evcc-io/evcc #17589: Display of charging sessions incorrect?

mbay0r, 2024-12-04:
### Describe the bug

[image]

I often have the problem that the charging sessions are often in the future.
Is this problem known and can it perhaps be fixed somehow?

Thank you

### Steps to reproduce

The problem persists even after a restart

### Configuration details

```yaml
interval: 60s
loadpoints:
  - title: CFOS
    charger: cfos
    meter: cfos
    vehicle: 500e

meters:
  - name: cfos
    type: template
    template: cfos
    usage: charge
# Modbus TCP
    modbus: tcpip
    id: 1
    host: XX
    port: 4701
  - name: my_grid
    type: template
    template: shelly-pro-3em
    usage: grid
    host: XX
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.28

vehicles:
  - name: 500e
    type: template
    template: fiat
    title: 500e
    user: XX
    password: XX
    capacity: 37
    pin: 4020
    phases: 3
#    mode: off
    mode: now
    identifiers:
      - XX
#    cache: 1m
chargers:
  - name: cfos
    type: template
    template: cfos
    host: XX # IP-Adresse oder Hostname
```

### Log details

```text
.
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.8

andig, 2024-12-04:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-12-04 by andig

mbay0r, 2024-12-04:
completed? really?

log since the last restart

```
[main  ] INFO 2024/12/04 21:07:25 evcc 0.131.8
[main  ] INFO 2024/12/04 21:07:25 using config file: /etc/evcc.yaml
[db    ] INFO 2024/12/04 21:07:25 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/12/04 21:07:25 listening at :7070
```

mbay0r, 2024-12-04:
@andig 

Should I now open a new issue or what was the purpose of setting the issue to resolved?