evcc-io/evcc #17606: Geladen wird nicht zurückgesetzt

stefan7201, 2024-12-05:
Habe leider immer wieder das Problem das die Geladen Anzeige nict zurückgesetzt wird.

EVCC auf Proxmox
Sungrow SH10RT + SBR160 + Sungrow Wallbox

Hat noch wer das Problem?

andig, 2024-12-05:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-12-05 by andig

stefan7201, 2024-12-06:
root@EVCC:~# evcc charger
[main  ] INFO 2024/12/06 16:18:25 evcc 0.131.8
[main  ] INFO 2024/12/06 16:18:25 using config file: /etc/evcc.yaml
[db    ] INFO 2024/12/06 16:18:25 using sqlite database: /root/.evcc/evcc.db
Power:          6910W
Energy:         1135.9kWh
Current L1..L3: 10.2A 10.2A 10.3A
Voltage L1..L3: 222V 224V 226V
Charge status:  C
Enabled:        true
Charged:        3.8kWh
Phases:         3

im Log steht Charged: 3,8 kWh 
[image]

in der übersicht steht aber 15,9kwh