evcc-io/evcc #17612: BMW 15min cache time not always considered - Out of call volume quota

endurance1968, 2024-12-06:
### Describe the bug

Sometime (observed it mainly at the end of a charging session but this might be accidentely) the 15min cache time to request new data from ar api it not kept.

### Steps to reproduce

1. just charge the car and watch out for quota exceeded warnings
2. in the logs attached you see request at 5:32 and then 5:36 again
3. "Out of call volume quota. Quota will be replenished in 00:21:37." as result

the door lock POST stuff after this I also do not understand

### Configuration details

```yaml
vehicles:
  - name: bmw-i3
    type: template
    template: bmw
    title: BMW i3
    hcaptcha: -
    user: -
    password: -
    vin: -
    capacity: 14
    icon: car
    phases: 1
    cache: 15m
    mode: PV
    minCurrent: 6
    maxCurrent: 32
    identifiers: # Kann meist erst später eingetragen werden
    priority: 10
    region: EU
    welcomecharge: true
```

### Log details

```text
Dec 06 05:16:53 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:16:53 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1733458613059
Dec 06 05:16:54 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:16:54 {"state":{"isLeftSteering":true,"lastFetched":"2024-12-06T04:16:53.979Z","lastUpdatedAt":"2024-12-06T04:16:54Z","isLscSupported":true,"range":87,
Dec 06 05:32:03 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:32:03 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1733459523071
Dec 06 05:32:03 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:32:03 {"state":{"isLeftSteering":true,"lastFetched":"2024-12-06T04:32:03.494Z","lastUpdatedAt":"2024-12-06T04:32:03Z","isLscSupported":true,"range":89,
Dec 06 05:38:23 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:38:23 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1733459903104
Dec 06 05:38:23 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:38:23 { "statusCode": 403, "message": "Out of call volume quota. Quota will be replenished in 00:21:37." }
Dec 06 05:39:33 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:39:33 POST https://cocoapi.bmwgroup.com/eadrax-vrccs/v3/presentation/remote-commands/***/door-lock
Dec 06 05:39:33 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:39:33 {"eventId":"f781e0f6-faa2-49a3-8b08-ec92fbf98804","creationTime":"2024-12-06T04:39:33.133069Z"}
Dec 06 05:40:53 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:40:53 POST https://cocoapi.bmwgroup.com/eadrax-vrccs/v3/presentation/remote-commands/***/door-lock
Dec 06 05:40:53 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:40:53 {"eventId":"e8a2537d-7e09-4182-a99c-1d8dd2b79bc3","creationTime":"2024-12-06T04:40:53.143431Z"}
Dec 06 05:42:13 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:42:13 POST https://cocoapi.bmwgroup.com/eadrax-vrccs/v3/presentation/remote-commands/***/door-lock
Dec 06 05:42:13 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:42:13 {"eventId":"b57d5397-3cf5-47f0-ba19-0617ad57d2a9","creationTime":"2024-12-06T04:42:13.129183Z"}
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.8

andig, 2024-12-06:
/door-lock

has nothing to do with cache. It's trying to wakeup the car since apparently charging did not start as expected.

closed 2024-12-06 by andig

endurance1968, 2024-12-06:
Hmm car was charging already almost finished, means the start of charging was working perfectly.
[image]

Topic was 
Dec 06 05:38:23 lx-evcc evcc[438673]: [bmw   ] TRACE 2024/12/06 05:38:23 { "statusCode": 403, "message": "**Out of call volume quota. Quota will be replenished in 00:21:37." }**

andig, 2024-12-06:
A single request certainly doesn't kill the volume.

endurance1968, 2024-12-06:
Vorher sind alle 15min Daten problemlos abgefragt worden.
Ich als Nutzer sehe ja auch nur die Fehlermeldung. Das tritt immer mal wieder auf. Es ist eben auffällig, dass in solchen Fällen der Abstand der Abfrage(n) deutlich kleiner 15min ist. Im obigen Fall 6min. Warum geht dann überhaupt eine Abfrage raus wenn cache auf 15 steht.

endurance1968, 2024-12-17:
```
Dec 16 23:04:16 lx-evcc evcc[60842]: [bmw   ] TRACE 2024/12/16 23:04:16 POST https://customer.bmwgroup.com/gcdm/oauth/token
Dec 16 23:04:17 lx-evcc evcc[60842]: [bmw   ] TRACE 2024/12/16 23:04:17 grant_type=refresh_token&redirect_uri=com.bmw.connected%3A%2F%2Foauth&refresh_token=HEl4vo-
Dec 16 23:04:17 lx-evcc evcc[60842]: [bmw   ] TRACE 2024/12/16 23:04:17 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1734386656959
Dec 16 23:04:18 lx-evcc evcc[60842]: [bmw   ] TRACE 2024/12/16 23:04:18 {"state":{"isLeftSteering":true,"lastFetched":"2024-12-16T22:04:17.993Z","lastUpdatedAt":"2024-12-16T18:46:37Z","isLscSupported":true,"range":47,"doorsState":{"combinedSecurityState":"SECURED","leftFront":"CLOSED","leftRear":"CLOSED","rightFront":"CLOSED","rightRear":"CLOSED","combinedState":"CLOSED","hood":"CLOSED","trunk":"CLOSED"},"windowsState":{"leftFront":"CLOSED","rightFront":"CLOSED","combinedState":"CLOSED"},"currentMileage":135533,"climateTimers":[{"isWeeklyTimer":false,"timerAction":"DEACTIVATE","timerWeekDays":[],"departureTime":{"hour":7,"minute":0}},{"isWeeklyTimer":true,"timerAction":"DEACTIVATE","timerWeekDays":["MONDAY"],"departureTime":{"hour":7,"minute":0}},{"isWeeklyTimer":true,"timerAction":"DEACTIVATE","timerWeekDays":["MONDAY"],"departureTime":{"hour":7,"minute":0}}],"departurePlan":{},"requiredServices":[],"checkControlMessages":[],"chargingProfile":{"chargingControlType":"WEEKLY_PLANNER","reductionOfChargeCurrent":{"start":{"hour":6,"minute":0},"end":{"hour":12,"minute":0}},"chargingMode":"IMMEDIATE_CHARGING","chargingPreference":"CHARGING_WINDOW","departureTimes":[{"id":1,"timeStamp":{"hour":12,"minute":0},"action":"DEACTIVATE","timerWeekDays":[]},{"id":2,"timeStamp":{"hour":10,"minute":0},"action":"DEACTIVATE","timerWeekDays":["SUNDAY"]},{"id":3,"timeStamp":{"hour":8,"minute":0},"action":"DEACTIVATE","timerWeekDays":["WEDNESDAY"]},{"id":4,"action":"DEACTIVATE","timerWeekDays":[]}],"climatisationOn":false,"chargingSettings":{"targetSoc":100,"idcc":"NO_ACTION","hospitality":"NO_ACTION"}},"electricChargingState":{"chargingLevelPercent":48,"range":47,"isChargerConnected":true,"chargingStatus":"NOT_CHARGING","chargingTarget":100},"combustionFuelLevel":{},"driverPreferences":{"lscPrivacyMode":"OFF"},"securityOverviewMode":null,"vehicleSoftwareVersion":{"puStep":{"month":7,"year":20},"iStep":{"seriesCluster":"I001","year":20,"month":7,"iStep":510}}},"capabilities":{"remoteServices":{"departureTimerControl":{"id":"departureTimerControl","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"inCarCamera":{"id":"inCarCamera","state":"NOT_AVAILABLE","executionApi":"RS_360_V4"},"inCarCameraDwa":{"id":"inCarCameraDwa","state":"NOT_AVAILABLE","executionApi":"RS_360_V4"},"remote360":{"id":"remote360","state":"NOT_AVAILABLE","executionApi":"RS_360_V4"},"surroundViewRecorder":{"id":"surroundViewRecorder","state":"NOT_AVAILABLE","executionApi":"RS_360_V4"},"windowControl":{"id":"windowControl","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"centralLockControl":{"id":"centralLockControl","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"batteryPreconditioningControl":{"id":"batteryPreconditioningControl","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"doorLock":{"id":"doorLock","state":"ACTIVATED","executionApi":"RS_API_V6"},"doorUnlock":{"id":"doorUnlock","state":"ACTIVATED","executionApi":"RS_API_V6"},"hornBlow":{"id":"hornBlow","state":"ACTIVATED","executionApi":"RS_API_V6"},"lightFlash":{"id":"lightFlash","state":"ACTIVATED","executionApi":"RS_API_V6"},"telematicsWakeup":{"id":"telematicsWakeup","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"wakeup":{"id":"wakeup","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"interiorPreconditioningControl":{"id":"interiorPreconditioningControl","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"}},"a4aType":"USB_ONLY","climateNow":true,"isClimateTimerWeeklyActive":false,"climateFunction":"AIR_CONDITIONING","horn":true,"isBmwChargingSupported":true,"isCarSharingSupported":false,"isChargeNowForBusinessSupported":true,"isChargingHistorySupported":true,"isLocationBasedChargingSettingsSupported":false,"isChargingHospitalityEnabled":false,"isChargingLoudnessEnabled":false,"isChargingPlanSupported":true,"isChargingPowerLimitEnabled":false,"isChargingSettingsEnabled":false,"isBatteryPreconditioningSupported":false,"isChargingTargetSocEnabled":false,"isCustomerEsimSupported":false,"isDataPrivacyEnabled":false,"isDCSContractManagementSupported":true,"isEasyChargeEnabled":false,"isMiniChargingSupported":false,"isEvGoChargingSupported":true,"isRemoteHistoryDeletionSupported":false,"isRemoteEngineStartSupported":false,"isRemoteServicesActivationRequired":false,"isRemoteServicesBookingRequired":false,"isScanAndChargeSupported":true,"lastStateCallState":"ACTIVATED","lights":true,"lock":true,"sendPoi":true,"unlock":true,"vehicleFinder":false,"vehicleStateSource":"LAST_STATE_CALL","isRemoteHistorySupported":true,"isWifiHotspotServiceSupported":false,"isNonLscFeatureEnabled":false,"isSustainabilitySupported":false,"isSustainabilityAccumulatedViewEnabled":false,"specialThemeSupport":[],"isRemoteParkingSupported":false,"isRemoteParkingEes25Active":false,"remoteChargingCommands":{},"digitalKey":{"state":"NOT_AVAILABLE","bookedServicePackage":"NONE","vehicleSoftwareUpgradeRequired":false,"isDigitalKeyFirstSupported":false},"isPersonalPictureUploadSupported":false,"personalPictureUpload":{"state":"NOT_AVAILABLE","aspectRa [... cut ...] ommerceFeatures":{"parking":false,"fueling":false,"reservations":false}}}
```
Wieder mal aber da das zufällig verteilt passiert, glaube ich auch nicht das es (nur) durch EVCC verursacht wird. Warum sollte sonst um 23.50:17 ein Quota fehler kommen und um 23:50:36 geht es wieder obwohl die Quato meldung sagte dauert noch 9min?
Wüsste nur nicht wer/was sonst die Daten mitten in der Nacht noch abfragen sollte. Kann man das Irgendwo bei BMW einsehen? Evtl. auch nur ein Bug auf der BMW Server seite...