evcc-io/evcc #17627: Grid Charging Control Based on Price Differences

Mosaik66, 2024-12-07:
Requirement Description:

The grid charging control for storage systems should be enhanced to allow management not only based on a maximum electricity price but also on a dynamic price difference. In this case, the future electricity price should be compared to the storage's own electricity price. The storage electricity price consists of the current feed-in tariff plus charging and discharging losses, which are considered at 15% by default. Grid charging should then occur when the future electricity price is at least 20% higher than this calculated storage price.

### Details:
Calculation of Storage Electricity Price:
Storage electricity price = Feed-in tariff + 15% charging/discharging losses.

**Trigger for Grid Charging:**
If the future electricity price > (storage electricity price + 20%), grid charging is activated.

**Flexibility:**
The percentages for charging/discharging losses (15%) and the price difference (20%) should be configurable.

Users should have the option to enable this function or choose between maximum price control and the new price difference-based control.

**Advantages of the Requirement:**
More efficient use of energy storage through dynamic adaptation to price fluctuations.

Reduction of costs for electricity procurement from the grid.

Greater flexibility for users with variable electricity tariffs or dynamic feed-in tariffs.

****Implementation Goal:
The feature should be intuitively configurable via the user interface and supported by clear parameters and visualizations.

Thanks!

andig, 2024-12-07:
All this logic is very specific and does imho not belong into evcc. Instead, we could expose the grid charging API and let an external party control that.

Mosaik66, 2024-12-07:
Thank you for your feedback! I understand your point that this specific logic might be highly tailored. However, since EVCC already supports grid charging functionality based on dynamic tariffs and price thresholds, this proposal could be viewed as a targeted enhancement rather than entirely new logic.

The suggested functionality—introducing configurable parameters for price differences and including factors like feed-in tariffs and charging/discharging losses—would build on the existing grid charging feature. This would provide more flexibility for users with specific economic considerations, without over-complicating the core EVCC logic.

Would it be feasible to extend the current grid charging functionality with these additional parameters? This way, the feature remains within EVCC but gains added configurability for advanced use cases. I’d be happy to assist with refining the details or testing any implementation to ensure it integrates seamlessly.

Looking forward to your thoughts!

andig, 2024-12-07:
„Advanced use cases“ is exactly what apis are for.

andig, 2024-12-07:
Pls join https://github.com/evcc-io/evcc/issues/17424

closed 2024-12-07 by andig