evcc-io/evcc #17634: Fronius Wechselrichter Primo 3.0-1: PV Ertrag wird doppelt gezählt

KingKahn123, 2024-12-07:
### Describe the bug

Ich habe 2 Wechselrichter in EVCC konfiguriert. Leider wird der Gesamt-PV-Ertrag nicht korrekt angezeigt. So wie es aussieht wird der PV-Ertrag von meinem Nebenwechselrichter (Primo 3.0-1) doppelt gezählt:

EVCC Meter:
[image]

Fronius Dashboard:
[image]

### Steps to reproduce

n/a, siehe Konfiguration unten

### Configuration details

```yaml
meters:
- name: HOMESWEETHOME_FRONIUS
  type: template # examples see https://github.com/andig/evcc-config#meters
  template: fronius-gen24
  usage: grid
  host: 192.168.33.70
  port: 502
- name: Wechselrichter_PV_Haupt
  type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.33.70
  port: 502
- name: Wechselrichter_PV_Neben
  type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.33.71
  port: 502
- name: Fronius_Batterie
  type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.33.70 # IP-Adresse oder Hostname
  port: 502 # Port (optional)

#Site definition (Standort und Hausinstallation)
site:
  title: Familie
  meters:
    grid: HOMESWEETHOME_FRONIUS
    pv:
      - Wechselrichter_PV_Haupt
      - Wechselrichter_PV_Neben
    battery: Fronius_Batterie
```

### Log details

```text
Siehe screenshots oben.
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

StefanSchoof, 2024-12-07:
Ist das ein Primo oder ein Gen24 Primo? Das Gen24 Template ist nur für die Gen24 Geräte.

benesolar, 2024-12-07:
Ist ein alter Primo. Erkennt man an der Weboberfläche vom Screenshot.

andig, 2024-12-07:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-12-07 by andig

KingKahn123, 2024-12-07:
Ja ist ein alter. Also müsste ich ein ganz anderes Template dafür nutzen? Für fronius selber gibt es ja nur die gen24.

benesolar, 2024-12-07:
Da gibt es kein eigenes Template.
Nur die Fronius Solar API welche mir zu langsam reagiert.

Nimm das:

  - name: Wechselrichter_PV_Neben
    type: template
    template: sunspec-inverter
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.33.71
    port: 502

KingKahn123, 2024-12-07:
@benesolar danke! Habe ich eingerichtet und werde morgen mal prüfen, ob es damit passt.