evcc-io/evcc #17640: Falsche Meter Anzeige

userwithoutpassword, 2024-12-07:
### Describe the bug

[image]
[image]
 Die Meter Anzeige ist inkorrekt und zeigt den identischen Start und Endwert an obwohl 2kWh Unterschied vorhanden sind.

Die Easee Wallbox ist in EVCC eingebunden. EVCC sendet per MQTT an Home Assistant den Zählerstand damit ich ihn auch in HA habe. In HA wird der Wert korrekt angezeigt, also muss EVCC den richtigen Wert eigentlich gehabt haben. Unter "Sessions" ist er jedenfalls falsch.

### Steps to reproduce

siehe oben

### Configuration details

```yaml
siehe oben
```

### Log details

```text
siehe oben
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.8

andig, 2024-12-07:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-12-07 by andig

userwithoutpassword, 2024-12-07:
log file added