evcc-io/evcc #17644: Meter Anzeige bei Wallbox falsch

userwithoutpassword, 2024-12-07:
### Describe the bug

[image]
[image]
Die Meter Anzeige ist inkorrekt und zeigt den identischen Start und Endwert an obwohl 2kWh Unterschied vorhanden sind.

Die Easee Wallbox ist in EVCC eingebunden. EVCC sendet per MQTT an Home Assistant den Zählerstand damit ich ihn auch in HA habe. In HA wird der Wert korrekt angezeigt, also muss EVCC den richtigen Wert eigentlich gehabt haben. Unter "Sessions" ist er jedenfalls falsch.

### Steps to reproduce

keine

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 15s # control cycle interval

influx:
  url: http://192.168.1.14:8086
  database: evcc # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  token: xxx
  org: Home

# mqtt message broker
mqtt:
  broker: 192.168.1.12:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: evcc
  user: evcc
  password: xxx

sponsortoken: xxx

meters:
  - name: grid
    type: custom
    # usage: grid
    power: # Leistung (W)
      source: mqtt
      broker: 192.168.1.12:1883
      user: evcc
      password: xxx
      topic: rct_power_storage/grid/power
      timeout: 30s
    Energy: # Energy (kWh)
      source: mqtt
      broker: 192.168.1.12:1883
      user: evcc
      password: xxx
      topic: rct_power_storage/grid/energy
      timeout: 30s
      scale: 0.001 # Wh => kWh

  - name: pv1
    type: custom
    # usage: pv
    power: # Leistung (W)
      source: mqtt
      broker: 192.168.1.12:1883
      user: evcc
[... 269 lines cut ...]
Dec 07 16:36:47 evcc evcc[15060]: [lp-1  ] INFO 2024/12/07 16:36:47 car disconnected
Dec 07 17:36:02 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:36:02 grid energy: outdated
Dec 07 17:36:17 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:36:17 grid energy: outdated
Dec 07 17:36:32 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:36:32 grid energy: outdated
Dec 07 17:36:47 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:36:47 grid energy: outdated
Dec 07 17:37:02 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:37:02 grid energy: outdated
Dec 07 17:37:17 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:37:17 grid energy: outdated
Dec 07 17:37:32 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:37:32 grid energy: outdated
Dec 07 17:37:47 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:37:47 grid energy: outdated
Dec 07 17:38:02 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:38:02 grid energy: outdated
Dec 07 17:38:17 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:38:17 grid energy: outdated
Dec 07 17:38:32 evcc evcc[15060]: [site  ] ERROR 2024/12/07 17:38:32 grid energy: outdated
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.131.8

andig, 2024-12-07:
Bitte erstmal die Unmengen Fehler im Log korrigieren, dann können spezifische Fehler analysiert werden.

closed 2024-12-07 by andig

userwithoutpassword, 2024-12-07:
Die Fehler sind Wechselrichter spezifisch