evcc-io/evcc #17652: Sungrow changed modbus registers

evilmon, 2024-12-08:
### Describe the bug

It looks like sungrow has changed some modbus registers.
in evcc I see that the home-battery is discharging but in fact the battery is charging with the same value. The value of my home consumpmtion is also wrong

### Steps to reproduce

1. start evcc
2. the the wrong values
3.
...

### Configuration details

```yaml
meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.40
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.40
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.40
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
```

### Log details

```text
no log errors
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.8

evilmon, 2024-12-08:
PS: I updated the inverter firmware yesterday to the latest version.

closed 2024-12-08 by andig

evilmon, 2024-12-08:
What is the original? Didn't find an open ticket with sungrow.

andig, 2024-12-08:
You're welcome to use the search function: https://github.com/search?q=repo%3Aevcc-io%2Fevcc+sungrow&type=issues