evcc-io/evcc #17662: Beschreibung für `minsoc` ändern?

VolkerK62, 2024-12-08:
Die Beschreibung in
https://github.com/evcc-io/evcc/blob/782f83c3e2826428d9e1c94b6fcab1bd70ec3873/util/templates/defaults.yaml#L157
bezieht sich auf die Fahrzeug Funktionalität, wird aber gleichzeitig auch bei dem steuerbaren Batteriespeichern genutzt (leider identische Bezeichnung). Dort passt die Beschreibung gar nicht.
Den Parameter `minsoc` gibt es zwar beim Vehicle noch, der wird doch aber nur noch im UI gesetzt.
Dann könnte man in der defaults.yaml die Beschreibung doch ändern, oder?

andig, 2024-12-10:
Blöd... ich sehe 3 Möglichkeiten:

1. Lowtech: Hilfstext bei jeder Verwendung neu und konsistent überschreiben. Nachteil: Pflegealptraum.
1. EInfach: Eine der Parameterverwendungen umbenennen. Nachteil: muss in allen Templates gemacht werden, breaking Change für die Anwender (@naltatis oder fällt uns da eine Migration ein?)
1. Hightech: Parameter identischen Namens aber unterschiedlicher Ausprägung Implementierung und je nach Kontext (einschließlich "fake" contextes wie z.B. "vehicle" anders interpretieren.

VolkerK62, 2024-12-10:
zu 2) "vehicle minsoc" gibt es doch gar nicht mehr im Template, oder?

naltatis, 2024-12-10:
@VolkerK62 Ja das ist so. Das Minsoc Property am Fahrzeug ist deprecated und der hinterlegte Text kommt nirgends mehr zur Anzeige. Im UI kommt die entsprechende Beschreibung aus Weblate. Heißt wir können einfach den Beschreibungstext auf den neuen Anwendungsfall (Batteriespeicher) anpassen.

andig, 2024-12-11:
Variante 0: einfach das Problem lösen. Top!

andig, 2024-12-11:
@naltatis doch nicht (ganz) top. Das ist ein temporärer Zustand- perspektivisch würde das Setting ja wieder in die Gerätekonfig wandern.

naltatis, 2024-12-11:
@andig nicht schlimm. Selbst wenn wir den MinSoc später wieder ans Fahrzeug (Config UI) verschieben fänd ich es nicht schlimm das zu einem festen Feld in der UI zu machen. Das gilt ja für alle Fahrzeuge (außer Offline).

closed 2024-12-23 by github-actions[bot]