evcc-io/evcc #17666: Database settings can't be deleted or overwritten from yaml

KanteTaete, 2024-12-08:
### Describe the bug

If - for Mercedes vehicles - the tokens are re-created and stored in the yaml configuration file, then the respective data in the database can't be overwritten by the data from the yaml file (e.g. by `evcc migrate --reset` and `evcc migrate` again).
If the data shall be entered via the UI config, then this is not accepted due to "invalid token".

### Steps to reproduce

All steps executed as sudo to prevent access permission problems:
1. Stop evcc service: `sudo systemctl stop evcc.service`
2. Reset database: `sudo evcc migrate --reset`
3. Re-generate tokens: `sudo evcc token [Mercedes_EQA250] `
4. Re-migrate from yaml: `sudo evcc migrate`
5. restart evcc service: `sudo systemctl start evcc.service`

### Configuration details

```yaml
Only respective vehicles section:

vehicles:
- name: Mercedes_EQA250
  title: Mercedes EQA250
  type: template
  template: mercedes
  capacity: 66 # battery capacity kWh
  user: <email-address>
  region: EMEA
  accesstoken: ....
  refreshtoken: 23qH...
  phases: 3
  mode: PV
  minCurrent: 6
  maxCurrent: 16
```

### Log details

```text
[main  ] INFO 2024/12/08 19:36:09 evcc 0.131.8
[main  ] INFO 2024/12/08 19:36:09 using config file: /etc/evcc.yaml
[db    ] INFO 2024/12/08 19:36:09 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/12/08 19:36:09 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/12/08 19:36:09 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/12/08 19:36:09 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/12/08 19:36:09 SELECT * FROM `settings` 35 <nil>
[db    ] TRACE 2024/12/08 19:36:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/12/08 19:36:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/12/08 19:36:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/12/08 19:36:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/12/08 19:36:10 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/12/08 19:36:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/12/08 19:36:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main  ] INFO 2024/12/08 19:36:10 listening at :7070
[db    ] TRACE 2024/12/08 19:36:10 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
...
[db    ] TRACE 2024/12/08 19:36:11 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
...
[db    ] TRACE 2024/12/08 19:36:11 SELECT * FROM `configs` WHERE `configs`.`class` = 3 0 <nil>
[mercedes] DEBUG 2024/12/08 19:36:11 identity.NewIdentity - database token found
[mercedes] DEBUG 2024/12/08 19:36:11 identity.NewIdentity - refreshToken started
[mercedes] TRACE 2024/12/08 19:36:11 POST https://id.mercedes-benz.com/as/token.oauth2
[mercedes] TRACE 2024/12/08 19:36:11 grant_type=refresh_token&refresh_token=VM2N...B0
--
{"error_description":"unknown, invalid, or expired refresh token","error":"invalid_grant"}
[main  ] ERROR 2024/12/08 19:36:11 creating vehicle Mercedes_EQA250 failed: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': token expired
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.8

KanteTaete, 2024-12-08:
Refer also discussion #17610

naltatis, 2024-12-08:
@KanteTaete Du hast das Fahrzeug nur über yaml konfiguriert, nicht über Config UI angelegt, korrekt?

Hintergrund: Der `evcc migrate` Befehl macht aktuell (noch) nichts mit Fahrzeugdaten. Aktuell migriert er nur die geräteunabhängigen Einstellungen aus der evcc.yaml. Daher ist das aktuelle Verhalten erst mal erwartungskonform. Perspektivisch wird das migrate Kommande aber auch Fahrzeuge, Zähler, Ladepunkte und Charger mit migrieren.

Zum eigentlichen Problem: @andig das klingt, als ob die Fahrzeugimplementierung beim Wechsel des Tokens in der YAML nicht korrekt mit einem existierenden Wert in der Datenbank umgeht. Das ist unabhängig von Config UI. Falls das nicht automatisch erkennbar ist, könnten wir auch ein dediziertes Token-Reset Kommando einführen. Schöner wäre natürlich ohne.

andig, 2024-12-08:
Ich kann zu Mercedes nichts sagen.

andig, 2024-12-09:
Das geht so einfach nicht da yaml Daten und migrierte DB Daten nix mit dem Problem zu tun haben. Das Mercedes „Runtime“ Token steht in der `settings` Tabelle, nicht in `config`.  Daher muss es- um es zu überschreiben- mit 

    evcc settings

gelöscht werden.