evcc-io/evcc #17669: Tesla Fahrzeugnamen wird nicht mehr angezeigt

mrdisc1479, 2024-12-09:
### Describe the bug

Seit Neuestem wird der Name meiner Autos nicht mehr angezeigt, sondern nur noch die definierte Variable aus der evcc.yaml. Access und refresh token funktionieren, die Namen werden von der Tesla-Api auch korrekt übermittelt. Siehe Log Details.
[image]

ich kann den Namen der Autos auch nicht manuell ändern. Im Ergebnis bringt das die Auswertung durcheinander.

[image]

### Steps to reproduce

1. Auto anschließen
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: tesla-command
    title: # Wird in der Benutzeroberfläche angezeigt (optional)
    accessToken: xxxxxxxxx
    refreshToken: xxxxxxxxxx
    vin: 5YJxxxxxxxxxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 77 # Akkukapazität in kWh (optional)
    mode: MinPV # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll 
  - name: my_car3
    type: template
    template: tesla-command
    title: # Wird in der Benutzeroberfläche angezeigt (optional)
    accessToken: xxxxxxxxxxxxxxx
    refreshToken: xxxxxxxxxxxxxxxxxxxxxxxxxxxx
    vin: XP7xxxxxxxxxxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 58 # Akkukapazität in kWh (optional)
    #icon: car # Icon in der Benutzeroberfläche (optional)
     mode: MinPV # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll
```

### Log details

```text
TRACE 2024/12/09 07:47:01 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/xxxxxxxxx/vehicle_data
[tesla ] TRACE 2024/12/09 07:47:01 {"response":{"id":929857208922601,"user_id":143782962866,"vehicle_id":1689275669228264,"vin":"xxxxxxx","color":null,"access_type":"OWNER","granular_access":{"hide_private":false},"tokens":null,"state":"online","in_service":false,"id_s":"929857208922601","calendar_enabled":true,"api_version":84,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false,"supercharger_payment_needed":"false","supercharging_enabled":true,"charge_state":{"battery_heater_on":false,"battery_level":41,"battery_range":92.44,"charge_amps":16,"charge_current_request":16,"charge_current_request_max":16,"charge_enable_request":true,"charge_energy_added":7.96,"charge_limit_soc":80,"charge_limit_soc_max":100,"charge_limit_soc_min":50,"charge_limit_soc_std":80,"charge_miles_added_ideal":32.5,"charge_miles_added_rated":32.5,"charge_port_cold_weather_mode":false,"charge_port_color":"FlashingGreen","charge_port_door_open":true,"charge_port_latch":"Engaged","charge_rate":42.7,"charger_actual_current":16,"charger_phases":2,"charger_pilot_current":16,"charger_power":11,"charger_voltage":231,"charging_state":"Charging","conn_charge_cable":"IEC","est_battery_range":84.3,"fast_charger_brand":"\u003cinvalid\u003e","fast_charger_present":false,"fast_charger_type":"\u003cinvalid\u003e","ideal_battery_range":92.44,"max_range_charge_counter":0,"minutes_to_full_charge":160,"not_enough_power_to_heat":null,"off_peak_charging_enabled":false,"off_peak_charging_times":"all_week","preconditioning_enabled":false,"preconditioning_times":"all_week","scheduled_charging_mode":"Off","scheduled_charging_pending":false,"scheduled_charging_start_time":null,"scheduled_departure_time":null,"supercharger_session_trip_planner":false,"time_to_full_charge":2.67,"timestamp":1733726821488,"trip_charging":false,"usable_battery_level":41,"user_charge_enable_request":null},"climate_state":{"allow_cabin_overheat_protection":true,"auto_seat_climate_left":true,"auto_seat_climate_right":true,"auto_steering_wheel_heat":true,"battery_heater":false,"battery_heater_no_power":null,"bioweapon_mode":false,"cabin_overheat_protection":"On","cabin_overheat_protection_actively_cooling":false,"climate_keeper_mode":"off","cop_activation_temperature":"High","defrost_mode":0,"driver_temp_setting":20.5,"fan_status":0,"hvac_auto_request":"On","inside_temp":5.3,"is_auto_conditioning_on":false,"is_climate_on":false,"is_front_defroster_on":false,"is_preconditioning":false,"is_rear_defroster_on":false,"left_temp_direction":656,"max_avail_temp":28.0,"min_avail_temp":15.0,"outside_temp":5.5,"passenger_temp_setting":20.5,"remote_heater_control_enabled":false,"right_temp_direction":656,"seat_heater_left":0,"seat_heater_rear_center":0,"seat_heater_rear_left":0,"seat_heater_rear_right":0,"seat_heater_right":0,"side_mirror_heaters":false,"steering_wheel_heat_level":0,"steering_wheel_heater":false,"supports_fan_only_cabin_overheat_protection":true,"timestamp":1733726821488,"wiper_blade_heater":false},"drive_state":{"power":-10,"shift_state":null,"speed":null,"timestamp":1733726821488},"gui_settings":{"gui_24_hour_time":true,"gui_charge_rate_units":"kW","gui_distance_units":"km/hr","gui_range_display":"Rated","gui_temperature_units":"C","gui_tirepressure_units":"Bar","show_range_units":false,"timestamp":1733726821488},"vehicle_config":{"aux_park_lamps":"NaPremium","badge_version":1,"can_accept_navigation_requests":true,"can_actuate_trunks":true,"car_special_type":"base","car_type":"modely","charge_port_type":"CCS","cop_user_set_temp_supported":true,"dashcam_clip_save_supported":true,"default_charge_to_max":true,"driver_assist":"TeslaAP3","ece_restrictions":true,"efficiency_package":"MYRWDEU2022","eu_vehicle":true,"exterior_color":"SolidBlack","exterior_trim":"Black","exterior_trim_override":"","has_air_suspension":false,"has_ludicrous_mode":false,"has_seat_cooling":false,"headlamp_type":"Global","interior_trim_type":"White2","key_version":2,"motorized_charge_port":true,"paint_color_override":"","performance_package":"Base","plg":true,"pws":true,"rear_drive_unit":"PM216MOSFET","rear_seat_heaters":1,"rear_seat_type":0,"rhd":false,"roof_color":"RoofColorGlass","seat_type":null,"sentry_preview_supported":false,"spoiler_type":"None","sun_roof_installed":null,"supports_qr_pairing":false,"third_row_seats":"None","timestamp":1733726821488,"trim_badging":"50","use_range_badging":true,"utc_offset":3600,"webcam_selfie_supported":true,"webcam_supported":true,"wheel_type":"Induction20Black"},"vehicle_state":{"api_version":84,"autopark_state_v2":"unavailable","calendar_supported":true,"car_version":"2024.44.25.2 288ce84bb43e","center_display_state":0,"dashcam_clip_save_available":true,"dashcam_state":"Recording","df":0,"dr":0,"fd_window":0,"feature_bitmask":"fbdffbff,ae5cbc7f,5","fp_window":0,"ft":0,"is_user_present":false,"locked":false,"media_info":{"a2dp_source_name":"Claudi XIII PRO","audio_volume":1.0,"audio_volume_increment":0.333333,"audio_volume_max":10.333333,"media_playback_status":"Playing","now_playing_album":"","now_playing_artist":"","now_playing_duration":0,"now_playing_elapsed":0,"now_playing_source":"13","now_playing_station":"DAB Beats Radio","now_playing_title":"Beats Radio ist eine Audiobrand von Klassik Radio."},"media_state":{"remote_control_enabled":true},"notifications_supported":true,"odometer":9895.987903,"parsed_calendar_supported":true,"pf":0,"pr":0,"rd_window":0,"remote_start":false,"remote_start_enabled":true,"remote_start_supported":true,"rp_window":0,"rt":0,"santa_mode":0,"sentry_mode":false,"sentry_mode_available":true,"service_mode":false,"service_mode_plus":false,"software_update":{"download_perc":0,"expected_duration_sec":2700,"install_perc":1,"status":"","version":" "},"speed_limit_mode":{"active":false,"current_limit_mph":76.922177,"max_limit_mph":120.0,"min_limit_mph":50.0,"pin_code_set":false},"timestamp":1733726821488,"tpms_hard_warning_fl":false,"tpms_hard_warning_fr":false,"tpms_hard_warning_rl":false,"tpms_hard_warning_rr":false,"tpms_last_seen_pressure_time_fl":1733724067,"tpms_last_seen_pressure_time_fr":1733724067,"tpms_last_seen_pressure_time_rl":1733724067,"tpms_last_seen_pressure_time_rr":1733724099,"tpms_pressure_fl":2.675,"tpms_pressure_fr":2.7,"tpms_pressure_rl":2.7,"tpms_pressure_rr":2.675,"tpms_rcp_front_value":2.9,"tpms_rcp_rear_value":2.9,"tpms_soft_warning_fl":false,"tpms_soft_warning_fr":false,"tpms_soft_warning_rl":false,"tpms_soft_warning_rr":false,"valet_mode":false,"valet_pin_needed":false,"vehicle_name":"F.R.Y.D.A.Y.","vehicle_self_test_progress":0,"vehicle_self_test_requested":false,"webcam_available":true}}}
-------------------------------------------------------------------------------------------------------
[... cut ...]
[db ] TRACE 2024/12/09 08:24:01 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`odometer`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`,`charge_duration`,`solar_percentage`,`price`,`price_per_kwh`,`co2_per_kwh`) VALUES ("2024-12-09 08:24:01.086","0000-00-00 00:00:00","Carport","","My_car",64299.47705354765,1564.89,NULL,0,NULL,NULL,NULL,NULL,NULL) RETURNING `id` 1 <nil>
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.8

andig, 2024-12-09:
Was steht unter /api/state?

VolkerK62, 2024-12-09:
Du hast keinen `title` angegeben. Ich denke mal, dann wird per default `name` genommen.

andig, 2024-12-09:
@VolkerK62 bei Tesla (und wenigen Anderen) sollte der aus dem API kommen.

mrdisc1479, 2024-12-09:
meinst du das?

{
  "result": {
    "auth": {

    },
    "battery": [
      {
        "power": -19,
        "energy": 581.74,
        "soc": 1,
        "controllable": true
      }
    ],
    "batteryCapacity": 1,
    "batteryDischargeControl": true,
    "batteryEnergy": 581.74,
    "batteryGridChargeActive": false,
    "batteryMode": "unknown",
    "batteryPower": -19,
    "batterySoc": 1,
    "bufferSoc": 95,
    "bufferStartSoc": 0,
    "currency": "EUR",
    "fatal": null,
    "greenShareHome": 0.204,
    "greenShareLoadpoints": 0,
    "gridConfigured": true,
    "gridCurrents": [6.16, 1.97, 0.82],
    "gridEnergy": 16194.768,
    "gridPower": 1775.85,
    "gridPowers": [1440.38, 283.94, 51.53],
    "hems": {

    },
    "homePower": 467.067,
    "influx": {
      "url": "",
      "database": "",
      "org": "",
      "user": "",
      "insecure": false
    },
    "interval": 30,
    "loadpoints": [
      {
        "batteryBoost": false,
        "chargeCurrent": 6,
        "chargeCurrents": [5.96600008, 0, 0],
        "chargeDuration": 0,
        "chargePower": 1384.983,
        "chargeRemainingDuration": 84506,
        "chargeRemainingEnergy": 32511.111,
        "chargeTotalImport": 1568.866,
        "chargedEnergy": 0,
        "chargerFeatureHeating": false,
        "chargerFeatureIntegratedDevice": false,
        "chargerIcon": null,
        "chargerPhases1p3p": true,
        "chargerPhysicalPhases": null,
        "chargerStatusReason": "unknown",
        "charging": true,
        "connected": true,
        "connectedDuration": 0,
        "disableDelay": 180,
        "disableThreshold": 0,
        "effectiveLimitSoc": 100,
        "effectiveMaxCurrent": 16,
        "effectiveMinCurrent": 6,
        "effectivePlanSoc": 0,
        "effectivePlanTime": null,
        "effectivePriority": 0,
        "enableDelay": 60,
        "enableThreshold": 0,
        "enabled": true,
        "limitEnergy": 0,
        "limitSoc": 0,
        "maxCurrent": 16,
        "minCurrent": 6,
        "mode": "minpv",
        "phaseAction": "scale1p",
        "phaseRemaining": 179,
        "phasesActive": 3,
        "phasesConfigured": 0,
        "phasesEnabled": 0,
        "planEnergy": 0,
        "planOverrun": 0,
        "planProjectedEnd": null,
        "planProjectedStart": null,
        "planTime": null,
        "priority": 0,
        "pvAction": "inactive",
        "pvRemaining": 0,
        "sessionCo2PerKWh": null,
        "sessionEnergy": 0,
        "sessionPrice": null,
        "sessionPricePerKWh": null,
        "sessionSolarPercentage": 0,
        "smartCostActive": false,
        "smartCostNextStart": null,
        "title": "Carport",
        "vehicleClimaterActive": null,
        "vehicleDetectionActive": false,
        "vehicleLimitSoc": 80,
        "vehicleName": "my_car",
        "vehicleOdometer": 0,
        "vehicleRange": 189,
        "vehicleSoc": 42,
        "vehicleWelcomeActive": false
      }
    ],
    "mqtt": {
      "broker": "Localhost:1883",
      "topic": "evcc",
      "user": "Mqtt",
      "insecure": true
    },
    "network": {
      "schema": "http",
      "host": "192.168.10.139",
      "port": 7070
    },
    "prioritySoc": 50,
    "pv": [
      {
        "power": 63,
        "energy": 1849.87
      },
      {
        "power": 32.19999695,
        "energy": 3442.352051
      }
    ],
    "pvEnergy": 5292.222,
    "pvPower": 95.2,
    "residualPower": 100,
    "siteTitle": "Home",
    "smartCostType": "pricestatic",
    "sponsor": {
      "name": "mrdisc1479",
      "expiresAt": "2027-08-31T12:00:00Z"
    },
    "statistics": {
      "30d": {
        "avgCo2": 0,
        "avgPrice": 0.241189862271489,
        "chargedKWh": 751.618,
        "solarPercentage": 3.52405509140443
      },
      "365d": {
        "avgCo2": 0,
        "avgPrice": 0.182886608735789,
        "chargedKWh": 1814.3665,
        "solarPercentage": 37.7884711385312
      },
      "thisYear": {
        "avgCo2": 0,
        "avgPrice": 0.182886608735789,
        "chargedKWh": 1814.3665,
        "solarPercentage": 37.7884711385312
      },
      "total": {
        "avgCo2": 0,
        "avgPrice": 0.182886608735789,
        "chargedKWh": 1814.3665,
        "solarPercentage": 37.7884711385312
      }
    },
    "tariffFeedIn": 0,
    "tariffGrid": 0.25,
    "tariffPriceHome": 0.199,
    "tariffPriceLoadpoints": 0.25,
    "vehicles": {
      "my_car": {
        "title": "My_car",
        "icon": "car",
        "capacity": 77,
        "features": [
          "CoarseCurrent"
        ]
      },
      "my_car3": {
        "title": "My_car3",
        "icon": "car",
        "capacity": 58,
        "features": [
          "CoarseCurrent"
        ]
      }
    },
    "version": "0.131.8"
  }
}

andig, 2024-12-09:
Wir holen den `display_name`, hier ist es in `vehicle_state.vehicle_name`. Um das zu verstehen bräuchte ich bitte ein vollständiges Trace Log des ganzen Logins und dann die Zugangsdaten (=Tokens), bitte an info@evcc.io.

mrdisc1479, 2024-12-09:
Hallo Andi,
Mail ist raus. 
VG

andig, 2024-12-09:
Die Antwort ist klar:

    "display_name": null,

Frage an Tesla: wird der Nickname jetzt an anderer Stelle gespeichert oder ist das ein temporärer Fehler im API? Früher war er da...

closed 2024-12-11 by andig