evcc-io/evcc #17670: eebus lets evcc crash on docker

swestland85, 2024-12-09:
### Describe the bug

Trying to get eebus up and running on a evcc running on docker on a synology NAS (DS416play). 

When adding eebus to the yaml config I get the following error:

> [main  ] FATAL 2024/12/03 14:09:12 failed configuring eebus: could not get machineid: open /etc/machine-id: no such file or directory; for manual configuration use plant: caf5c742810cf7bda8e093d2787c9c88b16762700a9b8125ad8807f60bcdcf85

Adding file machine-id file to the file mapping and adding the suggested machine-id. 

Now evcc will start, however the webconfig never becomes accessible and the docker will reboot after 15 minutes going into an infinite loop. 

Adding plant: to the yaml creates the same problem. 

### Steps to reproduce

1. add eebus to yaml 
2. add plant: for the machine-id

### Configuration details

```yaml
#plant: fd2371f0ba8725a310c23f7e8e708afa77461493ac9b9794ae67f82da7bc4d9f
site:
  - title: Blaricum # display name for UI
    meters:
      grid: my_grid # grid meter reference
    residualPower: 100
tariffs:
  grid:
    type: template
    template: tibber
    token: removed
    homeid: removed
loadpoints:
  - title: Oprit # display name for UI
    charger: my_charger # charger reference
    vehicle: BMW # reference to standard vehicle
    mode: minpv # charge mode (off, now, minpv, pv)    
vehicles:
  - name: BMW
    type: template
    template: bmw
    title: X5
    user: removed # User account (e.g. email address, user id, etc.)
    password: remvoed # Password of the user account (use single quotes in case of leading zeros)
    capacity: 26 # Battery capacity in kWh (optional)
    hcaptcha: [hcaptcha]
meters:
  - name: Enphase
    type: template
    template: enphase
    usage: pv
    host: 192.168.2.213 # IP address or hostname
    token: [token]
eebus:
  shipid: EVB-510-061-355
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
      MIIBvTCCAWOgAwIBAgIRAcyYJVawVRQ+VsR8DLaj9qkwCgYIKoZIzj0EAwIwPjEL
      MAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0MxCTAHBgNVBAsTADEVMBMGA1UEAwwM
      RVZDQ19IRU1TXzAxMB4XDTI0MTIwMzEyMzAwNVoXDTM0MTIwMTEyMzAwNVowPjEL
      MAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0MxCTAHBgNVBAsTADEVMBMGA1UEAwwM
      RVZDQ19IRU1TXzAxMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAESOkdLGhclC5J
      FKD6w9oNElBrd2WOHOfUgMbqUS9JyFCXTkCk0yxhQEM9vsW2okyc70sp0NrK0K/F
      Xfv+5Seq0aNCMEAwDgYDVR0PAQH/BAQDAgeAMA8GA1UdEwEB/wQFMAMBAf8wHQYD
      VR0OBBYEFGQd3P1IkhdZr6dF2xQgDGlLTTzNMAoGCCqGSM49BAMCA0gAMEUCIQCU
      POMAl3b/SUPruDNWHAQd+ufzZr3yxw8eqEWHQk8nxAIgBRH/Al5sH0V8qMGI3iAS
      znwH07ONCF5/f2LhRcRCOFA=
      -----END CERTIFICATE-----
    private: |
      -----BEGIN EC PRIVATE KEY-----
      MHcCAQEEIGpNtxIpfMFOHXlR4vDHbgyHZsC3aoKBg+w0lBR/9aS7oAoGCCqGSM49
      AwEHoUQDQgAESOkdLGhclC5JFKD6w9oNElBrd2WOHOfUgMbqUS9JyFCXTkCk0yxh
      QEM9vsW2okyc70sp0NrK0K/FXfv+5Seq0Q==
      -----END EC PRIVATE KEY-----
chargers:
  - name: my_charger
    type: template
    template: eebus
    ski: EVB-510-061-355
#    ski: mmm5PVZj69EGjnpxVa41gWQiU
    ip: 192.168.2.187 # IP address (optional)
```

### Log details

```text
[main  ] FATAL 2024/12/03 14:09:12 failed configuring eebus: could not get machineid: open /etc/machine-id: no such file or directory; for manual configuration use plant: caf5c742810cf7bda8e093d2787c9c88b16762700a9b8125ad8807f60bcdcf85
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.8

andig, 2024-12-09:
There is no crash. Error message says it all.

closed 2024-12-09 by andig

swestland85, 2024-12-09:
> There is no crash. Error message says it all.

Well if I add plant: or map machine-id there is no error, it just hangs and keep restarting.

swestland85, 2024-12-09:
> > Well if I add plant: or map machine-id there is no error, it just hangs and keep restarting.
> 
> Do you have a log?

Yes: 

> 
> 2024-12-09T09:56:24.296574089Z	stdout	[main  ] INFO 2024/12/09 10:56:24 listening at :7070
> 2024-12-09T09:56:24.289152171Z	stdout	[eebus ] INFO 2024/12/09 10:56:24 Local SKI: 641ddcfd48921759afa745db14200c694b4d3ccd
> 2024-12-09T09:56:23.894180721Z	stdout	[db    ] INFO 2024/12/09 10:56:23 using sqlite database: /root/.evcc/evcc.db
> 2024-12-09T09:56:23.838477672Z	stdout	[main  ] INFO 2024/12/09 10:56:23 using config file: /etc/evcc.yaml
> 2024-12-09T09:56:23.830159339Z	stdout	[main  ] INFO 2024/12/09 10:56:23 evcc 0.131.8
> 
> 

That's it, webconfig doesnt come online and it will reboot in 15 minutes.

andig, 2024-12-09:
Again, error message says it all.

    cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'eebus': timeout