evcc-io/evcc #17678: Polestar - creating vehicle polestar failed: resumePath not found

fblix, 2024-12-09:
### Describe the bug

Hallo zusammen,

es scheint mal wieder ein Problem mit der Polestar API zu geben.
Mein EVCC zeigt aktuell 0% Ladestand bei meinem Polestar 2 an - auch nach einem Neustart.
Sonderlich viel eingestellt habe ich nicht, den Log mit dem Fehler habe ich angehängt.

### Steps to reproduce

1. EVCC starten
2. Polestar anschließen

### Configuration details

```yaml
N/A
```

### Log details

```text
[main ] ERROR 2024/12/09 23:58:49 creating vehicle polestar failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': login failed: resumePath not found
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.131.8

andig, 2024-12-10:
Seems Polestar is making changes to the Api. Someone will need to find out which those are. /cc @loebse

fblix, 2024-12-10:
FYI: Although evcc seems to have issues with those changes, [pyPolestar](https://github.com/pypolestar/polestar_api) seems to continue to work just fine.

andig, 2024-12-10:
That's good to know. Which changes did pyPolestar make?

fblix, 2024-12-10:
Just did some light digging. Seems I was wrong and pyPolestar experiences the same issue:

`This error originated from a custom integration.

Logger: custom_components.polestar_api.polestar
Source: custom_components/polestar_api/polestar.py:84
integration: Polestar API (documentation, issues)
First occurred: December 9, 2024 at 4:01:28 PM (1 occurrences)
Last logged: December 9, 2024 at 4:01:28 PM

Unexpected Error on update data cannot access local variable 'resumePath' where it is not associated with a value`

fblix, 2024-12-10:
This issue on the pyPolestar repo suggets the auth flow has been changed: https://github.com/pypolestar/polestar_api/issues/261

BJReplay, 2024-12-10:
> Do we have a trace of the requests executed?

@andig I've provided one (it's a bit of a kludge of screen grabs because I don't have a paid copy of HTTP Toolkit) of the Android app auth flow via PM to @jschlyter to assist with https://github.com/pypolestar/polestar_api/issues/261 - if he gets pypolestar auth up and running, I imagine you'll see the changes PR'd into that repo pretty quickly.

jahoooo, 2024-12-10:
Good that it's already being addressed!
Due to this issue and polestar/evcc(?) reporting 0% SOC, evcc just started charging full throttle.
I have set minimum SOC 40% in evcc. Guess the PV excess triggered the loading of the car and then due to 0% falsely reported it went to full throttle.

rostbeule, 2024-12-10:
The login process stopped working due to changes in the Polestar authentication system, which now requires the use of the OAuth 2.0 Authorization Code Flow with PKCE (Proof Key for Code Exchange). This update enforces additional security measures by introducing PKCE parameters (code_verifier and code_challenge) during the authentication process. As a result of this update, the workaround involving the GraphqlService is no longer necessary.

If you're familiar with C#, feel free to take a look at my repository [PolestarApi](https://github.com/rostbeule/PolestarApi). I've outlined and documented the authentication flow there. The logic just needs to be translated into Go, and everything should work smoothly again. Unfortunately, I'm not skilled enough to do that myself.

By the way, consent handling should no longer be necessary, but I’ve still left it in the code for now.

loebse, 2024-12-10:
This became a bit of a bigger change based on your example @rostbeule but it works for me again:

https://github.com/evcc-io/evcc/pull/17683

rostbeule, 2024-12-10:
@loebse 
Du bist ein Fuchs :D Ich wollte mich soeben dran setzen. Ich muss mal das Projekt lokal zum laufen bringen, dann kann ich  besser unterstützen. Besten Dank!

closed 2024-12-10 by andig