evcc-io/evcc #17681: evcc GELADEN kWh springt von 0 auf 1 bei 0,75 go-eCharger, auf 2 bei 1,75 ...

jfraundo251158, 2024-12-10:
### Describe the bug

Hallo, 
in meiner Installation werden in evcc 0.126.2 die geladenen kWh auf Dezimalkomma hochgezählt und dargestellt ( 1,0 1,1 1,2 ... ). In späteren Versionen und der aktuellen 0.131.8 springt die Darstellung von 1,0 auf 2,0 auf 3,0 usw. und dies bei go-eCharger 0,75, 1,75, 2,75 Werten usw.

Bitte nachfolgende Screenshots betrachten
Bild 1: 14:09 evcc - geladen 2,0 kWh
[image]

Bild 2: 14:09 Charger - geladen 2,72 kWH
[image]

Bild 3: 14:10 evcc - geladen 3,0 kWh
[image]

Bild 4: 14:10 Charger - geladen 2,75 kWH
[image]

Attached Debug und Trace Logs
[evcc-20241210-141131-debug.log.zip](https://github.com/user-attachments/files/18081054/evcc-20241210-141131-debug.log.zip)
[evcc-20241210-141141-trace.log.zip](https://github.com/user-attachments/files/18081057/evcc-20241210-141141-trace.log.zip)

Was tun?

Danke,
   Juergen

### Steps to reproduce

.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

#log: info, debug, trace
log: info
levels:
  cache: error

# unique installation id
plant: ???? - entfernt

interval: 10s # control cycle interval

sponsortoken: ???? - entfernt

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.329 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.09 # [currency]/kWh

meters:
- type: template
  template: sma-home-manager 
  usage: grid
  host: 192.168.178.44
  name: grid1
- type: template
  template: sma-energy-meter
  usage: pv
  host: 192.168.178.115
  name: pv2
#- type: template
#  template: sma-inverter
#  usage: pv
#  host: 192.168.178.46
#  password: PV-anlage1
#  name: pv2
- type: template
  template: sma-inverter 
  usage: battery
  host: 192.168.178.57
  password: PV-anlage1  
  name: battery3

chargers:
- type: template
  template: go-e
  host: 192.168.178.43
  name: wallbox4

loadpoints:
- title: Car Port
  charger: wallbox4
  mode: off

site:
  title: Weitnau Home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100
```

### Log details

```text
Log Files in der Beschreibung attached
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.8

VolkerK62, 2024-12-10:
schau mal, könnte das der Grund sein?  https://github.com/evcc-io/evcc/issues/16296

andig, 2024-12-10:
Leider: https://github.com/evcc-io/evcc/pull/14730

closed 2024-12-10 by andig

jfraundo251158, 2024-12-10:
Habe mir #14730 angeschaut - aber nicht verstanden.
Werde dann wohl bei der 0.126.2 bleiben : - (

tm-107, 2024-12-11:
Also wenn ich es richtig verstanden habe, dann wird der Gesamtenergie-Zähler (eto) benutzt [welcher nur ganze kWh anzeigt], weil der Ladeeneergie-Zähler (dws) von der Box auf null gesetzt wird, sobald während des Ladevorgangs von 1p auf 3p (oder umgekehrt) umgeschaltet wird oder der Ladevorgang zu oft unterbrochen wird.

Der Fix gilt jetzt leider für alle go-e-Wallboxen, obwohl scheinbar nur die v3 von dem Problem betroffen sind.
Die v1 & v2 können überhaupt keine Phasenumschaltung. Und an meiner v2 habe ich mein Fahrzeug über Tage angeschlossen gehabt mit unzähligen Ladeunterbrechnungen, der Session-Wert (dws) wurde dabei nie resettet.

Die 0.126.2 kann ich leider nicht verwenden, da ich den Nissan-Fix der 0.126.4 brauche.
Theoretisch müssten aber alle Versionen bis 0.128.1 genaue Werte anzeigen, da der genannte Fix erst danach kam. Das teste ich jetzt mal ...
Wenn ich es könnte, würde ich versuchen eine aktuelle evcc-Version ohne diesen Fix zu kompilieren ... kann ich aber nicht :(

jfraundo251158, 2024-12-15:
Danke für die Erklärung tm-107. Würde ich auch so machen - alternativ eine Fallunterscheidung zu den Wallboxen implementieren. Musste ich in meinem Metier / Branche auch öfters machen. Geht nicht immer alles nur nach der reinem Lehre : - )

andig, 2024-12-15:
Wenn jemand in der Lage ist das Problem auf bestimmte Boxen ein zugrenzen die wir auch erkennen können dann lasst uns das gerne wieder ändern!

premultiply, 2024-12-15:
Alternativ könnte der Hersteller auch seine Firmware fixen und den Gesamtladezähler mit entsprechend größerer Genauigkeit exportieren. Problem gelöst.

andig, 2024-12-15:
Gute Idee. Hat schon jemand ein Ticket bei GoE aufgemacht? Gabs eine Reaktion?

tm-107, 2024-12-16:
Das habe ich in der Tat mal getan und mich über die grobe Auflösung des Gesamtzählers bei der v2 beschwert. 
Als Antwort kam dann sinngemäß, daß eine Auflösung von 1 kWh für den Gesamtzähler vollkommen ausreichend sei, da ja der Ladezähler in dWs aufgelöst ist (was tatsächlich schon sehr fein ist).
Und dieser wird bei der v2 -wie schon erwähnt- erst auf null gesetzt, wenn die Verbindung zum Fahrzeug unterbrochen wird (zumindest bei mir).

Beim go-e würde ich bisher dabei bleiben, dass Probleme mit zurückgesetzten Zählern nur bei der v3 auftreten (weil v2 ja keine Phasenumschaltung kann). Und da ist die Unterscheidung ja schon übers Template gegeben.

andig, 2024-12-16:
> Als Antwort kam dann sinngemäß, daß eine Auflösung von 1 kWh für den Gesamtzähler vollkommen ausreichend sei, da ja der Ladezähler in dWs aufgelöst ist (was tatsächlich schon sehr fein ist).

Bei der v3 geht das ja offensichtlich nicht weil die Sessionlogik kaputt ist. Gibts dazu ein Ticket?

tm-107, 2024-12-17:
Ticket [223](https://github.com/goecharger/go-eCharger-API-v2/issues/223) und [248](https://github.com/goecharger/go-eCharger-API-v2/issues/248) dürften mit dem Problem zu tun haben, gab aber jeweils keine Reaktion.
Ich habe meine Antwort damals auch nur über den direkten Mail-Support bekommen.

Da ich keine v3 habe, kann ich jetzt nur raten: Der API-Wert "wh" wird bei Phasenumschaltung auf null gesetzt?

Bei der [alten API-Version](https://github.com/goecharger/go-eCharger-API-v1/blob/master/go-eCharger%20API%20v1%20DE.md) war das ja bei v2 und v3 "dws", jetzt ist es bei den neuen Modellen und der [neuen API](https://github.com/goecharger/go-eCharger-API-v2/blob/main/apikeys-de.md) aber wohl "wh".

tm-107, 2024-12-17:
Mir ist aber heute auch wieder aufgefallen, dass evcc vor dem workaround (ich verwende derzeit 0.128.1) den Wert selber auf null setzt, obwohl das Fahrzeug nicht getrennt wurde ...
Ich hatte kurz schnell geladen und dann auf PV mit Plan umgestellt. Dabei wurde die geladene Energie auf null gesetzt, obwohl von der go-e weiterhin der korrekte Wert übermittelt wurde (API-Wert dws).

andig, 2024-12-17:
Es lohnt sich nicht über vermeintliche Fehler in 1 Jahr alter Software zu sprechen. Falls das eine aktuelle Version wäre und es ein Logfile gäbe würden wir das beheben.

tm-107, 2024-12-17:
Ob das Problem in der aktuellen Version ebenfalls existiert kann ich nicht sagen, da aufgrund des Workaround für die go-e ja nur der Gesamtzähler ausgewertet wird ;-)

tm-107, 2025-01-08:
> Wenn jemand in der Lage ist das Problem auf bestimmte Boxen ein zugrenzen die wir auch erkennen können dann lasst uns das gerne wieder ändern!

Kann ich hier noch auf Aktivität hoffen?
Wie gesagt, der go-e v2 ist nach bisherigem Kenntnisstand nicht von dem Fehler betroffen.

andig, 2025-01-08:
Bei Go-E? Das weiss nur der Hersteller. Ansonsten hat sich seit https://github.com/evcc-io/evcc/issues/17681#issuecomment-2543984232 nichts geändert. Lässt sich eine v2 erkennen? Wie?

tm-107, 2025-01-08:
Zur Unterscheidung habe ich in https://github.com/evcc-io/evcc/issues/13726#issuecomment-2547240236 was geschrieben ...

andig, 2025-01-08:
@tm-107 könntest Du den PR bitte testen?

tm-107, 2025-01-08:
Sorry, aber da fehlen mir die mentalen Voraussetzungen zum kompilieren ...
Ich kann leider nur release & nightly testen :-(

tm-107, 2025-01-19:
Okay, Release ist raus ... ich werde es testen.
Aber leider erst Ende der Woche ...

jfraundo251158, 2025-01-22:
Auf v0.133.0 aktualisiert - alles OKAY - VIELEN DANK !