evcc-io/evcc #17682: Arrival charging plan: Allow selecting "Min+Solar" instead of "Fast"

capi, 2024-12-10:
**Is your feature request related to a problem? Please describe.**
Not a problem, just an improvement: When I arrive at home with a SoC below my target minimum, I really appreciate the automatic charging until the selected minimum charge. However, most of the days I'm not in a hurry and would rather prefer "Min+Solar" instead of "Fast" charging mode. The reasoning behind it is, that it a) reduces load on my main power connection and b) that it maximizes chance that there might be some more solar charging involved.

**Describe the solution you'd like**
In the "Arrival" tab, add a "Charging mode" selector that allows for selecting "Fast" (current behavior) or "Min+Solar" for the desired charging mode. The last selected mode should be remembered per load-point.

As a mock:
[image]

**Describe alternatives you've considered**
None. I've considered writing an external script that monitors EVCC via the API and adjusts plans/limits/... dynamically to achieve this, but I'd rather prefer this to be a built-in solution.

naltatis, 2024-12-10:
Your desired per-vehicle-option already exists. You can achieve this now by setting the `mode` property in your vehicle entry in the evcc.yaml. See the advanced options in the [docs](https://docs.evcc.io/docs/devices/vehicles).

> The last selected mode should be remembered per load-point.

This is the default behavior if you dont have a default mode specified at you loadpoint or vehicle the last mode will be rememberd.

In experimental UI you can set the default mode per vehicle.
[image]

With the next update this will also be possible per loadpoint. see https://github.com/evcc-io/evcc/pull/12958

closed 2024-12-10 by naltatis

capi, 2024-12-10:
@naltatis The default mode when connecting the vehicle works, but when I have an arrival plan, it does so in "Fast mode" regardless of the configured charging mode of the loadpoint. What I want to influence is the mode when charging to the minimum configured arrival SoC.

naltatis, 2024-12-10:
Ok, so you want an option to modify the MinSoC charging behavior in general?

One of the larger topics on our roadmap is a mode redesign. Our goals is to offer more flexibility for different needs and usecases. E.g. situations without pv (dynamic prices only). We have a discussion thread to collect ideas around this: https://github.com/evcc-io/evcc/discussions/3530
If you post your use-case there, we'll have it present once we address this topic. I don't see this as a short-term option that we'll implement before this redesign.

capi, 2024-12-10:
Thanks for the feedback and will add the case in the mentioned ticket. Thanks for taking time to answer here.

naltatis, 2024-12-10:
Thanks for clarifying!