evcc-io/evcc #17687: Fronius GEN 24 - Load battery from grid - set max loading speed by config

florianfiby, 2024-12-11:
**Is your feature request related to a problem? Please describe.**
When charging from grid, fronius-gen24.yaml sets charging speed to 100%. This leeds on my 8kw fronius gen24 to 7,5kw grid power usage. When my ID.4 is charging at night with 11kw and my heat pump is running I'll get over 20kw grid usage.

**Describe the solution you'd like**
As a first step it would be great to set a optional config value between 1 and 100% charging speed over fronius-gen24.yaml
Also for not stressing the battery, I'would like to load the battery with no more than 4kw.

**Describe alternatives you've considered**
I'will test circuits and check what happens when I'll set 15kw for my overall grid usage.

**Additional context**

andig, 2024-12-11:
> When my ID.4 is charging at night with 11kw and my heat pump is running I'll get over 20kw grid usage.

Is that an issue?

> I'will test circuits and check what happens when I'll set 15kw for my overall grid usage.

As documented, battery grid charging is not handled by the load management.

closed 2024-12-13 by andig

savus4, 2024-12-13:
@florianfiby Could you already try this? I just set the  `maxchargerate: 35` , but the battery was still charged with full power (in my case 10kW) from the grid. Did I understand it wrong?

andig, 2024-12-17:
Did you open a ticket with Fronius? What do they say?

florianfiby, 2024-12-17:
> @florianfiby Could you already try this? I just set the `maxchargerate: 35` , but the battery was still charged with full power (in my case 10kW) from the grid. Did I understand it wrong?

I will give it next week a try.

NFea-r, 2024-12-17:
@andig Seems Fronius makes it extra hard to contact them. Unfortunately I did not find any support address on their website. They seem to offer support only to partners. Nevertheless, in case I find a way to contact their support, is there any information about the way evcc sets maxchargerate? E.g. a specific register that is adressed via modbus to get the information to the inverter?

andig, 2024-12-17:
Ich hab mir das grad nochmal angeschaut. Wir verwenden Sunspec Model 124 `OutWRte = -100%` fürs Laden. Das ist eigentlich schon sehr merkwürdig und ich frage mich darum das _überhaupt_ funktioniert. Eigentlich müsste man dafür eher `InWRte = 100%` verwenden. Das müsste aber mal jemand ausprobieren, auch für die anderen Modi. 

/cc @MikeTheTux von dem die Idee mal kam https://github.com/evcc-io/evcc/issues/10912 (und  @Steph-74 https://github.com/evcc-io/evcc/discussions/14986)

Steph-74, 2024-12-17:
Ich bin jetzt nicht der Sunspec-Kenner, habe aber eine eigene Ladesteuerung für meinen Fronius Gen 24 10.0 mit 10KWh BYD über Modbus in Home Assistant gebaut, damit der Heimspeicher "sanft" geladen wird (max. 0,5C, SoC nach oben hin noch viel weniger). Ich weiß nicht, ob das Sunspec-Standard ist, aber im Gen24 bedeuten die Werte folgendes:

- InWRte begrenzt die Ladeleistung. z.B. auf 50 % (~5kW). Die Batterie gönnt sich **max.** ~5 kW Ladeleistung.
(Wenn mehr wie 5KW von der PV kommen wird der Rest ins Netz eingespeist)
- Für eine Zwangsladung muss der OutWRte negativ gesetzt werden, z.B. -50% für Zwangsladung mit **min.** ~5kW. 
(Wenn die PV Produktion nicht ausreicht, wird der Rest aus dem Netz geholt)
- Die erzwungene Aufladung wird wiederum durch den InWRte begrenzt, sprich steht der InWRte auf 50 % und der OutWRte auf -50% wird mit konstant ~5kW geladen bis das BMS eingreift und nach oben hin die Leistung reduziert.

Zwangsentladung der Batterie geht genau umgekehrt, sprich InWRte muss negativ sein und der OutWRte begrenzt die Entladeleistung.

Musste ziemlich viel rumfrickeln, bis ich das im Home Assistant stabil am laufen hatte ... wenn ihr mal jemanden zum testen braucht bin ich gerne bereit zu unterstützen.

andig, 2024-12-17:
Anscheinend passiert das nicht. Mit https://github.com/evcc-io/evcc/pull/17691/files#diff-2f5afe5b69f083165dced037c8eaafecbc65e4b7f8ff14d2856f49bb3bd90755R218 wird die Ladung durch OutWRte begrenzt, scheint aber dennoch mit voller Leistung zu erfolgen. Das ist dann ggf. ein FW-spezifischer Bug oder das zusätzliche Setzen von InWRte ist zusätzlich notwendig?

NFea-r, 2024-12-17:
Ich verwende Firmware 1.34-5.1 (es gibt noch eine neuere, die nach Chagelog aber nur einen Bug während des Updates adressiert). Laut der Doku von Fronius unter [https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42,0410,2649.pdf](url) Seite 34 wird wohl für die Begrenzung der Ladeleistung InWRte benötigt. OutWRte ist für die Entladeleistung.

Steph-74, 2024-12-17:
@NFea-r: ich habe das ausgiebig getestet:

Ja, der InWRte begrenzt die Ladeleistung, sprich aus meinem Beispiel 50%: wenn 10 kW von der PV kommen gehen nur 5 kW in die Batterie. Du musst dem WR aber über einen negativen (!) OutWRte sagen, dass eine Ladung erzwungen (!) werden soll. Nur dann lädt die Batterie auch aus dem Netz (sofern Laden aus dem Netz im Techniker Menü aktiviert ist, der Storage Control richtig gesetzt ist, nicht mehr wie 5kW von der PV kommen etc.)

@andig, aus meiner Erfahrung: InWRtr >= |OutWRte| (Absoluter OutWRte). 

ich bin kein Developer, aber wenn ich es richtig sehe wird in der defaults.yaml der maxchargepower auf 100 gesetzt. Wenn ich den OutWRte  über die gen24.yaml dann auf -100 setze lädt er mit voller Leistung, da die -100% die minimale Ladeleistung sind...

andig, 2024-12-17:
Könnt ihr einfach mal probieren, wie wir die forcierte Ladeleistung begrenzen können? These: `OutWRte` ist wirkungslos (muss aber auf -100% bleiben) und `InWRte` muss zusätzlich gesetzt sein?

Steph-74, 2024-12-17:
Hallo @andig,
wenn ich es richtig sehe ist das im aktuellen Release 0.131.11 implementiert. Ich probiere das mal heute Abend aus ... ich glaube, ich komme im Home Assistant nicht an defaults.yaml ran, um die 100 auf z.B. 50 zu ändern (zumindest weiß ich nicht, wie ich sie ändern kann). 

Bzgl. These ... leider nein: Im Normalfall lade ich nur zwangsweise nach, wenn die PV nicht genügend bringt. Dann kann ich das ausschließlich über den negativen OutWRte steuern, der den Minimumwert für die Zwangsladung darstellt. 
Wenn die PV aber 10kW runterknallt und ich will nur mit 5 kW Laden, muss ich den InWRte auf 50 setzen, um die Ladeleistung zu begrenzen. 
Will ich den Akku konstant mit 5kW zwangsladen müssen beide Werte gesetzt werden, OutWRte = -50 um eine Zwangsladung mit min. 5kW zu erzwingen und InWRte auf 50 um die Ladung auf 50 zu begrenzen, wenn die PV mir z.B. 10 kW runterknallt.

Gruß
Stephan

andig, 2024-12-17:
> Will ich den Akku konstant mit 5kW zwangsladen müssen beide Werte gesetzt werden, OutWRte = -50 um eine Zwangsladung mit min. 5kW zu erzwingen

Was passiert denn ohne PV? 5kW? Dann scheint das Verhalten bei anderen Anwendern anders zu sein...

Steph-74, 2024-12-17:
Hallo @andig ,
ich habe jetzt mal eine "maxchargerate: 10" wie folgt schnell getestet:

```  - type: template
    template: fronius-gen24
    usage: battery
    host: 192.168.10.999
    port: 502
    name: battery
    maxchargerate: 10
    capacity: 10.2
```
Funktioniert genau so, wie es funktionieren soll, die Batterie lädt mir ~1kW Leistung
[image]

Die "reading_*"-Werte sind die Modbus-Rohdaten für OutWRte (-1.000 = -10%) und InWRte (10.000 = 100%)

Gruß
Stephan

NFea-r, 2024-12-17:
Hallo @Steph-74 , Du hast aber im Homeassistant die InWRte noch zusätzlich gesetzt? Weil bei meiner config steht
```
- name: grid1
  type: template
  template: fronius-gen24
  usage: grid
  host: 192.168.1.18
  port: 502
  maxchargerate: 35
```
Einziger Unterschied scheint mir, dass ich die maxchargerate im grid hinterlegt habe (wie in der Doku beschrieben). Aber das dürfte nicht das Problem sein, wenn ich mir den Code anschaue. Mit dieser Config ist dann aber nicht bei 3,5 kW Schluss, sondern es geht Schrittweise hoch, bis 10kW vom Netz in die Batterie gehen.

Steph-74, 2024-12-17:
Hi @NFea-r,
vielleicht haben wir jetzt den Fehler, wenn Du schreibst: "wie in der Doku beschrieben". Ich habe mir eine neue config über "evcc configure" als "Batterie Wechselrichter" erstellt und diese dann in meine yaml kopiert. evcc configure schreibt die maxchargerate bei battery:
[image]

Ist das vielleicht die Doku falsch? Kannst Du den maxchargerate-Eintrag mal in battery verschieben?

Die Einstellungen im Screenshot von Home Assistent wurden alle über evcc gesetzt. Der InWRte wurde nicht angefasst, sprich war davor schon bei 100 %.

Gruß
Stephan

NFea-r, 2024-12-17:
@Steph-74 Super gesehen, das war es! maxchargerate unter battery eingetragen: Funktioniert! Dann ist die Doku dbzgl. nicht korrekt (und meine „Interpretation“ des Codes war dann wohl auch falsch). Danke für Eure Unterstützung.👍

andig, 2024-12-18:
https://github.com/evcc-io/evcc/pull/17784 sollte den Fehler künftig vermeiden.