evcc-io/evcc #17696: Tibber Integration homeassistant

0163deralex, 2024-12-12:
### Describe the bug

Nach Installation der neusten evcc Version 0.131.9 wird zur vollen Stunde die Tibber Integration in home assistant "rausgeworfen". Es werden dann keine Werte mehr in homeassistant angezeigt.
Der Strompreis wird aber weiterhin in evcc angezeigt.
Sobald auf die vorherige Version gedowngradet wird und das home assistant System neu gestartet wird, wird wieder alles angezeigt. 

### Steps to reproduce

1. Neuste Version installieren 
2. Auf volle Stunde warten 
3. Keine Werte mehr in homeassistant 
...

### Configuration details

```yaml
Keine Änderung zu bisher
```

### Log details

```text
Keine Daten
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

V0.131.9

andig, 2024-12-12:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-12-12 by andig

0163deralex, 2024-12-12:
Evcc funktioniert doch, da gibt's keine Fehler(logs).
Die Tibber Integration (das ist eine andere, die wie in evcc) wird dann von
Home Assistant nicht mehr erkannt. Und dieses log habe ich eingetragen.

mucki12, 2024-12-13:
Habe Tibber in evcc und HA eingebunden und kann den Fehler nicht nachstellen.