evcc-io/evcc #17699: Checking Teslamate config fails, but in general working

souhl, 2024-12-12:
### Describe the bug

When checking the teslamate config in the UI I get a "failed" and the following error:
`chargeStatus: strconv.ParseBool: parsing "16": invalid syntax`
Beside that everything seems to work fine.

### Steps to reproduce

1. Configure Teslamate
2. Configure Teslamate in EVCC UI
3. Store settings

[image]

...

### Configuration details

```yaml
#uri: 0.0.0.0:7070 # uri for ui
interval: 30s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: xxxx

# log settings
log: info
levels:
  site: info
  lp-1: info
  lp-2: info
  cache: info
  db: error
  mqtt: info
  proxy: info
  #tronity: debug
  opel: info
  tesla: info

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
#- name: grid1
#  type: template
#  template: solaredge-hybrid
#  usage: grid
  
#  modbus: tcpip
#  id: 1
#  host: 192.168.178.108 # Hostname
# port: 1502 # Port
#  timeout: 10s # optional
#  watchdog: 60s # optional
  
#- name: pv-dach
#  type: template
#  template: solaredge-hybrid
#  usage: pv
#  modbus: tcpip
#  id: 1
#  host: 192.168.178.108 # Hostname
#  port: 1502 # Port
#  timeout: 10s # optional
 
  
#- name: byd
#  type: template
#  template: solaredge-hybrid
#  usage: battery
[... 192 lines cut ...]
[lp-1  ] INFO 2024/12/12 14:15:22 loadpoint 1:
[lp-1  ] INFO 2024/12/12 14:15:22   mode:        pv
[lp-1  ] INFO 2024/12/12 14:15:22   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/12/12 14:15:22   meters:      charge ✓
[lp-1  ] INFO 2024/12/12 14:15:22     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/12/12 14:15:22 loadpoint 2:
[lp-2  ] INFO 2024/12/12 14:15:22   mode:        pv
[lp-2  ] INFO 2024/12/12 14:15:22   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-2  ] INFO 2024/12/12 14:15:22   meters:      charge ✓
[lp-2  ] INFO 2024/12/12 14:15:22     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2024/12/12 14:15:22 car disconnected
[lp-2  ] INFO 2024/12/12 14:15:53 car disconnected
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.9

andig, 2024-12-12:
/cc @Hofyyy 16 klingt nicht sehr boolish :O

andig, 2024-12-13:
Closed in https://github.com/evcc-io/evcc/commit/140858b16acf773aa3d3265bde0d80bfd0425a64

andig, 2024-12-18:
ping @Hofyyy

Hofyyy, 2024-12-18:
ich verstehe die frage nicht. Bei mir funktioniert es top. nutze aber auch kein Homeassist

andig, 2024-12-18:
Anscheinend gibts hier als "connected" Status eine 16. Das kann nicht als Bool verarbeitet werden. Wenns da keine Abhilfe gibt können wir aber auch zu machen. Im Zweifel bräuchte es wohl ein trace log für teslamate.

closed 2024-12-18 by andig

Hofyyy, 2024-12-18:
fair enough ich stelle aktuell noch alles direkt im yaml ein. könnte mal ein aufsetzten von Null per UI testen. würde ich allerdings ungern :-)

Beme99, 2025-01-01:
Ich habe 

> Anscheinend gibts hier als "connected" Status eine 16. Das kann nicht als Bool verarbeitet werden. Wenns da keine Abhilfe gibt können wir aber auch zu machen. Im Zweifel bräuchte es wohl ein trace log für teslamate.

Hallo und frohes neues Jahr allen!
Ich bekomme den gleichen Fehler wie @Hofyyy

Ich meine, dass die Template-Definition für Teslamate an der Stelle nicht passt, da für den Charging-Status das Mqqt-Topic "charger_actual_current" ausgelesen wird, in dem der Ladestrom definiert ist, der vom Fahrzeug genutzt wird.
https://github.com/evcc-io/evcc/blob/a558aa2d3ab9664b1ab00e2aa378430c0ddc4dfd/templates/definition/vehicle/teslamate.yaml#L32

[image]

Das richtige Topic von teslamate sollte m.E. "charging_state" sein

[image]

andig, 2025-01-01:
Gerne PR!

Hofyyy, 2025-01-01:
Nach deinen ganzen infos schlage ich vor, das wir state == charging nutzen, da dieser dokumentiert ist. ich gucke mir das am we mal an, wenn es bis dahin nicht schon wer anders eingevaut hat :-)

Beme99, 2025-01-01:
Habe mich mal dran versucht, nach meinen lokalen Tests funktioniert. 

Zur fehlenden Doku zum Topic charging_state habe ich auch einen PR in deren Projekt gestellt. Habe vorher getestet welche Status es annimmt: Charging, Disconnected, Stopped, NoPower, Complete

Hofyyy, 2025-01-03:
> Fix for issue #17699: Using correct MQTT topic for charging state #18005

Chatgpt sagt:
topic: "{{ if eq (teslamate/cars/{{ .id }}/state) 'charging' }}true{{ else }}false{{ end }}"

Kannst du das mal prüfen?

oder in schön:

topic: 
  value: teslamate/cars/{{ .id }}/state
  condition: "{{ eq .value 'charging' }}"
  
 Hab leider gerade keine Umgebung zum Testen.

jk, 2025-03-08:
Passiert hier noch was? Ich frage, weil das Issue als "closed" gilt. Ich habe heute von der bisherigen (jetzt abgeschalteten) Tesla-Integration auf Teslamate gewechselt und habe den gleichen Fehler, dass der Integer in charger_actual_current nicht als bool für Charing aufgefasst werden kann.

Seit dem ich gewechselt habe, klebt mir jetzt die Ladeleistung im PV-Modus auf 1,4 kW fest, obwohl es Netzeinspeisung gibt - hängt das hiermit zusammen?

Ich würde dann @Beme99 's bisherige Bemühungen wieder aufnehmen mit den Anmerkungen von @Hofyyy und einen neuen PR erstellen. Da schien ja leider die Motivation flöten gegangen zu sein.

andig, 2025-03-08:
Seit https://github.com/evcc-io/evcc/issues/17699#issuecomment-2566970557 scheint es keine neuen Erkenntnisse gegeben zu haben. Please feel free.

MarioSDR93, 2025-04-04:
Das Template nutzt charger_actual_current um den Charging state festzustellen. Die jq abfrage gibt true zurück, wenn der current größer 0 ist.