evcc-io/evcc #17704: UI Planner falsche Anzeige wenn Ziel nicht erreicht.

TobiasHuber1980, 2024-12-12:
@maschga

Planner heute auf 15:30 gestellt, Ladeziel konnte auf Grund zu wenig Zeit bis Zieluhrzeit nicht erreicht werden. UI spring dann auf "morgen" lädt aber trotzdem mit "schnell" weiter um das Ziel heute noch zu erreichen.

[Slack Message](https://evccgroup.slack.com/archives/C01321PUJAD/p1734015672328349?thread_ts=1733484744.273659&cid=C01321PUJAD)

andig, 2024-12-12:
Das wäre ja erstmal richtig- macht der Planner sonst auch. Er sollte in diesem Fall aber weiter „heute“ als Ziel zeigen?

Maschga, 2024-12-12:
Ja, genau. Ansonsten verwirrt das - meiner Meinung nach - den Benutzer.

andig, 2024-12-13:
@TobiasHuber1980 gibts ein Logfile? Ich vermute hier wird bei repeating plans der alte Plan bereits beendet was bei laufendem Plan nicht passieren dürfte.

naltatis, 2024-12-13:
@andig ist technisch ein etwas größeres Thema. Wir haben die Logik, dass laufende Pläne nicht abgebrochen werden sofern das Ziel noch nicht erreicht ist. Das ist das was hier auch zum richtigen Ladeverhalten führt. Für die Anzeige, aber auch um den Plan SoC auf den geladen wird, zu ermitteln betrachten wir momentan immer stumpf die aktuelle Uhrzeit und suchen dann den nächsten Plan raus. In dem Beispiel von @TobiasHuber1980 wird die Ladung nicht abgebrochen weil seine nächste Planzeit (morgen 15:30) auch 80% als Ziel hat. Wäre der nächste Plan unter dem aktuellen Ladestand gewesen hätte er abgebrochen. Heißt wir müssen an die Anzeige und ans Verhalten ran.

Ich bin mir noch nicht ganz sicher, was hier eine gute und nicht zu komplizierte technische Lösung ist. Vmtl. müssen wir die Infos zum aktiven Ladevorgang (aktiv wie "in progress"), also das Ladeziel und den Zeitpunkt noch mal als Kopie halten und erst beim "finish" wieder auf die Suche nach dem nächsten Ladeziel anhand der aktuellen Zeit switchen.

TobiasHuber1980, 2024-12-13:
Ja, solange der Ladevorgang des bereits "abgelaufenen" Planners noch läuft, sollte wie beim bisherigen Planner "die Restzeit/Ziel wann" in orange angezeit werden. Wäre mich mich logischer (so wie im Screenshot.)

Log:
```
[lp-2 ] DEBUG 2024/12/12 17:59:54 plan: charge 25m1s between 2024-12-13 17:00:00 +0100 CET until 2024-12-13 17:30:00 +0100 CET (power: 11040W, avg cost: 0.234)
[lp-2 ] DEBUG 2024/12/12 17:59:54 plan: continuing until end of slot
```

andig, 2024-12-13:
Wir müssen wohl das aktuelle Ziel mit publishen um die Logik aus dem UI rauszuhalten?

naltatis, 2024-12-13:
@andig Das aktuelle Ziel wird mWn. schon separat für die UI gepublisht. "Nur" die Logik muss angepasst werden.

andig, 2025-01-14:
Idee für einen Algorithmus siehe https://github.com/evcc-io/evcc/issues/18212#issuecomment-2589410309

jeffborg, 2025-03-26:
This could be improved. Any plan where the SOC is already at the target could be discarded for the purpose of considering the next plan

I have a repeating plan for 3pm every day to be 50% which was done by about 2pm due to cleanest slots it was done by about 1:30pm 
I also have a plan for 70% by 11am next day. I had to disable the repeating plan as the cleanest slots are up till 3pm and a final slot 10:30-11am.

andig, 2025-04-05:
@jeffborg are you saying that we will NOT charge the 11am plan (even if we should) since we still think the 3pm plan is active?

andig, 2025-04-05:
@TobiasHuber1980 Wir haben aktuell keine richtig gute Idee für eine bessere Heuristik. Daher wontfix. Das verbleibende Issue aus https://github.com/evcc-io/evcc/issues/17704#issuecomment-2780888118 wird per https://github.com/evcc-io/evcc/pull/20414 gelöst.

closed 2025-04-05 by andig

jeffborg, 2025-04-06:
@andig #20414 is a reasonable quick solution to the problem, just take the next plan where the soc hasn't been reached. In my [comment](https://github.com/evcc-io/evcc/issues/17704#issuecomment-2753152377) this would start the 70% plan as soon as 50% was reached and would continue to 3pm