evcc-io/evcc #17705: UI - kleine Verbesserungen für "neue Version" und dynamische Tarife

vkr2791, 2024-12-12:
**Is your feature request related to a problem? Please describe.**
Nur eine 2 kleine Anregungen ...

**Describe the solution you'd like**
Aktuell bekommt man immer nur "neue Version verfügbar" angezeigt, besser wäre hier noch die letzte verfügbare Version mit anzuzeigen, manche Versionen sind für einen erstmal gar nicht so relevant, dann will man nicht direkt updaten, aber dann bekommt man nicht im UI mit, dass es ggf. schon weitere neuere Versionen gibt, daher wäre die Anzeige der letzten verfügbaren Version sinnvoll. 

Ein anderer Punkt ist der, dass man bei dynamischen Stromtarifen immer nur den aktuellen Preis bei "Netzbezug" angezeigt bekommt.
Aktuell klicke ich immer auf die Einstellungen des Loadpoints um zu sehen, wie sich der Strompreis in den nächsten Stunden entwickelt oder man nutzt die Tibber-App. Aber denkbar wäre doch einen Link bei Netzbezug einzubauen mit einem eigenen neuen Sub-Fenster der den Strompreis detailierter anzeigt, auch direkt mit den Preisen pro Stunden, ohne das man einzelne Stunden anklicken muss.

**Describe alternatives you've considered**
Es wäre auch denkbar den dynamischen Stromtarif irgendwie immer direkt darzustelllen, das würde aber extra Platz im UI kosten 
ODER man würde es zusätzlich in den Balken des aktuellen Stromverbrauchs (ganz oben) integrieren, aber dann kann man keine konkreten Werte einbauen sondern nur eine Trendlinie einzeichnen.
Hier mal der Versuch einer (stümperhaften) Darstellung :-) :
[image]

[image]

Vielleicht so eine detailierte oder ähnliche Darstellung als Link hinter "Netzbezug":
[image]

andig, 2024-12-12:
> Aktuell klicke ich immer auf die Einstellungen des Loadpoints um zu sehen, wie sich der Strompreis in den nächsten Stunden entwickelt oder man nutzt die Tibber-App. Aber denkbar wäre doch einen Link [...]

Warum interessiert dich das wenn ein Ladeplan schon „billigst“ garantiert?

naltatis, 2024-12-13:
Hi @vkr2791 erstmal danke für deine Anregungen. Du hast hier allerdings viele Wünsche und Lösungsvorschläge zusammen in eins gepackt. So etwas macht es immer schwierig zu diskutieren. Ich probier jetzt trotzdem mal kurz auf die unterschiedlichen Punkte einzugehen:

- **Neue Versionsnummer anzeigen:** Ich würde hier sogar eher in die Gegenrichtung argumentieren. Die konkrete Versionsnummer hat für die allermeisten Nutzer keine große Bedeutung. Momentan zeigen wir die immer an, weil wir den Platz dafür vorgesehen haben. Im Alltag wäre es aber sicher auch kein Problem, wenn man die Versionsnummer und detailliertere Infos über mögliche Updates und deren neuen Funktionen über ein Menü bekommt und wir den Platz im Gegenzug für etwas anderes verwenden.

- **Liniendiagramm im Balkendiagram:** Das sehe ich aktuell nicht. Das würde die jetzige sehr aufgeräumte Ansicht unruhiger machen und, wie du schon sagst, direkt den Wunsch nach Achsenbeschriftung nachziehen :D. Dieses Element sollten wir nicht überladen.

> Aktuell klicke ich immer auf die Einstellungen des Loadpoints um zu sehen, wie sich der Strompreis in den nächsten Stunden entwickelt oder man nutzt die Tibber-App. 

Das kann ich gut verstehen und geht mir genau so. Da ist die aktuelle Lösung noch nicht optimal. Allerdings sehe ich hier keinen einfachen Fix. Was würdest du denn erwarten was sich öffnet, wenn du auf den Preis, wie in deinem Screenshot, klickst? Würdest du hier einen neuen Dialog sehen in dem man nur die Preisentwicklung angezeigt bekommt und sonst nichts machen kann? Das fühlt sich für mich noch zu unrund und dünn an.

Aber ich bin bei dir, dass wir hier noch mal grundsätzlicher drüber nachdenken müssen. Ein Feature, das noch auf unserer Roadmap steht ist die Einbindung von PV-Prognosen. Diese Daten würden sich gut mit den Preisvorhersagen vertragen. Das rechtfertigt dann ggf. auch einen eigenen Menüpunkt/Reiter. Bis dahin würde ich erst einmal bei der aktuellen Lösung bleiben.

Ich mach diesen Issue jetzt erst einmal zu, nehme aber das Thema "Preisprognose besser zugänglich machen" gedanklich mit, wenn wir uns mit PV-Prognose beschäftigen.

closed 2024-12-13 by naltatis

vkr2791, 2024-12-13:
@andig : Da hast du recht, der Ladeplan macht alles super.
Aber die Waschmaschine muss ich trotzdem noch manuell planen ;-)
und evcc ist in der Darstellung meiner Haus-Eckdaten so gut (PV-Analage, Haus-Batterie, aktueller Strombezug, ...),
dass ich mir immer über evcc ein Bild der aktuellen Lage verschaffe
und dieser kleiner Baustein wäre noch ein kleiner Komfortgewinn und würde sich gut einpassen in der Gesamtdarstellung.

Ist ja auch nur eine Anregung mit geringer Prio ;-)

@naltatis : 
Mit dem Liniendiagramm an der Stelle gebe ich dir Recht, das würde für mehr Unruhe sorgen, ich fände die Variante über den Link auf dem "Netzbezug" auch besser. 

> Was würdest du denn erwarten was sich öffnet, wenn du auf den Preis, wie in deinem Screenshot, klickst? Würdest du hier einen neuen Dialog sehen in dem man nur die Preisentwicklung angezeigt bekommt und sonst nichts machen kann? Das fühlt sich für mich noch zu unrund und dünn an.

Ich würde hier gar nichts erwarten, reine Information und da man an dem "Netzbezug" auch selbst nichts konfigurieren/beeinflussen kann (gibt ja Tibber vor), wäre für mich ein einfaches Info-Fenster auch vollkommen OK und würde der Erwartung des End-Users entsprechen.

> Einbindung von PV-Prognosen

Wenn dieser Punkt kommt, dann würde ich den nicht unter "Netzbezug" sehen sondern unter "Erzeugung", weil es wäre ja ein Forecast für die PV-Erzeugung.

naltatis, 2024-12-13:
> Wenn dieser Punkt kommt, dann würde ich den nicht unter "Netzbezug" sehen sondern unter "Erzeugung", weil es wäre ja ein Forecast für die PV-Erzeugung.

Ich denke für die PV-Prognose eher über eine separate Ansicht (Dialog o. eigene Seite) nach.