evcc-io/evcc #17706: Tibber geht nicht mehr seit Version 0.131.9

chrostek, 2024-12-13:
### Describe the bug

Seit dem Update auf [0.131.9](https://github.com/evcc-io/evcc/releases/tag/0.131.9) geht Tibber bei mir nicht mehr.
evcc startet in einer Endlosschleife (im 15 Minuten Takt) neu mit der Fehlermeldung:

> cannot create meter 'grid1': cannot create meter type 'template': cannot create meter type 'tibber-pulse': Message: 429 Too Many Requests, Locations: [], Extensions: map[code:request_error], Path: []

Log:
[49686a9f_evcc_2024-12-13T05-11-38.329Z.log](https://github.com/user-attachments/files/18121591/49686a9f_evcc_2024-12-13T05-11-38.329Z.log)

Da ich es neuerdings als HA Addon verwende, kann ich leider auch kein Downgrade machen 🙈

### Steps to reproduce

1. evcc starten mit konfiguriertem Tibber-Pulse Meter

### Configuration details

```yaml
meters:
- type: template
  template: tibber-pulse
  usage: grid
  token: <token>
  homeid: <homeid>
  name: grid1

tariffs:
  grid:
    type: tibber
    token: <token>
    homeid: <homeid>
```

### Log details

```text
[site  ] ERROR 2024/12/13 06:01:21 grid power: outdated
[site  ] ERROR 2024/12/13 06:01:51 grid power: outdated
[site  ] ERROR 2024/12/13 06:02:21 grid power: outdated
[site  ] ERROR 2024/12/13 06:02:51 grid power: outdated
[site  ] ERROR 2024/12/13 06:03:22 grid power: outdated
[site  ] ERROR 2024/12/13 06:03:52 grid power: outdated
[site  ] ERROR 2024/12/13 06:04:21 grid power: outdated
[site  ] ERROR 2024/12/13 06:04:51 grid power: outdated
[eebus ] ERROR 2024/12/13 06:04:53 websocket server error: http: Server closed

[...]

[main  ] FATAL 2024/12/13 06:04:54 cannot create meter 'grid1': cannot create meter type 'template': cannot create meter type 'tibber-pulse': Message: 429 Too Many Requests, Locations: [], Extensions: map[code:request_error], Path: []
[main  ] FATAL 2024/12/13 06:04:54 will attempt restart in: 15m0s
[eebus ] ERROR 2024/12/13 06:07:55 websocket server error: http: Server closed
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.9

andig, 2024-12-13:
Bitte Zugangsdaten an info@evcc.io

andig, 2024-12-13:
Closed in https://github.com/evcc-io/evcc/commit/140858b16acf773aa3d3265bde0d80bfd0425a64

closed 2024-12-13 by andig

phrozen77, 2024-12-13:
Weil ich die Fehlermeldung heute auch schon hatte (Neustart über die GUI hilft dann) habe ich, neugierdehalber, eben auf die 0.131.11 aktualisiert.

Was soll ich sagen - selber Fehler.

Da aktuell sowohl evcc, HomeAssistant und sogar die offizielle Tibber-App "unrund" laufen (Tibber-App hatte vorhin anstelle eines Messwerts einen durchgestrichenen Pulse angezeigt - daraufhin hab ich die Bridge mal aus- und wieder eingesteckt und beim Pulse mal kurz die Batterien entnommen) behaupte ich mal, dass das Problem eher auf Seiten Tibber zu finden ist.

Aktuell zeigt die App zwar einen Messwert an, aber anstatt dass der sich, wie üblich, innerhalb von 2-6 Sekunden aktualisiert ist das eher gerade eine Standuhr.

Unschön ist halt, das man sich aufgrund externer Probleme bzw. fehlender externer Konnektivität eine weitestgehend lokal agierende Software lahmlegen kann. 

Nach dem manuelle Neustart über die GUI habe ich übrigens wieder die Preise sichtbar - HA und auch die App sind immer noch außer Gefecht.

bastian1607, 2024-12-13:
Hi @phrozen77 ,
das Problem ist, dass das wohl bei Tibber liegt. Sobald meine Tibber-Integration stockt, erhalte ich auch einen Fehler unter developer.tibber.com mit "Network error". 
Ich hatte das heute morgen Tibber übern Chat gemeldet, aber mal sehen wie da die Response-Zeiten sind. Vorher hatte ich hier auch alles abgeklappert von Internet, APs, Integration neu installiert etc. pp. Auf der Statusseite von Tibber gibt es bisher keinen Eintrag.
Gut zu wissen, dass das scheinbar ein globales Problem ist.

phrozen77, 2024-12-13:
Kennt man ja von Tibber auch kaum anders 🤷‍♂️ 

Ist am Ende des Tages auch relativ egal, aber

> Unschön ist halt, das man sich aufgrund externer Probleme bzw. fehlender externer Konnektivität eine weitestgehend lokal agierende Software lahmlegen kann.

Und genau das ist mit 0.131.11 nicht behoben.

phrozen77, 2024-12-13:
...und jetzt, auch wenn das Prozedere heute schon mindestens 2x funktioniert hat, hilft auch der Neustart über die GUI (Konfiguration -> System -> Neu starten) nicht mehr - es bleibt beim 429'er:
```
Fehler beim Starten. Überprüfe deine Konfiguration und starte neu.
cannot create meter 'tibber': cannot create meter type 'template': cannot create meter type 'tibber-pulse': Message: 429 Too Many Requests, Locations: [], Extensions: map[code:request_error], Path: []
```

chrostek, 2024-12-13:
https://github.com/evcc-io/evcc/releases/tag/0.131.11
sieht bei mir gut aus damit. Danke!

bastian1607, 2024-12-13:
Hi @phrozen77 ,
Ich verstehe deinen Punkt durchaus, ähnliches Problem hatte ich mit Senec, weil lokal immer abgeschmiert ist. Ich hatte den Speicher dann per MQTT aus der Web-API angebunden.
Theoretisch könntest du ähnliches mit Tibber machen, der Wert wird dann aktualisiert mit dem aktuellsten aus der Integration bzw. baust dir einen Sensor extra und schickst den Wert über MQTT. Meine Sensoren, die per Tibber Web-API angebunden sind, haben gerade weniger Ausfall als die Ingetration selbst, weil sie selbst einfach wieder pollen.

Kurzum: Workaround einen eigenen Sensor erstellen und über Node-Red befüllen. Dann über MQTT dann den aktuellsten Wert an EVCC schicken lassen.