evcc-io/evcc #17712: vehicle soc: modbus: exception '4' (server device failure), function '3' bei KEBA P30 Wallbox

maxschloegl, 2024-12-13:
### Describe the bug

Ich habe 2 KEBA P30 Wallboxen und immer wieder kommt dieser Fehler in der GUI:
`vehicle soc: modbus: exception '4' (server device failure), function '3' `

### Steps to reproduce

nachdem ich evcc starte kommt dieser Fehler.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

#log: debug
log: trace
#levels:
#  cache: error
#  site: trace
#  modbus: trace

modbusproxy:
    # KEBA P30 Carport
  - port: 5021
    uri: 10.9.192.41:502
    readonly: false
    # KEBA P30 Garage
  - port: 5022
    uri: 10.9.192.44:502
    readonly: false
    # Sungrow Wechselrichter
  - port: 5023
    uri: 10.9.192.32:502
    readonly: false

# unique installation id
plant: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

sponsortoken: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

site:
  title: xxxxxxxxxxxxxxxxxxxxxxx
  residualPower: 100
  meters:
    grid: grid-sungrow-114
    pv:
      - pv-sungrow-114
    battery:
      - battery-sungrow-114

tariffs:
  grid:
    type: fixed
    price: 0.2491
  feedin:
    type: fixed
    price: 0.11

interval: 30s # control cycle interval

meters:
  - type: template
    template: sungrow-hybrid
    id: 1
    host: 127.0.0.1
    port: 5023
    usage: grid
    modbus: tcpip
    name: grid-sungrow-114
  - type: template
    template: sungrow-hybrid
    id: 1
    host: 127.0.0.1
    port: 5023
    usage: pv
    modbus: tcpip
    name: pv-sungrow-114
  - type: template
    template: sungrow-hybrid
    id: 1
    host: 127.0.0.1
    port: 5023
    usage: battery
    modbus: tcpip
    name: battery-sungrow-114
  - name: tasmota-heizstab
    type: custom
    power:
      source: mqtt
      broker: 10.90.3.21:1883
      topic: tele/esp/stg114/heizraum/puffer-heizstab/steckdose/SENSOR
      scale: 1
      jq: .ENERGY.Power
      timeout: 30s
    energy:  
      source: mqtt
      broker: 10.90.3.21:1883
      topic: tele/esp/stg114/heizraum/puffer-heizstab/steckdose/SENSOR
      scale: 1
      jq: .ENERGY.Current
      timeout: 30s

chargers: 
  - type: template
    template: keba-modbus
    name: wallbox-carport
    modbus: tcpip
    id: 255
    host: 127.0.0.1
    port: 5021
  - type: template
    template: keba-modbus
    name: wallbox-garage
    modbus: tcpip
    id: 255
    host: 127.0.0.1
    port: 5022
  - name: heizstab
    type: custom
    features:
      - integrateddevice
      - heating
    icon: waterheater
    status: # charger status A..F
      source: mqtt
      broker: 10.90.3.21:1883
      topic: tele/evcc/stg114/heizraum/puffer/heizstab/chargestatus
    enabled: # charger enabled state (true/false or 0/1)
      source: mqtt
      broker: 10.90.3.21:1883
      topic: tele/evcc/stg114/heizraum/puffer/heizstab/enabled
    enable: # set charger enabled true (true/false or 0/1)
      source: mqtt
      broker: 10.90.3.21:1883
      topic: cmd/evcc/stg114/heizraum/puffer/heizstab/enabled
      payload: ${enable:%d}
    maxcurrent: # set charger max current (A)
      source: mqtt
      broker: 10.90.3.21:1883
      topic: cmd/evcc/stg114/heizraum/puffer/heizstab/current
      payload: ${maxcurrent}
    

vehicles:
  - type: template
    template: bmw
    title: BMW i4 M50
    icon: car
    user: xxxxxxxxxxxxx
    password: xxxxxxxxxxxxxxx
    capacity: 80.7
    phases: 3
    cache: 15m
    mode: now
    minCurrent: 6
    maxCurrent: 16
    region: EU
    name: ev-bmw-i4-max
    hcaptcha: "xxxxxxxxxxxxxxxxxxxxxxxxx"
  - type: template
    template: audi
    title: AUDI A8 Hybrid
    icon: car
    user: xxxxxxxxxxxxxx
    password: xxxxxxxxxxxxxxxxxxx
    capacity: 19.7
    phases: 3
    cache: 15m
    mode: now
    minCurrent: 6
    maxCurrent: 16
    #region: EU
    name: ev-audi-a8-michael
  - type: custom
    title: Heizstab Puffer
    name: heizstab-puffer
    capacity: 80
    phases: 1
    soc: 
      source: mqtt
      broker: 10.90.3.21:1883
      topic: tele/evcc/stg114/heizraum/puffer/heizstab/puffer-temperatur

loadpoints:
  - title: Carport
    charger: wallbox-carport
    mode: now
    priority: 5
  - title: Garage
    charger: wallbox-garage
    mode: now
    priority: 5
  - title: Puffer
    charger: heizstab
    mode: pv
    meter: tasmota-heizstab
    vehicle: heizstab-puffer
    soc:
      estimate: false
```

### Log details

```text
[audi ] TRACE 2024/12/13 10:12:49 {"charging":{"batteryStatus":{"value":{"carCapturedTimestamp":"2024-12-12T09:00:18Z","currentSOC_pct":100,"cruisingRangeElectric_km":51}},"chargingStatus":{"value":{"carCapturedTimestamp":"2024-12-12T09:00:18Z","remainingChargingTimeToComplete_min":0,"chargingState":"chargePurposeReachedAndNotConservationCharging","chargeMode":"manual","chargeType":"unsupported"}},"plugStatus":{"value":{"carCapturedTimestamp":"2024-12-12T09:00:18Z","plugConnectionState":"connected","plugLockState":"locked","externalPower":"ready","ledColor":"green"}},"chargeMode":{"value":{"preferredChargeMode":"invalid","availableChargeModes":[]}}},"climatisation":{"climatisationSettings":{"error":{"message":"Bad Gateway","errorTimeStamp":"2024-12-13T09:12:47Z","info":"Upstream service responded with an unexpected status. If the problem persists, please contact our support.","code":4002,"group":2,"retry":true}},"climatisationStatus":{"error":{"message":"Bad Gateway","errorTimeStamp":"2024-12-13T09:12:47Z","info":"Upstream service responded with an unexpected status. If the problem persists, please contact our support.","code":4105,"group":2,"retry":true}},"climatisationTemperatureOutside":{"error":{"message":"Bad Gateway","errorTimeStamp":"2024-12-13T09:12:47Z","info":"Upstream service responded with an unexpected status. If the problem persists, please contact our support.","code":4002,"group":2,"retry":true}},"windowHeatingStatus":{"error":{"message":"Bad Gateway","errorTimeStamp":"2024-12-13T09:12:47Z","info":"Upstream service responded with an unexpected status. If the problem persists, please contact our support.","code":4105,"group":2,"retry":true}}},"fuelStatus":{"rangeStatus":{"value":{"carCapturedTimestamp":"2024-12-12T09:00:18Z","carType":"hybrid","primaryEngine":{"type":"gasoline","currentSOC_pct":70,"remainingRange_km":540,"currentFuelLevel_pct":70},"secondaryEngine":{"type":"electric","currentSOC_pct":100,"remainingRange_km":51},"totalRange_km":591}}},"measurements":{"rangeStatus":{"value":{"carCapturedTimestamp":"2024-12-12T09:00:18Z","gasolineRange":540,"totalRange_km":540}},"odometerStatus":{"value":{"carCapturedTimestamp":"2024-12-12T07:24:24Z","odometer":7776}},"fuelLevelStatus":{"value":{"carCapturedTimestamp":"2024-12-12T09:00:18Z","currentFuelLevel_pct":70,"currentSOC_pct":100,"primaryEngineType":"gasoline","secondaryEngineType":"electric","carType":"hybrid"}}}}
[main ] DEBUG 2024/12/13 10:12:49 vehicle status: B (AUDI A8 Hybrid (Michael))
[lp-2 ] INFO 2024/12/13 10:12:49 vehicle updated: unknown -> AUDI A8 Hybrid (Michael)
[lp-2 ] DEBUG 2024/12/13 10:12:49 set charge mode: now
[keba ] TRACE 2024/12/13 10:12:49 modbus: send 00 14 00 00 00 06 ff 03 03 ec 00 02
[proxy-5022] TRACE 2024/12/13 10:12:49 read holdings: id 255 addr 1004 qty 2
[proxy-5022] TRACE 2024/12/13 10:12:49 read holding: write tcp 172.18.0.8:52360->10.9.192.44:502: write: connection reset by peer
[keba ] TRACE 2024/12/13 10:12:49 modbus: recv 00 14 00 00 00 03 ff 83 04
[lp-2 ] ERROR 2024/12/13 10:12:49 vehicle soc: modbus: exception '4' (server device failure), function '3'
[keba ] TRACE 2024/12/13 10:12:49 modbus: send 00 15 00 00 00 06 ff 03 03 e8 00 02
[proxy-5022] TRACE 2024/12/13 10:12:49 read holdings: id 255 addr 1000 qty 2
[proxy-5022] TRACE 2024/12/13 10:12:49 read holding: 00000002
[keba ] TRACE 2024/12/13 10:12:49 modbus: recv 00 15 00 00 00 07 ff 03 04 00 00 00 02
[keba ] TRACE 2024/12/13 10:12:49 modbus: send 00 16 00 00 00 06 ff 06 13 bc 00 01
[proxy-5022] TRACE 2024/12/13 10:12:49 write holding: id 255 addr 5052 val 0001
[proxy-5022] TRACE 2024/12/13 10:12:49 write holding: 0001
[keba ] TRACE 2024/12/13 10:12:49 modbus: recv 00 16 00 00 00 06 ff 06 13 bc 00 01
[lp-2 ] DEBUG 2024/12/13 10:12:49 switched phases: 3p
[keba ] TRACE 2024/12/13 10:12:49 modbus: send 00 17 00 00 00 06 ff 06 13 8c 3e 80
[proxy-5022] TRACE 2024/12/13 10:12:49 write holding: id 255 addr 5004 val 3e80
[proxy-5022] TRACE 2024/12/13 10:12:49 write holding: 3e80
[keba ] TRACE 2024/12/13 10:12:49 modbus: recv 00 17 00 00 00 06 ff 06 13 8c 3e 80
[lp-2 ] DEBUG 2024/12/13 10:12:49 max charge current: 16A
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.11 + 0.131.10 (51075e29)

andig, 2024-12-13:
Gibts den Fehler ohne Modbus Proxy auch? Wofür brauchst du den Proxy? Tritt der Fehler auch auf wenn du externe Zugriffe (und Proxy!) deaktivierst?

maxschloegl, 2024-12-13:
Ohne Modbus Proxy tritt ein anderer Fehler auf. (**[lp-2 ] WARN 2024/12/13 14:55:58 charger out of sync: expected disabled, got enabled**)
Ich benötige den Proxy, da ich auch von anderen Slaves darauf zugreife. Diese habe ich jetzt für den Test deaktiviert.

```
[audi ] TRACE 2024/12/13 14:55:58 {"charging":{"batteryStatus":{"value":{"carCapturedTimestamp":"2024-12-13T13:55:50Z","currentSOC_pct":96,"cruisingRangeElectric_km":47}},"chargingStatus":{"value":{"carCapturedTimestamp":"2024-12-13T13:55:50Z","chargingState":"readyForCharging","chargeMode":"manual","chargeType":"unsupported"}},"plugStatus":{"value":{"carCapturedTimestamp":"2024-12-13T13:55:50Z","plugConnectionState":"connected","plugLockState":"locked","externalPower":"ready","ledColor":"none"}},"chargeMode":{"value":{"preferredChargeMode":"manual","availableChargeModes":["manual","timer"]}}},"climatisation":{"climatisationSettings":{"value":{"carCapturedTimestamp":"2024-12-13T13:54:29Z","targetTemperature_C":22,"targetTemperature_F":72,"climatisationWithoutExternalPower":true,"climatizationAtUnlock":false,"windowHeatingEnabled":false,"zoneFrontLeftEnabled":true,"zoneFrontRightEnabled":true,"zoneRearLeftEnabled":true,"zoneRearRightEnabled":true}},"climatisationStatus":{"value":{"carCapturedTimestamp":"2024-12-13T13:54:39Z","remainingClimatisationTime_min":-256,"climatisationState":"off"}},"climatisationTemperatureOutside":{"error":{"message":"Not Found","errorTimeStamp":"2024-12-13T13:55:57Z","info":"Upstream service responded with an unexpected status. If the problem persists, please contact our support.","code":4103,"group":2,"retry":true}},"windowHeatingStatus":{"value":{"windowHeatingStatus":[{"windowLocation":"front","windowHeatingState":"invalid"},{"windowLocation":"rear","windowHeatingState":"invalid"}]}}},"fuelStatus":{"rangeStatus":{"value":{"carCapturedTimestamp":"2024-12-13T13:55:54Z","carType":"hybrid","primaryEngine":{"type":"gasoline","currentSOC_pct":70,"remainingRange_km":540,"currentFuelLevel_pct":70},"secondaryEngine":{"type":"electric","currentSOC_pct":96,"remainingRange_km":47},"totalRange_km":587}}},"measurements":{"rangeStatus":{"value":{"carCapturedTimestamp":"2024-12-13T13:55:54Z","electricRange":47,"gasolineRange":540,"totalRange_km":587}},"odometerStatus":{"value":{"carCapturedTimestamp":"2024-12-13T13:55:32Z","odometer":7777}},"fuelLevelStatus":{"value":{"carCapturedTimestamp":"2024-12-13T13:55:54Z","currentFuelLevel_pct":70,"currentSOC_pct":96,"primaryEngineType":"gasoline","secondaryEngineType":"electric","carType":"hybrid"}}}}
[main ] DEBUG 2024/12/13 14:55:58 vehicle status: B (AUDI A8 Hybrid (Michael))
[lp-2 ] INFO 2024/12/13 14:55:58 vehicle updated: unknown -> AUDI A8 Hybrid (Michael)
[lp-2 ] DEBUG 2024/12/13 14:55:58 set charge mode: now
[db ] TRACE 2024/12/13 14:55:58 UPDATE `sessions` SET `created`="2024-12-13 14:55:53.578",`finished`="0000-00-00 00:00:00",`loadpoint`="Garage",`identifier`="",`vehicle`="AUDI A8 Hybrid (Michael)",`odometer`=NULL,`meter_start_kwh`=1291.2183,`meter_end_kwh`=NULL,`charged_kwh`=0,`charge_duration`=NULL,`solar_percentage`=NULL,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 360 1 <nil>
[keba ] TRACE 2024/12/13 14:55:58 modbus: send 00 81 00 00 00 06 ff 03 03 ec 00 02
[keba ] TRACE 2024/12/13 14:55:58 modbus: recv 00 81 00 00 00 07 ff 03 04 00 00 00 07
[keba ] TRACE 2024/12/13 14:55:58 modbus: send 00 82 00 00 00 06 ff 03 03 e8 00 02
[keba ] TRACE 2024/12/13 14:55:58 modbus: recv 00 82 00 00 00 07 ff 03 04 00 00 00 03
[lp-2 ] DEBUG 2024/12/13 14:55:58 vehicle status: B
[lp-2 ] DEBUG 2024/12/13 14:55:58 vehicle soc: 96%
[lp-2 ] DEBUG 2024/12/13 14:55:58 vehicle range: 47km
[keba ] TRACE 2024/12/13 14:55:58 modbus: send 00 83 00 00 00 06 ff 03 03 e8 00 02
[keba ] TRACE 2024/12/13 14:55:58 modbus: recv 00 83 00 00 00 07 ff 03 04 00 00 00 03
[lp-2 ] WARN 2024/12/13 14:55:58 charger out of sync: expected disabled, got enabled
[lp-2 ] DEBUG 2024/12/13 14:55:58 charger enable
[keba ] TRACE 2024/12/13 14:55:58 modbus: send 00 84 00 00 00 06 ff 06 13 bc 00 01
[keba ] TRACE 2024/12/13 14:55:58 modbus: recv 00 84 00 00 00 06 ff 06 13 bc 00 01
[lp-2 ] DEBUG 2024/12/13 14:55:58 switched phases: 3p
[keba ] TRACE 2024/12/13 14:55:58 modbus: send 00 85 00 00 00 06 ff 06 13 8c 3e 80
[keba ] TRACE 2024/12/13 14:55:58 modbus: recv 00 85 00 00 00 06 ff 06 13 8c 3e 80
[lp-2 ] DEBUG 2024/12/13 14:55:58 max charge current: 16A 
```

An einer anderen Stelle tritt auch folgender Fehler auf:
**[site ] ERROR 2024/12/13 14:56:53 battery mode: write tcp 172.18.0.7:59928->10.9.192.32:502: write: broken pipe**

```
[lp-2 ] DEBUG 2024/12/13 14:56:53 charge power: 6323W
[keba ] TRACE 2024/12/13 14:56:53 modbus: send 00 8d 00 00 00 06 ff 03 03 f0 00 02
[keba ] TRACE 2024/12/13 14:56:53 modbus: recv 00 8d 00 00 00 07 ff 03 04 00 00 39 03
[keba ] TRACE 2024/12/13 14:56:53 modbus: send 00 8e 00 00 00 06 ff 03 03 f2 00 02
[keba ] TRACE 2024/12/13 14:56:53 modbus: recv 00 8e 00 00 00 07 ff 03 04 00 00 38 78
[keba ] TRACE 2024/12/13 14:56:53 modbus: send 00 8f 00 00 00 06 ff 03 03 f4 00 02
[keba ] TRACE 2024/12/13 14:56:53 modbus: recv 00 8f 00 00 00 07 ff 03 04 00 00 00 02
[lp-2 ] DEBUG 2024/12/13 14:56:53 charge currents: [14.6 14.5 0.002]A
[battery-sungrow-114] TRACE 2024/12/13 14:56:53 modbus: send 01 50 00 00 00 06 01 06 32 f9 00 02
[site ] ERROR 2024/12/13 14:56:53 battery mode: write tcp 172.18.0.7:59928->10.9.192.32:502: write: broken pipe
[... cut ...]
[grid-sungrow-114] TRACE 2024/12/13 14:56:53 modbus: send 01 55 00 00 00 06 01 04 13 9a 00 01
[grid-sungrow-114] TRACE 2024/12/13 14:56:53 modbus: recv 01 55 00 00 00 05 01 04 02 08 c7
[pv-sungrow-114] TRACE 2024/12/13 14:56:53 modbus: send 01 56 00 00 00 06 01 04 32 ca 00 02
[pv-sungrow-114] TRACE 2024/12/13 14:56:53 modbus: recv 01 56 00 00 00 07 01 04 04 be 96 00 02
[site ] DEBUG 2024/12/13 14:56:53 pv power: 96W 
```

**Mein Equipment ist folgendes:**
- Wallboxen: 2x KEBA P30
- Wechselrichter: Sungrow SH10RT-V112 (Modbus TCP direkt angeschlossen)
- Batterie: Sungrow SBR192

maxschloegl, 2024-12-13:
PS: Die IP beim zweiten Fehler (10.9.192.32) ist die des Wechselrichters.

andig, 2024-12-13:
> Ich benötige den Proxy, da ich auch von anderen Slaves darauf zugreife. 

Wenn der server device failure ohne die anderen Slaves nicht auftaucht ist er dadurch bedingt- das ist dann kein evcc-Thema.

maxschloegl, 2024-12-13:
> Wenn der server device failure ohne die anderen Slaves nicht auftaucht ist er dadurch bedingt- das ist dann kein evcc-Thema.

Ich verstehe dich nicht ganz. Der Proxy dient ja genau dem Zweck, dass dann mehrere Devices zugreifen können, oder?
Der Fehler ist ja deshalb verschwunden, nachdem ich den Proxy deaktiviert habe. 
Ich habe jetzt den Proxy wieder aktiviert und es greifen keine Slaves außer evcc darauf zu. Habe alles andere deaktiviert.

Ähnlicher device failure in der letzten Zeile:
```
[grid-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: send 00 10 00 00 00 06 01 04 32 d1 00 02
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: id 1 addr 13009 qty 2
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: write tcp 172.18.0.5:55934->10.9.192.32:502: write: broken pipe
[grid-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: recv 00 10 00 00 00 03 01 84 04
[pv-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: send 00 11 00 00 00 06 01 04 13 98 00 02
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: id 1 addr 5016 qty 2
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: 00000000
[pv-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: recv 00 11 00 00 00 07 01 04 04 00 00 00 00
[pv-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: send 00 12 00 00 00 06 01 04 32 ca 00 02
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: id 1 addr 13002 qty 2
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: be970002
[pv-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: recv 00 12 00 00 00 07 01 04 04 be 97 00 02
[site ] DEBUG 2024/12/13 16:43:25 pv power: 0W
[battery-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: send 00 13 00 00 00 06 01 04 32 dd 00 01
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: id 1 addr 13021 qty 1
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: 0000
[battery-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: recv 00 13 00 00 00 05 01 04 02 00 00
[battery-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: send 00 14 00 00 00 06 01 04 32 c8 00 01
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: id 1 addr 13000 qty 1
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: 0028
[battery-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: recv 00 14 00 00 00 05 01 04 02 00 28
[battery-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: send 00 15 00 00 00 06 01 04 32 c8 00 01
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: id 1 addr 13000 qty 1
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: 0028
[battery-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: recv 00 15 00 00 00 05 01 04 02 00 28
[battery-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: send 00 16 00 00 00 06 01 04 32 e2 00 02
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: id 1 addr 13026 qty 2
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: 998d0000
[battery-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: recv 00 16 00 00 00 07 01 04 04 99 8d 00 00
[battery-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: send 00 17 00 00 00 06 01 04 32 de 00 01
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: id 1 addr 13022 qty 1
[proxy-5023] TRACE 2024/12/13 16:43:25 read input: 0000
[battery-sungrow-114] TRACE 2024/12/13 16:43:25 modbus: recv 00 17 00 00 00 05 01 04 02 00 00
[site ] DEBUG 2024/12/13 16:43:25 battery soc: 0%
[site ] DEBUG 2024/12/13 16:43:25 battery power: 0W
[site ] ERROR 2024/12/13 16:43:25 grid power: read failed: modbus: exception '4' (server device failure), function '4'
```

> Bzgl. Sungrow broken pipe siehe #17069.

Könntest du mir auf die Sprünge helfen in wie weit mir das helfen kann?

> Bleibt die Frage nach charger out of sync. Dafür reicht das kurze Log nicht aus. Auch dabei muss sicher gestellt sein, dass nur ein Client (=evcc) zugreift.

Welche Log-Informationen soll ich zur Verfügung stellen?

lg Max

maxschloegl, 2024-12-13:
PS: Ab dem Zeitpunkt wo ich über den modbus proxy auf die Devices zugreifen häufen sich die Fehler:
```
[site ] ERROR 2024/12/13 16:55:21 grid power: read failed: read tcp 127.0.0.1:41488->127.0.0.1:5023: i/o timeout
[site ] ERROR 2024/12/13 16:56:25 battery 1 power: mul[1]: add[0]: read failed: read tcp 127.0.0.1:45098->127.0.0.1:5023: i/o timeout
[site ] ERROR 2024/12/13 16:57:10 grid currents: div[0]: read failed: modbus: exception '4' (server device failure), function '4'
[lp-1 ] ERROR 2024/12/13 16:57:55 charge total import: write tcp 172.18.0.5:36438->10.9.192.41:502: write: connection reset by peer
[site ] ERROR 2024/12/13 16:58:10 pv 1 energy: read failed: modbus: exception '4' (server device failure), function '4'
[lp-2 ] ERROR 2024/12/13 16:58:10 charge total import: write tcp 172.18.0.5:48252->10.9.192.44:502: write: connection reset by peer 
```

mucki12, 2024-12-13:
Sorry, kann da wahrscheinlich auch nicht weiterhelfen.
Aber ich dachte bislang, dass evcc in der meters/ chargers Konfiguration nicht auf den eigenen Modbus Proxy referenzieren kann. Dachte immer, dass man dort die echte IP+Port des jeweiligen Geräts einträgt und anschließend in der modbusproxy Konfiguration nur einen Eintrag hat.

Wieder etwas dazu gelernt...

andig, 2024-12-13:
> Ich verstehe dich nicht ganz. Der Proxy dient ja genau dem Zweck, dass dann mehrere Devices zugreifen können, oder?

Ja. Der regelt- zumindest aktuell- aber auch nur gleichzeitige Zugriffe. Er erzwingt z.B. nicht bestimmte Delays zwischen den Zugriffen einzelner Clients. Was die KEBA hier _konkret_ stört kann Dir wohl nur der Hersteller sagen :(

> Könntest du mir auf die Sprünge helfen in wie weit mir das helfen kann?

Gerne in dem Thread fragen. Wenn ich das Thema richtig verstehe dann gibt es Abhängigkeiten zur Sungrow Firmware als auch zum gewählten Anschluss.

andig, 2024-12-14:
Also zurück zur Keba. Was sagt

    evcc meter --log trace

maxschloegl, 2024-12-14:
Hier der Output. Für mich unauffällig.
```
[main  ] INFO 2024/12/14 12:41:27 evcc 0.131.11 (d4cb42f7)
[main  ] INFO 2024/12/14 12:41:27 using config file: /etc/evcc.yaml
[db    ] INFO 2024/12/14 12:41:27 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/12/14 12:41:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/12/14 12:41:27 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/12/14 12:41:27 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/12/14 12:41:27 SELECT * FROM `settings` 26 <nil>
[mqtt  ] INFO 2024/12/14 12:41:28 connecting evcc-1786672911 at tcp://10.90.3.21:1883
[mqtt  ] DEBUG 2024/12/14 12:41:28 tcp://10.90.3.21:1883 connected
[db    ] TRACE 2024/12/14 12:41:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/12/14 12:41:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/12/14 12:41:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/12/14 12:41:28 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/12/14 12:41:28 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/12/14 12:41:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/12/14 12:41:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/12/14 12:41:28 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[heizstab] INFO 2024/12/14 12:41:28 connecting evcc-216418465 at tcp://10.90.3.21:1883
[heizstab] DEBUG 2024/12/14 12:41:28 tcp://10.90.3.21:1883 connected
[heizstab] DEBUG 2024/12/14 12:41:28 tcp://10.90.3.21:1883 subscribe tele/evcc/stg114/heizraum/puffer/heizstab/chargestatus
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 01 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 01 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 01 00 00 00 07 ff 03 04 00 04 c2 ca
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 02 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 01 00 00 00 07 ff 03 04 00 04 c2 df
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 02 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 03
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 03 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 03 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 03 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 03 00 00 00 07 ff 03 04 00 00 00 00
wallbox-garage
--------------
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 04 00 00 00 06 ff 03 03 fc 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 04 00 00 00 07 ff 03 04 00 69 4a 92
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 05 00 00 00 06 ff 03 04 0c 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 05 00 00 00 07 ff 03 04 00 c7 6f fc
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 06 00 00 00 06 ff 03 03 f0 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 06 00 00 00 07 ff 03 04 00 00 3d 51
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 07 00 00 00 06 ff 03 03 f2 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 07 00 00 00 07 ff 03 04 00 00 3e 14
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 08 00 00 00 06 ff 03 03 f4 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 08 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 09 00 00 00 06 ff 03 03 ec 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 09 00 00 00 07 ff 03 04 00 00 00 07
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 0a 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 0a 00 00 00 07 ff 03 04 00 00 00 03
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 0b 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 0b 00 00 00 07 ff 03 04 00 00 00 03
Power:          6900W
Energy:         1307.0kWh
Current L1..L3: 15.7A 15.9A 0A
Charge status:  C
Enabled:        true

wallbox-carport
---------------
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 04 00 00 00 06 ff 03 03 fc 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 04 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 05 00 00 00 06 ff 03 04 0c 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 05 00 00 00 07 ff 03 04 01 4a 39 44
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 06 00 00 00 06 ff 03 03 f0 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 06 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 07 00 00 00 06 ff 03 03 f2 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 07 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 08 00 00 00 06 ff 03 03 f4 00 02
[... cut ...]
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 0e 00 00 00 07 ff 03 04 00 00 00 01
[keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 0f 00 00 00 06 ff 03 05 dc 00 02
[keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 0f 00 00 00 07 ff 03 04 00 00 00 00
Power:          0W
Energy:         2164.2kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Status reason:  unknown
Enabled:        true
Phases:         3
Identifier:     00000000

heizstab
--------
Charge status: outdated
Enabled:       outdated
Features:      [IntegratedDevice Heating]
```

andig, 2024-12-14:
[keba ] TRACE 2024/12/14 11:39:25 modbus: send 00 28 00 00 00 06 ff 03 04 0c 00 02
    [lp-1 ] ERROR 2024/12/14 11:39:25 charge total import: write tcp 172.18.0.7:42820->10.9.192.41:502: write: connection reset by peer

vs

    [keba  ] TRACE 2024/12/14 12:41:28 modbus: send 00 05 00 00 00 06 ff 03 04 0c 00 02
    [keba  ] TRACE 2024/12/14 12:41:28 modbus: recv 00 05 00 00 00 07 ff 03 04 01 4a 39 44

Dann frag bitte KEBA was da los ist...

closed 2024-12-14 by andig