evcc-io/evcc #17715: Vehicle can not connect anymore with 0.131.11 (vw ID family)

timoglaess, 2024-12-13:
### Describe the bug

During config test in the vehicle configuration: "cannot create vehicle type 'template': cannot create vehicle type 'vw': missing state"

### Steps to reproduce

1. Updated to v.0.131.11
2. Vehicle VW ID.3 can not connect anymore (configured in experimental Configpage)
3.
...

### Configuration details

```yaml
configere via experimental console
```

### Log details

```text
[main ] INFO 2024/12/13 12:16:32 evcc 0.131.11
[main ] INFO 2024/12/13 12:16:32 using config file: /etc/evcc.yaml
[db ] INFO 2024/12/13 12:16:32 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/12/13 12:16:33 listening at :7070
[main ] ERROR 2024/12/13 12:16:34 creating vehicle db:5 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': missing state
[lp-1 ] WARN 2024/12/13 12:16:34 poll mode '{always 15m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site ] INFO 2024/12/13 12:16:44 site config:
[site ] INFO 2024/12/13 12:16:44 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/12/13 12:16:44 grid: power ✓ energy ✓ currents ✗
[site ] INFO 2024/12/13 12:16:44 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2024/12/13 12:16:44 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2024/12/13 12:16:44 vehicles:
[site ] INFO 2024/12/13 12:16:44 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1 ] INFO 2024/12/13 12:16:44 loadpoint 1:
[lp-1 ] INFO 2024/12/13 12:16:44 mode: off
[lp-1 ] INFO 2024/12/13 12:16:44 charger: power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/12/13 12:16:44 meters: charge ✓
[lp-1 ] INFO 2024/12/13 12:16:44 charge: power ✓ energy ✗ currents ✓
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.11

CarG70, 2024-12-13:
Everything is fine here with ID.4 and 0.131.11.
Vehicle in yaml, no poll mode and nothing to accept on myvolkswagen website. So the problem is not in general.

seppelw, 2024-12-13:
Get the same issue like OP. Vehicle is not in yaml but configured using the frontend!
Reconfiguring leads to the same issue.
[image]

jph76, 2024-12-13:
Can confirm with 0.131.9, “Ladebereit. Warte auf Fahrzeug”.

MySkoda 1.15.3 integration for Home Assistant shows correct charging power.

J0ergW, 2024-12-13:
Same for me with my ID.4 with 0.131.11

But I saw this issue with 0.131.6 too, where it was working well for a while.
Maybe related to changes in VW network.

timoglaess, 2024-12-13:
Right J0ergW. That's it!
It is the disclaimer confirmation blocking the log-in.
Need to log into myVolkswagen account and confirm the disclaimer warning one time, then evcc is able again to connect.

closed 2024-12-13 by timoglaess

jph76, 2024-12-13:
@timoglaess Please reconsider closing this issue. I've got no consent to agree to and it still does not work in evcc. Please note that MySkoda HA integration works.

MySkoda HA integration will btw detect a missing consent and notify me accordingly, which it currently does not.

jph76, 2024-12-13:
Did a downgrade from evcc 0.131.9 to 0.131.8 and now everythings works again.

MehrCurry, 2024-12-15:
Same here with evcc evcc 0.131.11. I logged in at Volkswagen add clicked the disclaimer but error persists.

CarG70, 2024-12-15:
> Same here with evcc evcc 0.131.11. I logged in at Volkswagen add clicked the disclaimer but error persists.

evcc restarted?

jph76, 2024-12-15:
Boring afternoon, tried something…

- evcc 0.131.8: works
- evcc 0.131.9: does not work
- evcc 0.131.11: works again

Issue resolved on my side.

Running stable evcc on Debian 12 in a LXC container. PV access is via modbus-proxy, that’s why Home Assistant shows in the logs. Car is a Skoda Enyaq but that’s the same VW stuff anyway.

Log excerpt from 0.131.9:
```
Dez 15 13:34:43 evcc systemd[1]: Started evcc.service - evcc.
Dez 15 13:34:43 evcc evcc[30576]: [main  ] INFO 2024/12/15 13:34:43 evcc 0.131.11
Dez 15 13:34:43 evcc evcc[30576]: [main  ] INFO 2024/12/15 13:34:43 using config file: /etc/evcc.yaml
Dez 15 13:34:43 evcc evcc[30576]: [db    ] INFO 2024/12/15 13:34:43 using sqlite database: /var/lib/evcc/evcc.db
Dez 15 13:34:44 evcc evcc[30576]: [mqtt  ] INFO 2024/12/15 13:34:44 connecting evcc-... at tcp://homeassistant:1883
Dez 15 13:34:44 evcc evcc[30576]: [eebus ] INFO 2024/12/15 13:34:44 Local SKI: ...
Dez 15 13:34:44 evcc evcc[30576]: [mqtt  ] DEBUG 2024/12/15 13:34:44 tcp://homeassistant:1883 connected
Dez 15 13:34:44 evcc evcc[30576]: [eebus ] DEBUG 2024/12/15 13:34:44 starting websocket server on :4712
Dez 15 13:34:44 evcc evcc[30576]: [eebus ] DEBUG 2024/12/15 13:34:44 mdns: announce
Dez 15 13:34:44 evcc evcc[30576]: [eebus ] DEBUG 2024/12/15 13:34:44 mdns: using zeroconf
Dez 15 13:34:44 evcc evcc[30576]: [main  ] INFO 2024/12/15 13:34:44 listening at :7070
Dez 15 13:34:44 evcc evcc[30576]: [eebus ] DEBUG 2024/12/15 13:34:44 mdns: new - ski: ... name: Elli-Wallbox-... brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-... register: false host: wallbox-....local. port: 4712 addresses: []
...
Dez 15 13:34:47 evcc evcc[30576]: [site  ] INFO 2024/12/15 13:34:47 site config:
Dez 15 13:34:47 evcc evcc[30576]: [site  ] INFO 2024/12/15 13:34:47   meters:      grid ✓ pv ✓ battery ✓
Dez 15 13:34:47 evcc evcc[30576]: [site  ] INFO 2024/12/15 13:34:47     grid:      power ✓ energy ✓ currents ✓
Dez 15 13:34:47 evcc evcc[30576]: [site  ] INFO 2024/12/15 13:34:47     pv 1:      power ✓ energy ✓ currents ✗
Dez 15 13:34:47 evcc evcc[30576]: [site  ] INFO 2024/12/15 13:34:47     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
Dez 15 13:34:47 evcc evcc[30576]: [site  ] INFO 2024/12/15 13:34:47   vehicles:
Dez 15 13:34:47 evcc evcc[30576]: [site  ] INFO 2024/12/15 13:34:47     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
Dez 15 13:34:47 evcc evcc[30576]: [lp-1  ] INFO 2024/12/15 13:34:47 loadpoint 1:
Dez 15 13:34:47 evcc evcc[30576]: [lp-1  ] INFO 2024/12/15 13:34:47   mode:        minpv
Dez 15 13:34:47 evcc evcc[30576]: [lp-1  ] INFO 2024/12/15 13:34:47   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
Dez 15 13:34:47 evcc evcc[30576]: [lp-1  ] INFO 2024/12/15 13:34:47   meters:      charge ✓
Dez 15 13:34:47 evcc evcc[30576]: [lp-1  ] INFO 2024/12/15 13:34:47     charge:    power ✓ energy ✗ currents ✓
Dez 15 13:34:47 evcc evcc[30576]: [site  ] DEBUG 2024/12/15 13:34:47 set buffer soc: 80
Dez 15 13:34:47 evcc evcc[30576]: [site  ] DEBUG 2024/12/15 13:34:47 set priority soc: 20
Dez 15 13:34:47 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:34:47 phase timer inactive
Dez 15 13:34:47 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:34:47 pv timer inactive
Dez 15 13:34:47 evcc evcc[30576]: [site  ] DEBUG 2024/12/15 13:34:47 ----
...
Dez 15 13:36:04 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:04 charger status: C
Dez 15 13:36:04 evcc evcc[30576]: [lp-1  ] INFO 2024/12/15 13:36:04 car connected
Dez 15 13:36:04 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:04 vehicle api refresh
...
Dez 15 13:36:07 evcc evcc[30576]: [main  ] DEBUG 2024/12/15 13:36:07 vehicle status: C (Das blaue Wunder)
Dez 15 13:36:07 evcc evcc[30576]: [lp-1  ] INFO 2024/12/15 13:36:07 vehicle updated: unknown -> Das blaue Wunder
Dez 15 13:36:07 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:07 vehicle status: C
Dez 15 13:36:07 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:07 vehicle soc: 53%
Dez 15 13:36:07 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:07 vehicle soc limit: 80%
Dez 15 13:36:07 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:07 vehicle range: 190km
Dez 15 13:36:07 evcc evcc[30576]: [lp-1  ] WARN 2024/12/15 13:36:07 charger logic error: disabled but charging
...
Dez 15 13:36:34 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:34 vehicle odometer: 30701km
Dez 15 13:36:34 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:34 charger status: C
Dez 15 13:36:34 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:34 soc estimated: 53.04% (vehicle: 53.00%)
Dez 15 13:36:34 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:34 vehicle soc: 53%
Dez 15 13:36:34 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:34 vehicle soc limit: 80%
Dez 15 13:36:34 evcc evcc[30576]: [lp-1  ] DEBUG 2024/12/15 13:36:34 vehicle range: 190km
```

Log from 0.131.11:

```
Dez 15 13:32:24 evcc systemd[1]: Started evcc.service - evcc.
Dez 15 13:32:24 evcc evcc[28546]: [main  ] INFO 2024/12/15 13:32:24 evcc 0.131.9
Dez 15 13:32:24 evcc evcc[28546]: [main  ] INFO 2024/12/15 13:32:24 using config file: /etc/evcc.yaml
Dez 15 13:32:24 evcc evcc[28546]: [db    ] INFO 2024/12/15 13:32:24 using sqlite database: /var/lib/evcc/evcc.db
[... 25 lines cut ...]
Dez 15 13:32:28 evcc evcc[28546]: [site  ] DEBUG 2024/12/15 13:32:28 ----
...
Dez 15 13:33:18 evcc evcc[28546]: [lp-1  ] DEBUG 2024/12/15 13:33:18 charger status: B
Dez 15 13:33:18 evcc evcc[28546]: [lp-1  ] INFO 2024/12/15 13:33:18 car connected
[... cut ...]
Dez 15 13:33:48 evcc evcc[28546]: [lp-1  ] INFO 2024/12/15 13:33:48 start charging ->
Dez 15 13:33:48 evcc evcc[28546]: [lp-1  ] DEBUG 2024/12/15 13:33:48 wake-up timer: stop
Dez 15 13:33:48 evcc evcc[28546]: [lp-1  ] DEBUG 2024/12/15 13:33:48 soc estimated: 54.00% (vehicle: 54.00%)
Dez 15 13:33:48 evcc evcc[28546]: [lp-1  ] DEBUG 2024/12/15 13:33:48 vehicle soc: 54%
Dez 15 13:33:48 evcc evcc[28546]: [lp-1  ] DEBUG 2024/12/15 13:33:48 vehicle soc limit: 80%
Dez 15 13:33:48 evcc evcc[28546]: [lp-1  ] DEBUG 2024/12/15 13:33:48 vehicle range: 190km
```