evcc-io/evcc #17750: Sungrow AC011E-01: cannot create charger type 'template': cannot create charger type 'ocpp'

shellfisch, 2024-12-15:
### Describe the bug

EVCC zeigt zunächst - vor dem Ladevorgang - im Log keine Fehlermeldungen. Ladevorgang sollte heute Nacht starten. Ca 2 Stunden nach dem Anstecken meldet das Log: 
cannot create charger type 'ocpp': timeout

### Steps to reproduce

1.Schwierig, weil zunächst fehlerfrei, dann nach ein paar stunden trat Fehler auf
2. mehrfacher Neustart der Box zeigte keine Änderung
3.
...

### Configuration details

```yaml
meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.42  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 10s  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.42  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.42  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 10s  
  name: battery3

chargers:
- type: template
  template: ocpp 
  stationid: A22xxxxxx 
  name: wallbox5
  metervalues: -Current.Offered
  remotestart: true
```

### Log details

```text
[main ] INFO 2024/12/15 21:57:04 evcc 0.131.11 (b14faf00)
[main ] INFO 2024/12/15 21:57:04 using config file: /etc/evcc.yaml
[db ] INFO 2024/12/15 21:57:04 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/12/15 21:57:04 listening at :7070
[A2292802444-1] DEBUG 2024/12/15 21:57:04 waiting for chargepoint: 5m0s
[main ] FATAL 2024/12/15 22:02:04 cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main ] FATAL 2024/12/15 22:02:04 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.11 (b14faf00)

andig, 2024-12-15:
> Schwierig, weil zunächst fehlerfrei, dann nach ein paar stunden trat Fehler auf

Unklar. Der Fehler passiert offensichtlich direkt beim Start.

closed 2024-12-15 by andig

shellfisch, 2024-12-15:
Was soll ich machen, bin ratlos

shellfisch, 2024-12-15:
Leider nein, Fehler trat nicht beim Start auf sondern 2 Stunden nach dem anstecken am Auto und ca 6 Stunden bevor der ladevorgang beginnen sollte . In der Zwischenzeit wurde EVCC oder das Auto oder die WB nicht angerührt

premultiply, 2024-12-15:
Der Logschnipsel zeigt nur dass evcc um 21:57:04 (neu)gestartet wurde und sich die Wallbox dann nicht innerhalb von 5 Minuten (wieder?) bei evcc gemeldet hat.
Mehr ist daraus nicht zu erkennen und für weitere Wahrsagereien müssten wir wieder auf unsere Glaskugel zurückgreifen, die sich aber leider immer noch in Reparatur befindet.

shellfisch, 2024-12-15:
Vielen Dank für die Rückmeldung. Mehr als das logschnipsel gibt das log leider nicht her. Ich habe bezgl. des Fehlers eine ganze Reihe anderer threads gelesen aber nichts gefunden. Und dass das log um 21.54 beginnt ist darauf zurückzuführen, dass ich evcc mehrfach gestartet habe, die Box mehrfach gestartet habe, auch mal 1/2 Stunde alles stromlos war, den Netzwerkeinteag in der Box kontrolliert habe. Sorry aber das log gibt einfach nicht mehr her und das mit der Glaskugel ist schon klar, habe aber definitiv auch keine Idee wie ich den Fehler besser eingrenzen oder beschreiben kann oder den Laden wieder ans laufen kriegen kann. Ich bin jetzt hier ausschließlich auf euer Wissen und Erfahrung angewiesen. selbst komme ich hier nicht mehr weiter. Vielen Dank nochmal

premultiply, 2024-12-16:
> die Box mehrfach gestartet habe, auch mal 1/2 Stunde alles stromlos war, den Netzwerkeinteag in der Box kontrolliert habe

Und die IP-Adresse des evcc-Systems ist dabei auch noch gleich geblieben und die Box hat somit eine theoretische Chance diese unter der konfigurierten Adresse auch zu erreichen?

andig, 2024-12-16:
Hättest du ein trace Log? Ich hab allerdings den Verdacht, dass sich die WB nie bei evcc meldet. Aber schauen wir mal sicherheitshalber nach.

shellfisch, 2024-12-16:
Guten Morgen, sorry ich war doch etwas voreilig was meine Mini-Log-Behauptung angeht. Hier kommt das komplette Log: 
[evcc-20241216-084629-trace.log](https://github.com/user-attachments/files/18146569/evcc-20241216-084629-trace.log)

premultiply, 2024-12-16:
Es bleibt dabei: Die Box meldet sich nicht bei evcc an.

shellfisch, 2024-12-16:
Hier sieht es doch so aus, als würde sich die Box verbinden, nachdem ich die Netzeinstellungen in der Box bestätigt - nicht geändert - habe: 
[evcc-20241216-092613-debug.log](https://github.com/user-attachments/files/18147276/evcc-20241216-092613-debug.log)
Dann habe ich wiederum ein anderes Log nach einem erneuten Neustart der Box: 
[evcc-20241216-093613-trace.log](https://github.com/user-attachments/files/18147309/evcc-20241216-093613-trace.log)
Ich weiß mir weiterhin keinen Rat; hat die Firmware einen Schuss, die Hardware defekt, Ohne Änderungen an der evcc.yaml, ohne eingreifen am Auto, der Box läuft der Laden nicht mehr nachdem er wochenlang problemlos lief. 
Was würdet ihr denn vermuten, ein Zurückgreifen auf den sungrow-Support ist ziemlich aussichtslos. Habt ihr evtl einen Vorschlag zur weiteren Vorgehensweise?
Herzlichen Dank nochmal

andig, 2024-12-16:
Leider nein :( Manche Boxen haben ein internes Log- vllt. kann Dir der Support ja doch einen Tipp geben...