evcc-io/evcc #17768: Excessive Modbus Trace Logging in Logfiles

igotsoul, 2024-12-17:
### Describe the bug

For a while I noticed excessive Modbus trace logging appears in the logs even when the default log level is set to "INFO" or lower. This results in bloated logfiles, making it difficult to analyze logs for errors or important information.

From the discussion, it seems that Modbus traces are logged unintentionally or that trace logging cannot be properly disabled.

### Steps to reproduce

1. Update EVCC to the latest version. (though behavior exists already for several releases)
2. Ensure the log level in evcc.yaml is set to info
3. Run EVCC and allow it to interact with a Modbus-based charger or meter. (Here SMA components)
4. Check the log output (evcc.log or console output).

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: x.x.x.x
  port: 7070

log: info
levels:
  cache: error
# modbus: info

# unique installation id
plant: xxx
sponsortoken: xxx
telemetry: true
interval: 15s # control cycle interval

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: x.x.x.x
  name: grid1
  interface: eth0
- type: template
  template: sma-inverter-modbus
  id: 3
  host: x.x.x.x
  port: 502
  usage: pv
  modbus: tcpip
  name: sunnyboy3.0
- type: template
  template: sma-inverter-modbus
  id: 3
  host: x.x.x.x
  port: 502
  usage: pv
  modbus: tcpip
  name: sunnyboy4.0
- type: template
  template: shelly-3em
  usage: pv
  host: x.x.x.x
  name: shelly
chargers:
- type: template
  template: openwb
  host: x.x.x.x
  name: wallbox

loadpoints:
- title: Carport
  charger: wallbox
[... 142 lines cut ...]
[sunnyboy4.0] TRACE 2024/12/17 13:13:59 modbus: recv 02 04 00 00 00 0b 03 03 08 00 00 00 00 01 65 c7 fe
[site ] DEBUG 2024/12/17 13:13:59 pv 2 power: 116W
[sunnyboy3.0] TRACE 2024/12/17 13:14:00 modbus: recv 02 04 00 00 00 0b 03 03 08 00 00 00 00 00 f3 ae 40
[site ] DEBUG 2024/12/17 13:14:00 pv 1 power: 85W
[site ] DEBUG 2024/12/17 13:14:00 pv 3 power: 18W
[site ] DEBUG 2024/12/17 13:14:00 pv power: 219W
[site ] DEBUG 2024/12/17 13:14:00 site power: 2370W
[lp-1 ] DEBUG 2024/12/17 13:14:00 charge total import: 15415.110kWh
[lp-1 ] DEBUG 2024/12/17 13:14:00 charger status: A
[site ] DEBUG 2024/12/17 13:14:14 ----
[lp-1 ] DEBUG 2024/12/17 13:14:14 charge power: 0W
[lp-1 ] DEBUG 2024/12/17 13:14:14 charge currents: [0.03 0.02 0.03]A
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.11

andig, 2024-12-17:
Kann ich nicht nachvollziehen:

```
❯ evcc meter -c logging.yaml
[main  ] INFO 2024/12/17 13:25:05 evcc 0.0.0
[main  ] INFO 2024/12/17 13:25:05 using config file: logging.yaml
[db    ] INFO 2024/12/17 13:25:05 using sqlite database: /Users/andig/.evcc/evcc.db
Power:          346W
Energy:         23745.1kWh
Current L1..L3: 0.932A 0.381A 1.65A
```

mit `logging.yaml`:

```yaml
log: info
meters:
  - name: pv
    type: template
    template: eastron
    modbus: rs485tcpip
    host: rs485.fritz.box
```

Hast du das Problem mit `evcc meter` auch?

GrimmiMeloni, 2024-12-17:
@naltatis where does the mentioned filtering happen? Is this in the frontend, i.e. something you can look into?

igotsoul, 2024-12-17:
Ich dachte, das hätte ich beantwortet. Ausschließlich im UI. Wir sprechen hier nur über die Darstellung im UI.

andig, 2024-12-17:
Naja- dein Issuetitel sagt da was anderes.

igotsoul, 2024-12-17:
Touché, ich hatte fälschlicherweise angenommen, dass die Ansicht in der WebUI dem Logfile/Journal entspricht.
Ja, ich hatte geprüft, im Logfile taucht es nicht auf. Das beschriebene Szenario ist nur in der WebUI Ansicht zur erkennen. Ich passen den Issue Titel an.

naltatis, 2024-12-17:
Ja, kann ich bestätigen.
@GrimmiMeloni das Filtern findet auf der Server-Seite statt. Nicht im UI. Da wird lediglich das Log-Level übergeben.

[image]

naltatis, 2024-12-17:
Ok, Problem gefunden. Das betrifft nicht `TRACE` sondern generell Areas mit Sonderzeichen. Bei @igotsoul wars der Punkt `sunnyboy3.0`, bei mir der Unterstrich `sungrow_battery`

GrimmiMeloni, 2024-12-18:
@andig grabbing this one.

andig, 2024-12-18:
@GrimmiMeloni es gibt schon einen PR ;)

closed 2024-12-18 by andig