evcc-io/evcc #17772: Web endpoint - power consumption  of meter in median average

Bevute, 2024-12-17:
### Web endpoint - power consumption  of meter in median average 

I am currently using the CMD command ‘evcc meter’ to read out the power status. I use this and write it to a sqlite DB. I calculate the median as an average. I also have a web endpoint in python that provides me with the current median value. 

**Reason for this:**
I need the PV surplus of the PV system and use evcc to read the smartmeter. At a certain size, in my case 4KW, I start the Smarthome hot water generation with the heat pump. 

**My feature request:**
I would like to have a web endpoint that displays this configurably natively and does not have to use the CMD commands for this. If you would like me to collaborate here, I would be happy to help.

andig, 2024-12-17:
We don't plan to do this as it's not required for evcc operations.

closed 2024-12-17 by andig

StefanSchoof, 2024-12-18:
Did you try to use the `api/state` endpoint for your information?
https://docs.evcc.io/en/docs/integrations/rest-api#state