evcc-io/evcc #17773: Fenster Konfiguration.Integrationen.Lastmanagment nicht mehr sichtbar nach Fehlkonfiguration des Lastmanagements

hauserandi, 2024-12-17:
### Describe the bug

Hallo Zusammen, 
da ich eine zweite Wallbox in EVCC integrieren möchte habe ich gestern mit der Integration Lastmanagement experimentiert.
Das hab ich leider nicht in der evcc.yaml Datei sondern über die graphische Konfigurationsoberfläche versucht.
Durch eine Fehlkonfiguration startet nun EVCC nicht mehr richtig hoch.
Ich kann den Fehler im Lastmanagement allerdings nicht mehr korrigieren, da ich das zugehörige Integrationsfenster nicht mehr öffnen kann. Das Fenster fehlt einfach in der Konfigurationsansicht:
[image]
Hier die aktuelle Log Datei
[evcc-20241217-160648-debug.log](https://github.com/user-attachments/files/18167405/evcc-20241217-160648-debug.log)
Das Start Fenster sieht so aus:
[image]
Mir scheint als wäre durch meine Tests in der Datenbank eine circuit "Halle" angelegt worden. 
Weiß vielleicht jemand wie ich diese Definition wieder aus der Datenbank bekomme?

Könnt ihr mir hier weiter helfen?

### Steps to reproduce

1. GUI Konfiguration öffnen
2. zu Integrationen navigieren
3. Lastmanagement öffnen
4. fehlerhafte Lastmanagement Konfiguration erstellen
5. EVCC neu starten

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ***

interval: 10s # control cycle interval

sponsortoken: ***

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: e3dc 
  usage: grid  
  host: 192.168.2.20  
  port: 502  
  name: grid1
- type: template
  template: e3dc 
  usage: pv  
  host: 192.168.2.20  
  port: 502  
  name: pv2
- type: template
  template: e3dc 
  usage: battery  
  host: 192.168.2.20  
  port: 502  
  name: battery3

chargers:
[... 109 lines cut ...]
time="2024-12-17T16:03:39+01:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2024-12-17T16:03:39+01:00" level=info msg="successfully connected to 192.168.2.20:5033"
time="2024-12-17T16:03:39+01:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2024-12-17T16:03:39+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2024-12-17T16:03:39+01:00" level=info msg="Connecting to 192.168.2.20:5033"
time="2024-12-17T16:03:39+01:00" level=info msg="successfully connected to 192.168.2.20:5033"
time="2024-12-17T16:03:39+01:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2024-12-17T16:03:39+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2024-12-17T16:03:39+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2024-12-17T16:03:39+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2024-12-17T16:03:39+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2024-12-17T16:03:39+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.11

andig, 2024-12-17:
Hier gibts nur manuellen Workaround- Du musst den circuit manuell aus der `configs` Tabelle löschen.

hauserandi, 2024-12-17:
Hi andig,
kannst du mir einen Tip geben wo ich die configs Tabelle auf meinem Raspberry Pi finde und wie ich diese bearbeiten kann?
Habe bisher nur mit Hilfe der Software MobaXterm mit der normalen Konfig Datei "/etc/evcc.yaml" gearbeitet.

andig, 2024-12-17:
Welche das ist siehst Du beim Start im Log:

    [db    ] INFO 2024/12/17 13:52:54 using sqlite database: /Users/andig/.evcc/evcc.db

andig, 2024-12-17:
Das ist eine SQlite DB die entsprechen mit SQlite und SQL bearbeitet werden muss. NICHT mit einem Texteditor  ;)

hauserandi, 2024-12-17:
... womit ich aktuell überfordert bin, da ich damit keine Erfahrung habe :-(

naltatis, 2024-12-17:
@andig zum konkreten Fehler: `circuit Halle has no meter and no loadpoint assigned`. Hier können bzw. sollten wir toleranter sein. Wir können nicht sicher stellen, dass Ladepunktzuordnung und Anlegen des Circuits immer in sync passieren. Heißt, wenn der Circuit keine Zuordnung hat sollten wir warnen, aber nicht failen.

naltatis, 2024-12-17:
@hauserandi die Holzhammermethode wäre die `evcc.db` einfach komplett zu löschen (oder umzubenennen). Dann legt evcc beim nächsten Start einfach eine frische an.

hauserandi, 2024-12-17:
Ja, mein Ziel wäre die Ladehistorie nicht zu verlieren.

evcc migrate --reset 
hatte ich bereits versucht. Greift allerdings wohl auf eine andere evcc.db zu.

EVCC_DATABASE_DSN=/var/lib/evcc/evcc.db evcc migrate --reset
säubert dann vermutlich die richtige db.

In der Kommandozeile sieht das wie folgt aus.
[image]

Allerdings helfen beide Befehle leider nicht um die db zu säubern.
Die Fehlermeldung nach EVCC Restart bleibt unverändert:
[image]

hauserandi, 2024-12-17:
Ja, hab evcc vorher gestoppt und danach wieder gestartet. Siehe Kommandozeilen Screenhot oben.
Mail mit DB is raus.

naltatis, 2024-12-17:
Danke. Gesäuberte Datenbank ist per Mail an dich zurück. Bei mir hat das Reset wie erwartet funktioniert. Merkwürdig, dass das bei dir scheinbar nicht funktioniert hat.

Hier noch mal meine Log-Ausgabe mit ` --log TRACE` Parameter damit man die SQL-Befehle sieht. Wäre spannend zu sehen was bei dir anders ist.

```
EVCC_DATABASE_DSN=/Users/michael/Downloads/evcc.db ./evcc migrate --reset --log TRACE
[main  ] INFO 2024/12/17 20:02:41 evcc 0.131.11 (d90c12c3e)
[main  ] INFO 2024/12/17 20:02:41 using config file: cmd/demo.yaml
[db    ] INFO 2024/12/17 20:02:41 using sqlite database: /Users/michael/Downloads/evcc.db
[db    ] TRACE 2024/12/17 20:02:41 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/12/17 20:02:41 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/12/17 20:02:41 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/12/17 20:02:41 SELECT * FROM `settings` 45 <nil>
[main  ] WARN 2024/12/17 20:02:41 resetting:
[main  ] DEBUG 2024/12/17 20:02:41 - global settings
[main  ] DEBUG 2024/12/17 20:02:41 - network
[main  ] DEBUG 2024/12/17 20:02:41 - mqtt
[main  ] DEBUG 2024/12/17 20:02:41 - influx
[main  ] DEBUG 2024/12/17 20:02:41 - hems
[main  ] DEBUG 2024/12/17 20:02:41 - eebus
[db    ] TRACE 2024/12/17 20:02:41 DELETE FROM `settings` WHERE `settings`.`key` = "eebus" 1 <nil>
[main  ] DEBUG 2024/12/17 20:02:41 - modbusproxy
[main  ] DEBUG 2024/12/17 20:02:41 - messaging
[main  ] DEBUG 2024/12/17 20:02:41 - tariffs
[main  ] DEBUG 2024/12/17 20:02:41 - circuits
[db    ] TRACE 2024/12/17 20:02:41 DELETE FROM `settings` WHERE `settings`.`key` = "circuits" 1 <nil>
```

hauserandi, 2024-12-17:
Oh ja mit 'sudo' voraus und dem richtigen .db Pfad hat es nun auch bei mir geklappt die evcc.db automatisch ohne SQLLite Fachwissen zu säubern.

`sudo EVCC_DATABASE_DSN=/var/lib/evcc/evcc.db evcc migrate --reset --log TRACE`

[image]

=> Definitiv klassisches Halbwissen auf Anwenderseite.
Die Konfigurations inkl. Ladehistorie ist gerettet.
Weite Tests zum Lastmanagement mach ich dann erstmal nur direkt in der evcc.yaml

Vielen, Vielen Dank für eure Hilfe! 👍 👍 👍 
I love EVCC :-)

naltatis, 2024-12-17:
Super, dass das geklappt hat. Das wir bei den fehlenden rechten keine Warnung ausgeben ist nicht optimal (da schau ich noch mal rein). Aber damit ist das Verhalten erklärt.

closed 2024-12-17 by naltatis