evcc-io/evcc #17774: Tessie - Remove welcomecharge

djfanatix, 2024-12-17:
### Describe the bug

@andig I notice that the Tessie vehicle is doing a welcomecharge when connecting.
Is it possible to disable it and how?
If you point me to the code, I will make pull request.

### Steps to reproduce

1. Connect ev

### Configuration details

```yaml
Na
```

### Log details

```text
Na
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

Latest

VolkerK62, 2024-12-17:
>is doing a welcomecharge

sure??  is it visible in the log?
there ist no `welcomecharge` configured within the Tessie template

djfanatix, 2024-12-17:
No a go-e.

there is nothing in the logs, but it starts charging for 5 minutes when connecting the cable.

i will try to reproduce a few times

closed 2024-12-17 by andig