evcc-io/evcc #17775: Add APsystems ECU-3 as PV device

lenvm, 2024-12-17:
**Is your feature request related to a problem? Please describe.**
APsystems ECU-3 is not supported by evcc yet, and I would like to be able to read the current power delivered by the PV system.

**Describe the solution you'd like**
I would like to integrate APsystems ECU-3 as PV device.

**Describe alternatives you've considered**
- A Home Assistant integration is available, although I would like to use evcc instead.
- Interaction via rest commands. From [doe-duurzaam.nl](https://doe-duurzaam.nl/artikel/je-apsystems-micro-omvormers-slim-uitzetten-via-web-of-home-assistant/)

```
rest_command:
  solaroff:
    url: "http://ip-adres/index.php/configuration/set_switch_all_off"
  solaron:
    url: "http://ip-adres/index.php/configuration/set_switch_all_on"
```

**Additional context**
- I have found the following implementation of a Home Assistant integration for an APsystems ECU-3: https://github.com/jeeshofone/ha-apc-ecu-3

I would be glad to understand how to add the ECU-3. Perhaps the [code for the EZ1](templates/definition/meter/apsystems-ez1.yaml) can be used as a starting point? Please point me in the right direction, then I'll try to do the development and submit a pull request.

andig, 2024-12-17:
Happy to take a PR

closed 2024-12-29 by andig

volsk, 2025-06-27:
I see it's listed as completed, but I don't see it on the list. Is this correct?

volsk, 2025-06-28:
> [#17775 (comment)](https://github.com/evcc-io/evcc/issues/17775#issuecomment-2549569794)

I got it working via the Sunspec Inverter template. 
First I had to enable Sunspec Modbus, via
``` http://192.168.1.50/index.php/management/modbus ```
And give each inverter an address. See screenshot.
The I simply added them. I have 7 inverters. So now I have 7 entries:

```
- type: template
  template: sunspec-inverter
  id: 1
  host: 192.168.1.50
  port: 502
  usage: pv
  modbus: tcpip
  name: pv1

```

And each pv was added to the list of my home.
```

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv1
    - pv2
    - pv3
    - pv4
    - pv5
    - pv6
    - pv7
    battery:
    - hwe-bat
 
```
[image]