evcc-io/evcc #17781: Ecoflow Integration

psczech, 2024-12-18:
**Is your feature request related to a problem? Please describe.**
It is not a problem, but an enhancement.

**Describe the solution you'd like**
The idea is to integrate EcoFlow storage devices (e.g., Delta Pro 2 or Delta Max) and the PowerStream micro-inverter into EVCC for managing them as storage and inverters.

**Describe alternatives you've considered**
-

**Additional context**
My use case: In addition to a Huawei 10kWp system with storage (already integrated into EVCC), I also operate two EcoFlow balcony solar systems. These include two PowerStream inverters, a Delta Pro 2, and a Delta Max. It would be great if both balcony systems could be integrated into EVCC, allowing surplus charging from the main solar system to be realized as well.

andig, 2024-12-18:
This needs documentation. Happy to take a PR.

closed 2024-12-29 by andig

psczech, 2024-12-29:
Great. Thank you. How is it working?

stephanschaefer1912, 2025-01-02:
Hello. This issue was closed recently but I can not see how to Integrate the EcoFlow Powerstream inverter. Any hints if there is a solution?

andig, 2025-01-02:
Nothing has changed since https://github.com/evcc-io/evcc/issues/17781#issuecomment-2550933137

no-banker, 2025-01-06:
The documentation of all? devices from ecoflow can be found here:

[https://developer-eu.ecoflow.com/us/document/powerStreamMicroInverter](url)

And there is as well an integration in home assistant avaialable:

[https://github.com/svenerbe/ecoflow_dynamic_power_adjustment](url)

It would be great if integration into evcc could be implemented.

andig, 2025-01-06:
Just do it- that‘s what the evcc Plugins are for! The core team is not involved here.

JohnWheely, 2025-03-06:
Hallo Andig, there ist Notion in them links

andig, 2025-03-06:
> https://github.com/evcc-io/evcc/issues/17781#issuecomment-2573034590

Afaikt there is a plethora of different devices with different APIs. Somebody with a device will need to work out a `custom` meter configuration. Then it needs be bundled as template. Happy to take PR(s).

JohnWheely, 2025-03-06:
Hi, who can work out and bundle as a Template?

wookie2k, 2025-04-09:
any update? would be willing to contribute in a custom meter integration.
Even if the core team will not handle this in general, it would be helpful to get support here.

Balou1986, 2025-07-01:
If there is a solituion for this I would like to use is as well and can offer to do some testing.
Unfortunately, I don't have enough knowledge to write the necessary documentation and a corresponding pull request.

cedricziel, 2025-07-16:
Here's how I integrated my ecoflow STREAM (1x Ultra, 1x AC Pro) into evcc, hope it helps someone 😄 

Preconditions: HomeAssistant add-on for [ecoflow](https://github.com/tolwi/hassio-ecoflow-cloud) - it would work just via mqtt  with ecoflows developer accounts, but since I wanted to integrate with HA either way, I'm reusing the values..

```yaml
site:
  title: Home
  meters:
    grid: heimnetz
    pv:
      - ecoflow-pv-ultra
    battery:
      - ecoflow-battery-ultra
      - ecoflow-battery-ac-pro

meters:
  - name: heimnetz
    type: template
    template: shelly-3em
    usage: grid
    host: 192.168.178.124

  - name: ecoflow-pv-ultra
    type: custom
    power:
      source: http
      uri: http://192.168.178.40:8123/api/states/sensor.stream_ultra_0150_power_pv_sum
      method: GET
      headers:
        - Authorization: Bearer xxx
      insecure: true
      jq: .state|tonumber

  - name: ecoflow-battery-ultra
    type: custom
    power: # Leistung (W)
      source: http
      uri: http://192.168.178.40:8123/api/states/sensor.stream_ultra_0150_power_sys_load_from_battery
      method: GET
      headers:
        - Authorization: Bearer xxx
      insecure: true
      jq: .state|tonumber
    energy: # Energiemenge (kWh)
      source: http
      uri: http://192.168.178.40:8123/api/states/sensor.stream_ultra_0150_power_battery_soc
      method: GET
      headers:
        - Authorization: Bearer xxx
      insecure: true
      jq: .state|tonumber
    soc: # Batterie SOC (%)
      source: http
      uri: http://192.168.178.40:8123/api/states/sensor.stream_ultra_0150_power_battery_soc
      method: GET
      headers:
        - Authorization: Bearer xxx
      insecure: true
      jq: .state|tonumber
    capacity: 1.92

  - name: ecoflow-battery-ac-pro
    type: custom
    power: # Leistung (W)
      source: http
      uri: http://192.168.178.40:8123/api/states/sensor.stream_ultra_0150_power_sys_load_from_battery
      method: GET
      headers:
        - Authorization: Bearer xxx
      insecure: true
      jq: .state|tonumber
    energy: # Energiemenge (kWh)
      source: http
      uri: http://192.168.178.40:8123/api/states/sensor.stream_ultra_0150_power_battery_soc
      method: GET
      headers:
        - Authorization: Bearer xxx
      insecure: true
      jq: .state|tonumber
    soc: # Batterie SOC (%)
      source: http
      uri: http://192.168.178.40:8123/api/states/sensor.stream_ultra_0150_power_battery_soc
      method: GET
      headers:
        - Authorization: Bearer xxx
      insecure: true
      jq: .state|tonumber
    capacity: 1.92
    #currents: # Stromstärke (A) pro Phase
    #  - source: # Phase 1 Plugin Typ
    #  - source: # Phase 2 Plugin Typ
    #  - source: # Phase 3 Plugin Typ
```

czam70, 2025-09-14:
As I recently added an "EcoFlow Stream AC Pro" to my existing PV system, I tried to integrate it into EVCC as well using this HA integration by "tolwi" today.

My **(!)** judgement: not all required paramaters exist in EcoFlow's API or this integration yet, so it will not work as needed by EVCC as of now.

First, it took me a while to figure out that you need to provide the custom repository "https://github.com/tolwi/hassio-ecoflow-cloud" to HACS in HA to install the "tolwi-version". I also had to find out what the "Bearer-lines" mena (use a token to access HA) and how to create a "long living" HA token (hint: see your HA profile and "Security"). But OK, I figured all of that out...

I was then able to configure my  "EcoFlow Stream AC Pro" with the "private_api" (using my EcoFlow credentials) and the Device Type "STREAM_AC". I chose to name it "ac_pro".
The **Power** value can actually be read, but the correct meter-code for EVCC is the following:

```
- name: ecoflow-battery-ac-pro
    type: custom
    power: # Leistung (W)
      source: http
      uri: http://192.168.178.40:8123/api/states/sensor.ac_pro**_power_ac_sys**
      method: GET
      headers:
        - Authorization: Bearer xxx
      insecure: true
      jq: .state|tonumber
```
But both parameters **energy** and **soc** are not supported be EcoFlow, as it seems. I could not read these values.
An energy entity does not even exist, so the line
_uri: http://192.168.178.40:8123/api/states/sensor.stream_ultra_0150_power_battery_soc_
in the yaml above is questionable to me.

A soc entity does exist, but cannot be read:

[image]

So I guess we have to wait untiol EcoFlwo improves the API, or if these paramters acutially do already exit in the API, until someone figures out how to access them in the correct way.
I also tried many other device types insted of "STREAM_AC", just to see if others would work - but no...

Jelledb, 2025-09-17:
I just received my Ecoflow PowerOcean today and I requested a developer account at Ecoflow.

I will try and make this integration work, also without HomeAssistant as I don't use HA.

Balou1986, 2025-09-29:
Hello,

In my experiance there won´t be any respone from Ecoflow about the developer Account.
At some time after the registration it just will be activated and you can Log in and work with the dev Account.
Without a notification that it was activated...

I have tried it and in the end have used a Tapo P110 WiFi Socket and configured this as Meter for my small 800W PV-System (Balkonkraftwerk)...
This is not the best solution, but I can visuallize the Input from the 800W Ecoflow Stream Inverter.

Jelledb, 2025-10-01:
I opened a PR for the PowerOcean: #24012

Please note for other devices of Ecoflow you'd probably need specific implementations as their APIs are different.