evcc-io/evcc #17782: Push-Notifications: "vehicleTitle" not set on "connected" and "disconnected" events

LiamHD, 2024-12-18:
### Describe the bug

See this snippet from my config:
[image]

The push notifications on telegram shows the vehicle title correctly e.g. when charging starts but not when I connect or disconnect (it shows <no value>". As you can see I did use the same variable in my texts. This worked before even though I cannot tell the evcc version with which it still worked.

### Steps to reproduce

see above

### Configuration details

```yaml
see above
```

### Log details

```text
none
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.10

TobiasHuber1980, 2024-12-30:
erinnert mich ein wenig an https://github.com/evcc-io/evcc/issues/14685
irgendwie kommt das messaging bei connect und disconnect mit `${vehicleTitle}` nicht klar.

andig, 2025-01-06:
Duplicate of https://github.com/evcc-io/evcc/issues/7193

closed 2025-01-06 by andig