evcc-io/evcc #17787: EVCC lief gut, bis VW die Nutzungsbestimmungen angepaßt haben wollte.

DidiHRB1975, 2024-12-18:
### Describe the bug

Hallo,

EVCC lief bei mir unter Windows ein halbes Jahr gut.
Ich habe einen VW ID3 angebunden. Die Konfiguration lief soweit gut.
Dann wurde seit ein paar Wochen der ID3 nicht mehr angezeigt, sondern nur noch das Gastfahrzeug.
Ich habe hier im Forum gelesen, was zu uns ist. Also bin ich in die APP und auch auf die Seite von VW und haben die Nutzungsbesdingungen bestätigt.
Aber leider bekomme ich jetzt die Meldung: Fehler beim Starten. Überprüfe deine Konfiguration und starte neu - failed configuration loadpoint. not fouind VWID3
Wenn ich jedoch die Konfiguration teste, dann werden Kilometerstand, SOC und ähnliches aus dem Fahrzeug ausgelesen.
evcc -c evcc.yaml -l debug vehicle
C:\DD_Temp\EVCC>evcc -c evcc.yaml -l debug vehicle
[main  ] INFO 2024/12/18 14:28:42 evcc 0.131.11
[main  ] INFO 2024/12/18 14:28:42 using config file: evcc.yaml
[db    ] INFO 2024/12/18 14:28:42 using sqlite database: C:\Users\dirkd\.evcc\evcc.db
Soc:            72%
Capacity:       58.0kWh
Charge status:  A
Range:          212km
Odometer:       21754km
Finish time:    2024-12-17 15:26:00 +0100 CET
Climate active: false
Limit Soc:      80%

Ich hatte auch eine 2. Konfiguration in einem anderen Verzeichnis probiert, aber leider kommt da der gleiche Fehler.
Was kann man noch tun?

Vielen Dank schon einmal für Eure Hilfe!

Gruß, Dirk

### Steps to reproduce

1. Start EVCC-Seite im Browser
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 3116faf89bf1c426778d387af296e19f31f2560ca33cee211699834b7f0716b1

interval: 30s # control cycle interval

sponsortoken: 
xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: goodwe-wifi 
  usage: grid  
  uri: 192.168.178.36  
  name: grid1
- type: template
  template: goodwe-wifi 
  usage: pv  
  uri: 192.168.178.36  
  name: pv2
- type: template
  template: goodwe-wifi 
  usage: battery  
  uri: 192.168.178.36  
  name: battery3

chargers:
- type: template
  template: easee 
  user: XXX
  password: XXX
  charger: EH3M47JM  
  timeout: 30s  
  authorize: true  
  name: wallbox5

vehicles:
- type: template
  template: vw 
  title: VWID3  
  user: XXX
  password: XXX
  vin: XXX
  capacity: 58  
  timeout: 10s  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: VWID3 # Referenz auf Standardfahrzeug
  mode: pv

site:
  title: DD_Harenberg11
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.33 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh
```

### Log details

```text
[main ] INFO 2024/12/18 14:39:51 evcc 0.131.11
[main ] INFO 2024/12/18 14:39:51 using config file: c:\dd_temp\evcc\evcc.yaml
[db ] INFO 2024/12/18 14:39:51 using sqlite database: C:\WINDOWS\system32\config\systemprofile\.evcc\evcc.db
[main ] INFO 2024/12/18 14:39:51 listening at :7070
[main ] FATAL 2024/12/18 14:39:54 failed configuring loadpoints: failed configuring loadpoint: not found: VWID3
[main ] FATAL 2024/12/18 14:39:54 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Windows

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.11

StefanSchoof, 2024-12-18:
Am loadpoint muss der name und nicht der Titel eingefügt werden. Also bei dir `ev4` und nicht `VWID3`.

DidiHRB1975, 2024-12-18:
Hallo Stefan,
alles klar. Habe ich gleich probiert und es funktioniert!
Vielen Dank für die schnelle Hilfe!
Schöne Feiertage!
Dirk

andig, 2024-12-18:
Es wäre wunderbar, wenn Discussions bitte dort aufgemacht werden könnten. Hier sind wir bei den Issues. Danke.