evcc-io/evcc #17801: Ladeplan sollte SmartCostLimit nicht überschreiben

naltatis, 2024-12-19:
Das ist die Logik im code dazu:
https://github.com/evcc-io/evcc/blob/597eeeb4dcabcdc740bc4beb3db228da8a095f26/core/loadpoint.go#L1849

Ich denke die Implementierung war schon immer oder schon lange so, aber es fiel bei sporadischen Plänen nicht weiter auf.

Jetzt, wo aber quasi immer ein aktiver Plan besteht, ist smart cost charging defacto nicht mehr möglich. 

Mein Usecase ist, an bestimmten Tagen über den Planner morgens je nach Tag einen gewissen Ladestand sicherzustellen, wenn es günstig ist / wenig co2 Ausstoß verursacht, möchte ich aber dennoch weiter laden bis zum eingestellten Limit (analog zum normalen Überschussladen im Sommer).

@naltatis wollt ihr euch das mal anschauen? Denke das könnte ziemlich einfach geändert werden.

_Originally posted by @mfuchs1984 in https://github.com/evcc-io/evcc/discussions/17798#discussioncomment-11613813_

naltatis, 2024-12-19:
Das wäre eine Verhaltensänderung. Bislang haben wir bewusst das SmartCostLimit bei aktivem Plan ignoriert um das bessere Ergebnis des Planers zu garantieren. Mit den wiederholenden Plänen ergibt das aber in der Tat wenig Sinn. Daher würde ich das Verhalten komplett entkoppeln und das Limit immer(!) zur Anwendung bringen wenn es unterschritten wird.

mfuchs1984, 2024-12-19:
Bzgl. der Frage 
> Spannend wäre für mich inhaltlich noch die Frage ob wir die Bedingung ganz rausnehmen oder nur die Repeating Plans ignorieren. 

Ich denke die aktuelle Implementierung bringt nur einen Vorteil: bei einem Ziel zu einer bestimmten Zeit, könnte dieses je nach konfigurierter Preisgrenze ggf günstiger erreicht werden, als mit "günstigem Netzladen". 
Aber andersrum gilt natürlich dasselbe, denn wenn zwischendrin das Auto verwendet wird, können die Preise beim Wiederanstecken höher sein. Bei wiederkehrenden Plänen wird zweiteres meines Erachtens (zumindest bei mir) definitiv oft vorkommen.  
Bei einmaligen Plänen hängt es davon ab, wie sie verwendet werden, durch die wiederkehrenden brauche ich sie aber nur noch selten, bspw. vor einer Urlaubsfahrt. Deshalb spielt es da für nicht eigentlich keine echte Rolle.

Eine dritte Option wäre noch die Konfigurierbarkeit über ein toggle switch je Plan oder alternativ global bei der Konfiguration für günstiges Netzladen.

andig, 2024-12-19:
Ich denke weniger Magie und damit besser wäre es wenn SmartCost zuschlägt wenn der Preis erreicht ist. Planner sollte an der Stelle ignoriert werden.

mfuchs1984, 2024-12-19:
Ich denke wir sind d'accord :)

closed 2024-12-19 by andig