evcc-io/evcc #17804: Modbus Issue @ Q.HOME+ ESS HYB-G3-3P

kjohann7, 2024-12-19:
### Describe the bug

Die Hausbatterie-Einstellung "Verhindere Entladung im Schnell-Modus und bei geplantem Laden." lässt sich zwar aktivieren. Beim Laden des Autos im Schnell-Modus oder bei geplantem Laden wird aber trotzdem zuerst die Batterie gelehrt.
Die Priorisierung Fahrzeugladen/Hausbatterie hingegen funktioniert.

Konfiguration:
- EVCC aktuelle Version auf Raspberry
- Hybridwechselrichter: Q.HOME+ESS HYB-G3-3P

Fehlermeldungen:
- Trace:
[grid1 ] TRACE 2024/12/19 09:47:23 modbus: recv 00 4e 00 00 00 07 01 04 04 ff 53 ff ff
[grid1 ] TRACE 2024/12/19 09:47:23 modbus: send 00 4f 00 00 00 06 01 04 00 4a 00 02
[grid1 ] TRACE 2024/12/19 09:47:23 modbus: recv 00 4f 00 00 00 07 01 04 04 db 83 00 08
[battery1] TRACE 2024/12/19 09:47:23 modbus: send 00 50 00 00 00 06 01 04 00 16 00 01
[battery1] TRACE 2024/12/19 09:47:23 modbus: recv 00 50 00 00 00 05 01 04 02 00 00
[pv2 ] TRACE 2024/12/19 09:47:23 modbus: send 00 51 00 00 00 06 01 04 00 0a 00 01
[pv2 ] TRACE 2024/12/19 09:47:23 modbus: recv 00 51 00 00 00 05 01 04 02 00 51 

Es gibt noch eine Fehlermeldung, die auf einen Modbus-Fehler hinweißt.

Kennt jemand eine funktionierende Konfiguration?
Kann das Problem mit dem Modbus-Proxy gelöst werden?
Wenn ja, welche Einstellungen, werden benötigt:
Hier das Bsp von der EVCC-Seite:
modbusproxy:
  - port: 5021
    uri: 192.0.2.2:502
  - port: 5022
    device: /dev/ttyUSB0
    baudrate: 9600
    comset: "8N1"
  - port: 5023
    uri: 192.0.2.3:502
    rtu: true

Ist die uri die IP-Adresse des Geräts, auf dem EVCC läuft?
Wird als port der Modbus-Standardport 502 benötigt?
Muss man rtu auf true oder false setzen?
Müssen die PV-Komponenten, die ohne Proxy alle die gleiche IP-Adresse haben, mit unterschiedlichen Adressen angegeben werden?
meters:
- type: template
  template: solax 
  id: 1  
  host: 192.168.11.102  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solax 
  id: 1  
  host: 192.168.11.102  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: solax 
  id: 1  
  host: 192.168.11.102  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery1

Schon jetzt vielen Dank für Hilfe,
Klaus

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solax 
  id: 1  
  host: 192.168.11.102  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solax 
  id: 1  
  host: 192.168.11.102  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: solax 
  id: 1  
  host: 192.168.11.102  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery1

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.11.104  
  name: wallbox1
- type: template
  template: go-e-v3 
  host: 192.168.11.103  
  name: wallbox2

vehicles:
- name: i4
  type: template
  template: bmw
  title: i4  
  icon: car  
  user: name@provider
  password: mypassword
  vin: xxx
  capacity: 67
  hcaptcha: xxx
  phases: 3  
  cache: 5m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  region: EU  

loadpoints:
- title: go-e 11kW
  charger: wallbox1
  mode: pv
- title: go-e 22kW
  charger: wallbox2
  mode: pv

site:
  title: Carport
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery1

tariffs:
  currency: EUR
  grid: # static grid price
    type: fixed
    price: 0.31 # [currency]/kWh
  feedin: # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.071 # [currency]/kWh
```

### Log details

```text
see above
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.11

andig, 2024-12-19:
> Es gibt noch eine Fehlermeldung, die auf einen Modbus-Fehler hinweißt.

Ist die geheim? Hier ist keine erkennbar.

kjohann7, 2024-12-20:
Nicht geheim - ich musste sie nur in alten Unterlagen suchen...
die Fehlermeldung lautet:
"[site ] ERROR 2024/10/13 09:18:50 battery mode: modbus: exception '4' (server device failure), function '4' ".

Der Customer-Service von Q.Cells kann oder will sich dazu nicht äußern.
Eine Anfrage an Solax habe ich eben gestellt.
In einem Forum (https://control.com/forums/threads/modbus-exception-code-04.27550/) habe ich gefunden:

Error code 04 = Slave Device Failure
'An unrecoverable error occured while the server was attempting to perform the requested action'
This error code is not seen often. To analyse what the reason in your case might be, more information on your slave device is needed. One possibility is that you try to write into a write protected area of the slave (I have seen this reaction on a device). 

Auf http://www.simplymodbus.ca/exceptions.htm findet man
...
4. The request is received without an error, but cannot be processed by the slave for another reason.  The slave replies with an exception response.

Wenn ich das mit meinem ungesunden Halbwissen richtig verstehe ist EVCC hier der Master und die Solax-Batterie der Slave. D.h. aus meiner Sicht, um dem Problem auf die Spur zu kommen, müsste man dem Solax-Helpdesk durchgeben, wie der von EVCC gesandte Befehl genau lautet.

(Link zu einer Modbus-Dokumentation: https://modbus.org/docs/PI_MBUS_300.pdf)

andig, 2024-12-20:
Das ist im trace Log zu sehen.

kjohann7, 2024-12-20:
Hallo andig,
wie soll ich Ihre Antwort und das Schließen dieser Anfrage interpretieren?
Falls Sie wie ich nur ein Anwender sind, der versucht zu helfen, wo es geht, danke ich Ihnen für ihre Hilfe.
Falls Sie für EVCC arbeiten: Wollen Sie mir sagen, dass EVCC, wenn ich nicht in der Lage bin, durch diese technischen Details durchzusteigen, nicht die richtige Software für mich ist?
MIt freundlichen Grüßen
Klaus Johann

andig, 2024-12-20:
Das Gerät reagiert auf eine gültige Anfrage mit einem Modbusfehler. Warum das so ist kann nur der Hersteller des Gerätes sagen. Also bitte bei QCells Support anfragen. Und _welche_ Anfrage das ist sieht man im Logfile. Da Du uns von diesem Fehler aber kein Logfile gegeben hast musst Du sie selber raus suchen. Meine Glaskugel ist leider immer noch zur Wartung...

kjohann7, 2024-12-20:
[evcc-20241220-103629-debug.log](https://github.com/user-attachments/files/18208770/evcc-20241220-103629-debug.log)
Anbei das gewünscht Logfile. Ich glaube nicht, dass das ein Problem der Solaranlage ist.
Gerade hat die Sonnen geschienen, so dass die Hausbatterie ein bisschen geladen war, und ich die Situation rekonstruieren konnte:

1. Obwohl die Einstellung "Verhindere Entladung im Schnell-Modus und bei geplantem Laden.", aktiviert ist, wird zuerst die Hausbatterie entladen.
2. Es ist keine Modbus-Fehlermeldung erschienen. Die ursprüngliche Vermutung, dass es da einen Zusammenhang gibt, scheint falsch zu sein. Die Motbus-Fehlermeldung trat dann wohl auf, wenn die Batterie einen SoC unter MinSoc hatte - frag mich nicht, wie es dazu kommt.

Ich hoffe, das hilft weiter
Viele Grüße

andig, 2024-12-21:
> Obwohl die Einstellung "Verhindere Entladung im Schnell-Modus und bei geplantem Laden.", aktiviert ist, wird zuerst die Hausbatterie entladen.

Das ist dann ein Fehler im `solax` Template.

/cc @premultiply

premultiply, 2024-12-29:
Im Log der beschriebene Fehler nirgends zu sehen.

kjohann7, 2024-12-30:
Ich möchte darum bitten, dass das wieder geöffnet wird. Auch wenn der Fehler sich nicht im Log widerspiegelt, ist er doch vorhanden und führt zu einem relevanten Fehlverhalten. Dass kein Modbus-Fehler im Log zu finden ist, ist meines Erachtens zusätzlich ein starker Hinweis, dass der Fehler auf der EVCC- undn nicht auf der Solax-Seite zu suchen ist: Würden die Solax-Komponenten, wenn dort ein nicht verarbeitbarer Befehl ankäme, nicht eine Fehlermeldung zurückgeben?

Lasst mich bitte wissen, wenn bzw. wie ich die Fehlersuche unterstützen kann. 

Rein intressehalber: Ich weiß nicht, was heutige Programmierumgebungen leisten. Früher musste man explizit mit in den Code reinschreiben, was im Log protokolliert werden sollte. Kein Hinweis im Log war in keiner Weise ein Beleg für einen fehlerfreien Code sondern viel mehr, dass man z.B. einen Spezialfall nicht bedacht hatte. Ist das heute anders?

andig, 2024-12-30:
> Auch wenn der Fehler sich nicht im Log widerspiegelt, ist er doch vorhanden und führt zu einem relevanten Fehlverhalten. 

Es geht um 

> Es gibt noch eine Fehlermeldung, die auf einen Modbus-Fehler hinweißt.

Davon ist nichts zu sehen. Es braucht bitte eine aussagekräftige Fehlerbeschreibung mit allen angefragten Informationen, dann können wir die auch bearbeiten. Hier gibt es absolut nichts was wir tun könnten.

kjohann7, 2024-12-30:
Describe the bug

Die Hausbatterie-Einstellung "Verhindere Entladung im Schnell-Modus und bei geplantem Laden." lässt sich zwar aktivieren. Beim Laden des Autos im Schnell-Modus oder bei geplantem Laden wird aber trotzdem zuerst die Batterie gelehrt.
Die Priorisierung Fahrzeugladen/Hausbatterie hingegen funktioniert.

Konfiguration:

    EVCC aktuelle Version auf Raspberry
    Hybridwechselrichter: Q.HOME+ESS HYB-G3-3P

Log-Ausschnitt zür Modbus-Kommunikation:

    Trace:
    [grid1 ] TRACE 2024/12/19 09:47:23 modbus: recv 00 4e 00 00 00 07 01 04 04 ff 53 ff ff
    [grid1 ] TRACE 2024/12/19 09:47:23 modbus: send 00 4f 00 00 00 06 01 04 00 4a 00 02
    [grid1 ] TRACE 2024/12/19 09:47:23 modbus: recv 00 4f 00 00 00 07 01 04 04 db 83 00 08
    [battery1] TRACE 2024/12/19 09:47:23 modbus: send 00 50 00 00 00 06 01 04 00 16 00 01
    [battery1] TRACE 2024/12/19 09:47:23 modbus: recv 00 50 00 00 00 05 01 04 02 00 00
    [pv2 ] TRACE 2024/12/19 09:47:23 modbus: send 00 51 00 00 00 06 01 04 00 0a 00 01
    [pv2 ] TRACE 2024/12/19 09:47:23 modbus: recv 00 51 00 00 00 05 01 04 02 00 51

Komplettes Log-File angehängt an einem meiner Kommentare von letzter Woche (s.o.)

Fehlermeldungen: KEINE
Im Zusammenhang mit dem oben beschriebenen Fehlverhalten sehe ich keine Fehlermeldungen. (s. Log von letzter Woche)

Modbus-Fehlermeldungen treten auf, wenn die Batterie unter MinSoc ist: Siehe
[evcc-20241230-134111-debug.log](https://github.com/user-attachments/files/18274665/evcc-20241230-134111-debug.log)

Lasst mich bitte wissen, wenn ihr mehr Informationen braucht.

kjohann7, 2024-12-30:
Danke
mach ich, sobald die Sonne geschienen hat und in der Batterie wieder was drin ist

kjohann7, 2025-01-06:
Hier der was sich in der Commandline abgespielt hat:

kjohann@raspberrypi:~ $ sudo systemctl stop evcc
kjohann@raspberrypi:~ $ evcc meter --battery-mode hold
[main  ] INFO 2025/01/06 16:17:23 evcc 0.131.11
[main  ] INFO 2025/01/06 16:17:23 using config file: /home/kjohann/evcc.yaml
[db    ] INFO 2025/01/06 16:17:23 using sqlite database: /home/kjohann/.evcc/evcc.db
[main  ] FATAL 2025/01/06 16:17:24 set battery mode: modbus: exception '4' (server device failure), function '4'

Und hier Grafik zur Batteriekonfiguration, Grafik, die zeigt, dass die Batterie trotzdem zum Laden der PKWs genutzt wird: Einmal geplantes Laden, einmal Schnellladen, sowie das aktuelle Logfile.
[image]
[evcc-20250106-163500-debug.log](https://github.com/user-attachments/files/18321353/evcc-20250106-163500-debug.log)
[image]

Ich hoffe, das hilft.

andig, 2025-01-06:
Perfekt. Den Screenshot brauchts nicht, der Fehler ist nachvollziehbar.

/cc @premultiply

kjohann7, 2025-01-06:
HIer noch ein paar Fehlermeldung, die später aufgetaucht sind, vielleicht aber auch helfen:

[site ] ERROR 2025/01/06 17:17:10 battery mode: modbus: exception '2' (illegal data address), function '4'
[site ] ERROR 2025/01/06 17:17:29 battery mode: modbus: exception '4' (server device failure), function '4'
[site ] ERROR 2025/01/06 17:17:32 battery mode: modbus: exception '2' (illegal data address), function '4'

premultiply, 2025-01-06:
> [main ] INFO 2025/01/06 16:17:23 using config file: /home/kjohann/evcc.yaml
> [db ] INFO 2025/01/06 16:17:23 using sqlite database: /home/kjohann/.evcc/evcc.db

Hier wird eine andere Konfigdatei als im Log verwendet.

Mal bitte die Ausgaben (bei angehaltenem Dienst) von
`evcc dump --cfg -c /etc/evcc.yaml`
`evcc meter --log trace,db:error -c /etc/evcc.yaml`
`evcc meter --battery-mode hold --log trace,db:error -c /etc/evcc.yaml`
hier einfügen.

kjohann7, 2025-01-07:
Muss und kann ich etwas dafür tun, dass keine andere Konfigdatei als im Log verwendet wird? Evtl. die ganze Kiste mal neu aufsetzen?

Hier die gewünschten Ausgaben:

kjohann@raspberrypi:~ $ sudo systemctl stop evcc
kjohann@raspberrypi:~ $ evcc dump --cfg -c /etc/evcc.yaml
[main  ] INFO 2025/01/07 08:52:49 evcc 0.131.11
[main  ] INFO 2025/01/07 08:52:49 using config file: /etc/evcc.yaml
[db    ] INFO 2025/01/07 08:52:49 using sqlite database: /home/kjohann/.evcc/evcc.db
[bmw   ] DEBUG 2025/01/07 08:52:49 identity.Login - no database token found. Proceeding to login via user, password and captcha.
[main  ] ERROR 2025/01/07 08:52:49 creating vehicle i4 failed: cannot create vehicle type 'template': cannot create vehicle type 'bmw': Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method): unexpected status: 401 (Unauthorized)
[lp-1  ] DEBUG 2025/01/07 08:52:49 charge total import: 4569.843kWh
[lp-2  ] DEBUG 2025/01/07 08:52:50 charge total import: 4183.262kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solax 
  id: 1  
  host: 192.168.11.102  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solax 
  id: 1  
  host: 192.168.11.102  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: solax 
  id: 1  
  host: 192.168.11.102  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery1

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.11.104  
  name: wallbox1
- type: template
  template: go-e-v3 
[... 63 lines cut ...]
[battery1] TRACE 2025/01/07 08:53:56 modbus: send 00 03 00 00 00 06 01 04 00 16 00 01
[battery1] TRACE 2025/01/07 08:53:56 modbus: recv 00 03 00 00 00 05 01 04 02 00 00
[battery1] TRACE 2025/01/07 08:53:56 modbus: send 00 04 00 00 00 06 01 04 00 1c 00 01
[battery1] TRACE 2025/01/07 08:53:57 modbus: recv 00 04 00 00 00 05 01 04 02 00 09
Power:        -0W
Soc:          9%
Controllable: true

pv2
---
[pv2   ] TRACE 2025/01/07 08:53:57 modbus: send 00 05 00 00 00 06 01 04 00 0a 00 01
[pv2   ] TRACE 2025/01/07 08:53:57 modbus: recv 00 05 00 00 00 05 01 04 02 00 00
[pv2   ] TRACE 2025/01/07 08:53:57 modbus: send 00 06 00 00 00 06 01 04 00 0b 00 01
[pv2   ] TRACE 2025/01/07 08:53:57 modbus: recv 00 06 00 00 00 05 01 04 02 00 2e
[pv2   ] TRACE 2025/01/07 08:53:57 modbus: send 00 07 00 00 00 06 01 04 00 94 00 02
[pv2   ] TRACE 2025/01/07 08:53:57 modbus: recv 00 07 00 00 00 07 01 04 04 b6 bd 00 02
Power:  46W
Energy: 17785.3kWh

kjohann@raspberrypi:~ $ evcc meter --battery-mode hold --log trace,db:error -c /etc/evcc.yaml
[main  ] INFO 2025/01/07 08:54:20 evcc 0.131.11
[main  ] INFO 2025/01/07 08:54:20 using config file: /etc/evcc.yaml
[battery1] TRACE 2025/01/07 08:54:20 modbus: send 00 01 00 00 00 06 01 06 00 20 00 00
[battery1] TRACE 2025/01/07 08:54:21 modbus: recv 00 01 00 00 00 07 01 04 04 b6 bd 00 02
[main  ] FATAL 2025/01/07 08:54:21 set battery mode: modbus: exception '4' (server device failure), function '4'

premultiply, 2025-01-07:
Thank you.
Could you please try if the error is the same for 
```
evcc meter --battery-mode normal --log trace,db:error -c /etc/evcc.yaml
```
and
```
evcc meter --battery-mode charge --log trace,db:error -c /etc/evcc.yaml
```
?

Maybe it is just an issue in case of empty battery or inverter in standby mode?

kjohann7, 2025-01-09:
Hallo Andig und Premultply,
vielleicht hilft das um der Lösung näher zu kommen: https://github.com/evcc-io/evcc/discussions/4016
Viele Grüße
Klaus

kjohann7, 2025-01-10:
Anbei 2 Fotos zur Modbus-Konfiguration der Solaranlage. Vielleicht hilft das ja weiter.
Oder ist der Fehler identifiziert und muss nur noch korrigiert werden, so dass ich nichts mehr liefern muss?

[image]
[image]

andig, 2025-01-10:
> vielleicht hilft das um der Lösung näher zu kommen

Gibts da neue Erkenntnisse? Passen die Register nicht? Uns liegen keine neuen Informationen vor.

xLennuk, 2025-01-10:
Moin, ich habe tatsächlich das gleiche Problem mit gleichem Wechselrichter.
Wechselrichter: Q.HOME+ ESS HYB-G3 10kW 3p
Stromspeicher: Q.SAVE BAT-G3

Das ist meine evcc.yaml-Datei:
```
sponsortoken: X

plant: X

network:
  port: 7070

log: debug

interval: 30s

meters:
  - name: my_grid
    type: template
    template: solax
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.47
    port: 502
    
  - name: my_pv
    type: template
    template: solax
    usage: pv
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.47
    port: 502

  - name: my_battery
    type: template
    template: solax
    usage: battery
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.47
    port: 502

chargers:
  - name: my_charger
    type: template
    template: go-e-v3
    host: 192.168.178.25

site:
  title: Zuhause
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery
  residualPower: 200

loadpoints:
  - title: Garage # display name for UI
    charger: my_charger # charger
    mode: now
    
tariffs:
  grid:
    type: template
    template: tibber
    token: X
    formula: math.Max((price + charges) * (1 + tax), 0.0) # Individuelle Formel zur Berechnung des Preises (optional)
  co2:
    type: template
    template: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: XXXXX # Deine Postleitzahl
```

Beim Laden im "Schnell-Modus" und gleichzeitig aktiviertem "Verhindere Entladung im Schnell-Modus und bei geplantem Laden.", kam folgendes im trace-log heraus. Ist das Verhindern der Entladung deaktiviert, kommen keine Fehlermeldungen. Vielleicht hilft es ja, der Lösung des Problems näher zu kommen?

```
[site  ] DEBUG 2025/01/08 13:53:47 ----
[lp-1  ] DEBUG 2025/01/08 13:53:47 charge power: 10580W
[lp-1  ] DEBUG 2025/01/08 13:53:47 charge currents: [14.8 14.9 14.9]A
[my_battery] TRACE 2025/01/08 13:53:47 modbus: send 00 33 00 00 00 06 01 06 00 20 00 00
[site  ] ERROR 2025/01/08 13:53:57 battery mode: read tcp 192.168.178.46:59646->192.168.178.47:502: i/o timeout
[my_grid] TRACE 2025/01/08 13:53:57 modbus: send 00 34 00 00 00 06 01 04 00 46 00 02
[site  ] DEBUG 2025/01/08 13:53:57 set battery discharge control: false
[my_grid] TRACE 2025/01/08 13:53:57 modbus: recv 00 34 00 00 00 07 01 04 04 e3 2f ff ff
[my_grid] TRACE 2025/01/08 13:53:57 modbus: send 00 35 00 00 00 06 01 04 00 4a 00 02
[my_grid] TRACE 2025/01/08 13:53:57 modbus: recv 00 35 00 00 00 07 01 04 04 ff 4d 00 12
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: send 00 36 00 00 00 06 01 04 00 0a 00 01
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: recv 00 36 00 00 00 05 01 04 02 06 ea
[my_battery] TRACE 2025/01/08 13:53:57 modbus: send 00 37 00 00 00 06 01 04 00 16 00 01
[my_battery] TRACE 2025/01/08 13:53:57 modbus: recv 00 37 00 00 00 05 01 04 02 00 00
[my_grid] TRACE 2025/01/08 13:53:57 modbus: send 00 38 00 00 00 06 01 04 00 46 00 02
[my_grid] TRACE 2025/01/08 13:53:57 modbus: recv 00 38 00 00 00 07 01 04 04 e3 2f ff ff
[site  ] DEBUG 2025/01/08 13:53:57 grid power: 7377W
[my_battery] TRACE 2025/01/08 13:53:57 modbus: send 00 39 00 00 00 06 01 04 00 16 00 01
[my_battery] TRACE 2025/01/08 13:53:57 modbus: recv 00 39 00 00 00 05 01 04 02 00 00
[site  ] DEBUG 2025/01/08 13:53:57 battery 1 power: -0W
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: send 00 3a 00 00 00 06 01 04 00 0a 00 01
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: recv 00 3a 00 00 00 05 01 04 02 06 ea
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: send 00 3b 00 00 00 06 01 04 00 0b 00 01
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: recv 00 3b 00 00 00 05 01 04 02 08 d2
[my_battery] TRACE 2025/01/08 13:53:57 modbus: send 00 3c 00 00 00 06 01 04 00 1c 00 01
[my_battery] TRACE 2025/01/08 13:53:57 modbus: recv 00 3c 00 00 00 05 01 04 02 00 0a
[my_grid] TRACE 2025/01/08 13:53:57 modbus: send 00 3d 00 00 00 06 01 04 00 4a 00 02
[my_grid] TRACE 2025/01/08 13:53:57 modbus: recv 00 3d 00 00 00 07 01 04 04 ff 4d 00 12
[my_battery] TRACE 2025/01/08 13:53:57 modbus: send 00 3e 00 00 00 06 01 04 00 1c 00 01
[my_battery] TRACE 2025/01/08 13:53:57 modbus: recv 00 3e 00 00 00 05 01 04 02 00 0a
[site  ] DEBUG 2025/01/08 13:53:57 battery 1 soc: 10%
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: send 00 3f 00 00 00 06 01 04 00 0b 00 01
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: recv 00 3f 00 00 00 05 01 04 02 08 d2
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: send 00 40 00 00 00 06 01 04 00 94 00 02
[site  ] DEBUG 2025/01/08 13:53:57 pv 1 power: 4028W
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: recv 00 40 00 00 00 07 01 04 04 73 89 00 03
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: send 00 41 00 00 00 06 01 04 00 94 00 02
[my_pv ] TRACE 2025/01/08 13:53:57 modbus: recv 00 41 00 00 00 07 01 04 04 73 89 00 03
[site  ] DEBUG 2025/01/08 13:53:57 site power: 7577W
[lp-1  ] DEBUG 2025/01/08 13:53:57 detected active phases: 3p
[go-e  ] TRACE 2025/01/08 13:53:57 GET http://192.168.178.25/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[go-e  ] TRACE 2025/01/08 13:53:57 {"alw":true,"eto":XXXXXXX,"car":X,"wh":XXXXX.XXXXX,"trx":null,"nrg":[XXX,XXX,XXX,X,XX.XXXXXXXX,XX.XXXXXXXX,XX,XXXX,XXXX,XXXX,X,XXXXX,XXX,XXX,XXX,X],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}
[lp-1  ] DEBUG 2025/01/08 13:53:57 charge total import: 3829.990kWh
[lp-1  ] DEBUG 2025/01/08 13:53:57 charger status: C
[site  ] DEBUG 2025/01/08 13:54:17 ----
```

andig, 2025-01-21:
Keine neuen Erkenntnisse. Da anscheinend niemand weiss, _wie_ battery hold zu implemnetieren ist können wir an der Stelle zu machen und/oder die aktuelle Umsetzung wieder ausbauen. Andere Lösungen gibt es nicht.

premultiply, 2025-01-22:
https://github.com/evcc-io/evcc/discussions/4016#discussioncomment-11813916

Scheint auch bei manchen zu funktionieren.

Ich habe hier ansonsten leider keinen Ansatz ohne weiteren externen Input.

Daher mache ich hier erstmal zu bis es weitere sachdienliche Erkenntnisse zur Ursache gibt.

closed 2025-01-22 by premultiply

kjohann7, 2025-01-28:
Lässt sich aus https://github.com/evcc-io/evcc/discussions/4016 überhaupt nichts ableiten?
Heißt das, dass nicht mehr weiter versucht wird, diesen Bug zu beheben?