evcc-io/evcc #17806: Entladesperre Hausbatterie solange dynamischer Tarif kleiner als Schwellwert ist

chris-ka1, 2024-12-19:
ich nutze die Funktion Netzladen der Hausbatterie. Dies funktioniert einwandfrei und ist im Winter sehr hilfreich. 
Ich setzte eine untere Schwelle und lade die Hausbatterie wenn der Tarif unter dem Schwellwert liegt.

Nachdem der Tarif über dem Schwellwert liegt fängt die Batterie sofort an ihre Energie wieder abzugeben. 

Optimal wäre es hier zusätzlich eine obere Schwelle zu definieren um die geladene Energie erst dann bereitzustellen wenn der dynamische Tarif oberhalb einer Schwelle liegt. 

Beispielszenario Dunkelflaute: (Ist so passiert am 12.12.2024)
in der Nacht geht der dynamische Tarif für 4h auf 30ct/kWh runter. Für den nächsten Tag ist ein Tarif von über 1€/kWh ab 15Uhr angekündigt.

Ich würde nun gerne ein Intervall definieren mit einer unteren und oberen Schwelle. Liegt der Tarif under der unteren Schwelle wird die Hausbatterie über das Netz geladen (das funktioniert aktuell bereits bestens). Solange der Tarif zwischen der unteren und oberen Schwelle liegt ist eine Entladesperre der Hausbatterie aktiv (das wäre die neue Funktion) die Batterie gibt dann erst ihre Energie ab wenn die obere Schwelle überschritten ist.

Dadurch kann sichergestellt werden, dass die Batterie noch Ladung hat wenn der Peak kommt.

andig, 2024-12-19:
Ich bin nicht sicher ob es diese Logik braucht da die Ansteuerung der Schwelle ohnehin dynamisch passieren müsste. Stattdessen- und das würde viel mehr use cases ermöglichen- könnten wir auch den Battery Mode (hold/charge) direkt von extern ansteuerbar machen?

/cc @naltatis

naltatis, 2024-12-19:
> könnten wir auch den Battery Mode (hold/charge) direkt von extern ansteuerbar machen?

Damit wäre ich vorsichtig. Würde ein einmaliges Setzen von außen dann jegliche Logik in evcc blockieren? Starten einer Schnellladung. Unterschreiten eines Preislimits? Würdest du einen "manuellen Modus" einführen bei dem unserer Automatismen alle aus sind?

Ich verstehe den Usecase, aber für mich gehört das in die Diskussion https://github.com/evcc-io/evcc/discussions/16364 und die konkrete Lösung wäre für mich abhängig davon wie wir uns für den nächsten Schritt "Netzladen der Batterie" Entscheiden.

VolkerK62, 2024-12-19:
@chris-ka1  unabhängig einer internen Lösung, gibt es jetzt schon einen workaround
https://github.com/evcc-io/evcc/wiki/aaa-Lifehacks#entladung-eines-steuerbaren-hausspeicher-preisgesteuert-oder-manuell-sperren

andig, 2024-12-19:
> Würdest du einen "manuellen Modus" einführen bei dem unserer Automatismen alle aus sind?

Ja. Das wäre Wiederverwendung der Battery-APIs. Alternativ: der (externe) Modus bleibt einmalig so lange stehen bis evcc eine Modusänderung durchführen möchte (Preisschwelle über/unterschritten, Fharzeugladung gestartet/beendet, Neustart).

Es gab schon mehrere Themen die sich über ein API lösen liessen.

naltatis, 2024-12-30:
Aus meiner Sicht können wir gerne eine "manuelle Steuerung" einführen. Ich würd die dann aber von unseren jetztigen Regeln/Automatismen (Preislimit, Schnelladen) trennen. Sonst können wir das in der UI nicht sauber kommunizieren.

Wir könnten `/batterymode/[auto|charge|hold]` machen. Default wäre `auto` (wie jetzt). Wenn `charge` oder `hold` gesetzt ist übersteuert das die Preislimit und Schnellade-Regeln. In der UI würde ich an den entsprechenden Stellen dann einen Hinweis einblenden.

andig, 2025-01-01:
> Das ist unproblematisch für die Schnellladung ("now"), da die überschüssige Leistung meist verwendet werden wird. Für die "Wintersperre" bei günstigem Netzstrom, bzw. um die gespeicherte Energie der Batterie zeitverzögert wirtschaftlich zu nutzen, ist es gelegentlich von Nachteil. 

Wenn das eine Einschränkung der Möglichkeiten des Geräts ist dann bleibt als Alternative nur die Entladesperre nicht zu verwenden.

> Vielleicht kann man "auto" umgehen, indem man die externen Vorgabe zeitlich befristet,

@naltatis hatte bereits vorgeschlagen, entweder extern gesteuert oder auto. Das ist klar und eindeutig.

softservicegit, 2025-01-08:
interessant wäre es den "Preis" des Stroms im Speicher zu berücksichtigen.
Wenn der Speicher von der Sonne gefüllt wird, dann kostes es nur die EInspeisevergütung. Wenn der Strom aus dem Netz kommt, dann halt mehr.
Dazu dann ein konfigurierbarer Aufschlag von z.b. 20 % für die Verluste und evcc könnte die Batteroe frei geben wenn der Strom in dem Akku billiger als der aktuelle Tarif ist.
Das wäre das optimum für mich.

naltatis, 2025-01-08:
> interessant wäre es den "Preis" des Stroms im Speicher zu berücksichtigen.

Ja, müssen wir auf jeden Fall adressieren. Jetzt von Batterie aus dem Netz laden ein häufigerer Anwendungsfall ist. Würde ich aber nicht als Teil von diesem Issue sehen. Ist eher ein eigener Themenkomplex.

closed 2025-01-20 by github-actions[bot]