evcc-io/evcc #17817: enphase: pv and power empty

othi, 2024-12-19:
### Describe the bug

My Enphase Envoy API returns the following at the URL `http://{{ .host }}/production.json`:
`{"production":[{"type":"inverters","activeCount":22,"readingTime":1734616504,"wNow":207,"whLifetime":6787031}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}`

The jq query as defined here https://github.com/evcc-io/evcc/blob/e33a68255e95cd80418d62e6bf399c3cfae1a468/templates/definition/meter/enphase.yaml#L44 does not return anything.

`(.production[] | select(.measurementType == "production").wNow) // (.production[] | select(.type == "inverters").wNow)` does work for me but I don't have an instance that returns `measurementType` so I can't be sure if the jq is correct.

The same is true for https://github.com/evcc-io/evcc/blob/e33a68255e95cd80418d62e6bf399c3cfae1a468/templates/definition/meter/enphase.yaml#L54 replacing `wNow` with `whLifetime`

### Steps to reproduce

1. Configure an Enphase meter in PV mode
2. Start evcc

### Configuration details

```yaml
- name: enphase
    type: template
    template: enphase
    usage: pv
    host: 192.168.xx.xx # IP-Adresse oder Hostname
    token: eyJraW...
```

### Log details

```text
[site ] ERROR 2024/12/19 14:46:02 pv 1 power: jq: empty result
[site ] DEBUG 2024/12/19 14:46:02 pv power: 0W
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.12

ivoks, 2024-12-20:
I see. Assumption was that the key are universal, but it seems that they are not. We need to check if they exist before we use them. I'll try to work on this today. @othi does production.json?details=1 return different set of keys? I see you are using http, and not https, so you have an older firmware too (not using tokens for auth)?

othi, 2024-12-20:
The `enphase`  template uses http itself:
https://github.com/evcc-io/evcc/blob/59025071b96edc1dc64d5de2338069cc17f07c1f/templates/definition/meter/enphase.yaml#L37
but it returns a 301 to the https URL anyway.

My gateway is on firmware 8.2.4264, which seems to be fairly recent (november 2024), and I do use a token.

As for the output of `production.json?details=1`:

```
# curl -H "Authorization: Bearer eyJra..." https://192.168.180.125/production.json?details=1
{"production":[{"type":"inverters","activeCount":22,"readingTime":1734687932,"wNow":442,"whLifetime":6788138}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}
```

ivoks, 2024-12-26:
Hey there

I believe I have an answer for this. I've tested with your production.json as well as mine and from few other folks and it gives the right answer:

jq 'if (.production | length) > 1 and (.production[] | select(.measurementType == "production").activeCount >= 1) then .production[] | select(.measurementType == "production").wNow else .production[] | select(.type == "inverters").wNow end'

jq 'if (.production | length) > 1 and (.production[] | select(.measurementType == "production").activeCount >= 1) then .production[] | select(.measurementType == "production").whLifetime else .production[] | select(.type == "inverters").whLifetime end'

If there are more than 1 member of production array, it checks if production's activeCount is, basically, not null (which means it's enabled). If all that is true, it returns the value from CT's wNow reading. If any other case it just returns inverter's wNow.

I'll prepare a patch for this...

PR #17948 (2024-12-29): Enphase: support for sparse production.json
In some cases production.json does not list CT meters, not even as a disabled resource. To support that use case, we need to check the length of array before iterating on, potentially non-existing, keys.

Fix #17817
closed 2024-12-29 by andig