evcc-io/evcc #17830: wirtschaftlich optimiertes Laden: PV-Leistung mit zusätzlichem Netzbezug voll ausnutzen

tipuraneo, 2024-12-20:
**Problem**
Der Modus "PV" arbeitet autarkieoptimiert. D. h., er setzt den max. Ladestrom so, dass möglichst viel PV-Leistung ohne Netzbezug geladen wird. Da fast alle Wallboxen nur in ganzen Ampere-Schritten regeln können und so gut wie nie exakt dieser Strom als Überschuss zur Verfügung steht, wird zwangsläufig ins Netz eingespeist. In der dunklen Jahreszeit reicht der PV-Überschuss oft nicht aus, um Autos komplett mit PV zu laden. D. h., früher oder später ist eh eine Ladung aus dem Netz notwendig. Gerade an sonnigen Tagen, an denen zwar viel Sonne scheint, aber nur über eine kurze Zeit, kommen mit PV doch ein paar kWh zusammen, die erst für wenig Geld ins Netz eingespeist und anschließend teurer wieder bezogen werden müssen.

_Beispiel 1: einphasiges Laden_
2.250 W PV-Überschuss ≙ 9,8 A (1p)
EVCC wird die nächst kleinere Stufe, 9 A bei 1p wählen, um Bezug zu vermeiden, mit 2.070 W laden und 180 W einspeisen.

_Beispiel 2: dreiphasiges Laden_
6.850 W PV Überschuss ≙ 9,9 A (3p)
EVCC wird die nächst kleinere Stufe, 9 A bei 3p wählen, um Bezug zu vermeiden, mit 6.210 W laden und 640 W einspeisen.

Wenn man davon ausgeht, dass die PV-Überschussleistungen zwischen zwei Ampere-Stufen gleichverteilt ist, ergibt sich eine durchschnittliche Einspeisung bei

- 1p von 1x 1 A * 230 V / 2 = 115 W
- 3p von 3x 1 A * 230 V / 2 = 345 W

**Lösung**
Da Netzbezug fast immer teurer ist als Eigenverbrauch, wäre es unter wirtschaftlichen Aspekten sinnvoller, PV-Überschuss komplett zu nutzen und bis zur nächsten Ampere-Stufe mit Netzstrom aufzufüllen. Dies wäre ein neuer Modus nach dem Prinzip "Min+PV" im "PV"-Modus. Also mindestens PV-Überschluss laden, aber nicht mehr als 230 W (1p) bzw. 690 W (3p) aus dem Netz. Dies wäre ein wirtschaftlich optimierter Modus. Der Modus könnte als "Min+PV+Netz" und "PV+Netz" in die GUI integriert oder als Konfigurationsvariable "operation_mode=[max-self-sufficientcy|min-feed-in]".

**Alternativen**
Es gibt zwar den Parameter [residualPower](https://docs.evcc.io/docs/reference/configuration/site#residualpower), der aber immer und nicht nur bei aktiven Ladevorgängen wirkt und nicht zwischen 1p und 3p unterscheidet. D. h., damit wäre nur ein fixes Offset möglich. Zudem lässt sich dieser Wert nicht vom Benutzer über die GUI ändern.

**Grafische Darstellung**
[image]

andig, 2024-12-20:
Klingt wie das Problem zur Lösung aus https://github.com/evcc-io/evcc/pull/16507 und Duplikat zu https://github.com/evcc-io/evcc/issues/14578?

tipuraneo, 2024-12-20:
Nein, die genannten Requests beziehen sich auf die Ein-/Ausschalthysterese. Mein Vorschlag bezieht sich jedoch auf das Regelverhalten während eines aktiven Ladevorgangs. Aktuell regelt evcc so, dass kein Strom aus dem Netz bezogen wird. Ich schlage jedoch hiermit vor, während eines Ladevorgangs nichts einzuspeisen und lieber minimal Strom aus dem Netz zu beziehen und die nächst höhere Amperestufe zu wählen.

andig, 2024-12-20:
Das ist _exakt_ https://github.com/evcc-io/evcc/issues/14578 "do not feed into grid".

closed 2024-12-20 by andig

tipuraneo, 2024-12-20:
Ja, der Gedanke in https://github.com/evcc-io/evcc/issues/14578 ähnelt meinem Vorschlag. Jedoch ist für mich schwer verständlich und technisch zu unspezifisch, da nicht so recht hervorgeht. wie genau evcc nach dem Start des Ladevorgangs die Leistung regeln soll. Insofern ergänzen sich beide. Im https://github.com/evcc-io/evcc/discussions/3530 werden auch schon weitere Modi (z. B. Smart) diskutiert. Ich werde diesen PR dort verlinken.

capi, 2024-12-20:
Habe im November mein erstes E-Auto erhalten, entsprechend bis jetzt nicht so viel Sonne gehabt. Das bisschen, dass ich habe, konnte ich aber mit folgender Konfiguration ganz gut ausnutzen:
```
loadpoints:
- title: Carport
  charger: go-e-gemini-flex
  vehicle: ev-dacia-spring-65-2024
  enable:
    threshold: -800
    delay: 1m
  disable:
    threshold: 500
    delay: 15m
  mode: pv
```

Aus meiner Sicht ist bei dem vorgeschlagenen Modus das Problem, ab wo du wirklich anfängst zu laden? Weil von 0 auf 6A (1.3kW) ist halt gleich mal ein Sprung, und du würdest also derzeit fast alles mit 6A aufladen, unabhängig was Sonne da ist?

SolarPowerEV, 2024-12-21:
Nur zur Info: der Wert "Residualleistung" lässt sich bereits in der GUI anpassen. Zum Glück sogar - anders als das dort ebenfalls einstellbare Interval - ohne den Dienst neu starten zu müssen:
Konfiguration → Regelverhalten → Residualleistung 

Das Problem mit 1p vs. 3p löst man damit natürlich nicht, aber zumindest kann man theoretisch on the fly den Residualwert während des aktiven Ladevorgangs noch nachziehen.