evcc-io/evcc #17835: Sonnenbatterie Netzladen funktioniert nicht

K4ba, 2024-12-20:
### Describe the bug

Sonnenbatterie Netzladen funktioniert nicht bzw. die Funktion wird in der Ui nicht angezeigt.

Ich habe in der config token, capacity und chargepower bei der batterie definiert.

Batterie aus der config entfernen und via Konfiguration in der Ui hinzufgen hilft auch nicht weiter (siehe: https://github.com/evcc-io/evcc/discussions/16503)

Ich habe ebenfalls probiert: evcc migrate --reset  bevor ich in der config etwas lösche/hinzufüge um da mögliche "überschreibungen" zu verhindern.

Ist das ein Bug oder habe ich was falsch?

Cheers
Kaba

### Steps to reproduce

1. write in config
2. delete in config
3. restart/stop/start evcc
4. delete database

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xxxxx

interval: 5s # control cycle interval

sponsortoken: xxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

tariffs:
  currency: EUR
  grid:    # static grid price
    type: fixed
    price: 0.347 # [currency]/kWh
  feedin:  # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: xxxx  
  port: xxx
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: xxx  
  port: xxxx
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: xxxx  
  port: xxx
  name: battery3
  token: xxxxxxx
  capacity: 10 
  chargepower: 4600 
  
chargers:
- type: template
  template: easee 
  user: xxxxx
  password: xxxxxxxx 
  charger: xxxxx
  name: wallbox5

vehicles:
- type: template
  template: id 
  title: ID4  
  user: xxxxxx  
  password: xxxxx
  vin: xxxx
  capacity: 77  
  phases: 3 
  icon: car  
  cache: 15m  
  mode: PV  
  # minSoc: 20  
  # targetSoc: 80  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev4

loadpoints:
- title: Tiefgarage
  charger: wallbox5
  vehicle: ev4
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  enable:       
   threshold: 0  
   delay: 1m
  disable:
   threshold: 0 
   delay: 30s
  soc:
   poll:
    mode: charging
    interval: 30m
   estimate: true

site:
  title: Home
  meters:
    grid: grid1
    pv: pv2
    battery: battery3
  residualPower: 50
```

### Log details

```text
**
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.12

VolkerK62, 2024-12-20:
Doku gibt die Antwort
> Du hast einen **dynamischen** Stromtarif und möchtest deinen Hausspeicher nachts mit günstigem Strom laden
Wenn dein Wechselrichter diese Funktion unterstützt und du **experimentelle Funktionen** aktiviert hast, erscheint im "Hausbatterie" Dialog der Reiter "Netzladen".

https://docs.evcc.io/docs/features/battery#netzladen

Um das bei einem fixen Stromtarif zu nutzen, braucht es einen "Trick"
https://github.com/evcc-io/evcc/wiki/aaa-Lifehacks#steuerbaren-hausspeicher-ohne-dynamischen-tarif-aus-dem-netz-laden

andig, 2024-12-20:
@naltatis das wäre ein weiterer potenzieller UC für ein Netzladen-API

closed 2024-12-20 by andig

naltatis, 2024-12-21:
Mir ist der Use Case noch nicht klar. Wieso möchte man das denn machen, wenn man keinen dynamischen Stromtarif hat?

K4ba, 2024-12-21:
Danke!

UC ohne dynamischen Tarif ist bsp. zur Batteriepflege im Winter.

andig, 2024-12-21:
Das klingt in der Tat sehr esotherisch- denn das mach sie ja selbst. Netzdienlichkeit wäre m.E. Ein UC.

K4ba, 2024-12-21:
Ich denke die Batterie auch Winter auf einem Level 20%+ zu halten ist sicherlich gesund und erhöht die Lebensdauer, die automatische Barrierepflege springt bei mir in den PV-Unterversorungsmonaten i.d.R. z.B. nur 2-3 mal an.

Zur Netzstabilität könnte man auf jeden Fall über eine Integration in EVCC nachdenken ggf. gekoppelt an die aktuelle Netzfrequenz (https://www.netzfrequenzmessung.de) oder EE Produktion. So als Idee.

Viele Grüße 
Kaba

andig, 2024-12-21:
> Ich denke die Batterie auch Winter auf einem Level 20%+ zu halten ist sicherlich gesund und erhöht die Lebensdauer, die automatische Barrierepflege springt bei mir in den PV-Unterversorungsmonaten i.d.R. z.B. nur 2-3 mal an.

Das mag sein, hat aber nix mit evcc zu tun. Wenns das braucht muss sich das BMS drum kümmern!

K4ba, 2024-12-21:
😊 was ein BMS „kann/könnte“ und „muss“ ist ja oft eine Diskrepanz aber für meinen Fall bin ich mit der Funktion Netzladen zufrieden.

Danke und schöne Feiertage.