evcc-io/evcc #17836: Probleme OCPP mit Heidelberg Amperfied connect.business

Saueee, 2024-12-20:
### Describe the bug

Beim Ändern des Lademodus (Aus/PV/PV+Min/Schnell) gibt die Wallbox ein "Not supported" zurück.
Anschließend passiert nicht. Es ist also nicht möglich das Laden zu Stoppen oder den Modus zu Ändern.

Alle Anderen OCPP Abfragen funktionieren ohne Probleme.

(Ich habe auch das Template ocpp-abb getestet da es sich evtl. um das gleiche Problem (stacklevel 0) handelt, aber ohne Erfolg)

### Steps to reproduce

1. Amperfied per OCPP verbinden
2. Auto anschließen und Laden
3. Lademodus wechseln

### Configuration details

```yaml
sponsortoken: XXX
interval: 30s
log: debug
site:
  title: XXX
  meters:
    grid: hager
    pv:
      - pv-anlage

loadpoints:
  - title: Wallbox 1
    charger: wb1
    mode: off
    enable:
      threshold: -4000
      delay: 2m
    disable:
      delay: 20m
      threshold: 1000
  - title: Wallbox 2
    charger: wb2
    mode: off
    enable:
      threshold: -4000
      delay: 2m
    disable:
      delay: 20m
      threshold: 1000
  - title: Wallbox 3
    charger: wb3
    mode: off
    enable:
      threshold: -4000
      delay: 2m
    disable:
      delay: 20m
      threshold: 1000
  - title: Wallbox 4
    charger: wb4
    mode: off
    enable:
      threshold: -4000
      delay: 2m
    disable:
      delay: 20m
      threshold: 1000
meters:
  - name: pv-anlage
    type: custom
    power:
      source: modbus
      id: 1
      uri: 10.10.10.10:502
      register:
        address: 19026
        type: input
        decode: float32
      scale: -1
      #rtu: true
      timeout: 5s
  - name: hager
    type: custom
    power:
      source: modbus
      id: 255
      uri: 10.10.10.11:502
      register:
        address: 50536
        type: holding
        decode: int32
      scale: 10
      timeout: 5s

chargers:
  - name: wb1
    type: template
    template: ocpp
    stationid: wb1
    meterinterval: 10s
    connecttimeout: 5m
  - name: wb2
    type: template
    template: ocpp
    stationid: wb2
    meterinterval: 10s
    connecttimeout: 5m
  - name: wb3
    type: template
    template: ocpp
    stationid: wb3
    meterinterval: 10s
    connecttimeout: 5m
  - name: wb4
    type: template
    template: ocpp
    stationid: wb4
    meterinterval: 10s
    connecttimeout: 5m

vehicles:
  - name: testauto
    title: Testauto
    type: template
    template: offline

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.29
```

### Log details

```text
[lp-3 ] DEBUG 2024/12/20 16:57:46 set charge mode: now
[lp-3 ] DEBUG 2024/12/20 16:57:46 charge power: 0W
[lp-3 ] DEBUG 2024/12/20 16:57:46 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2024/12/20 16:57:46 charge voltages: [228 0 0]V
[lp-3 ] DEBUG 2024/12/20 16:57:46 detected connected phases: 1p
[lp-3 ] DEBUG 2024/12/20 16:57:46 charge total import: 458.122kWh
[lp-3 ] DEBUG 2024/12/20 16:57:46 charger status: B
[lp-3 ] DEBUG 2024/12/20 16:57:46 vehicle soc: 0%
[ocpp ] TRACE 2024/12/20 16:57:46 send wb3: [2,"2002931585","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":0,"stackLevel":2,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-12-20T15:56:46Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16}]}}}]
[ocpp ] TRACE 2024/12/20 16:57:46 recv wb3: [3,"2002931585", { "status": "NotSupported" }]
[lp-3 ] ERROR 2024/12/20 16:57:46 max charge current 16A: set charging profile: NotSupported
[ocpp ] TRACE 2024/12/20 16:57:46 send wb3: [2,"1059617336","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/12/20 16:57:46 recv wb3: [3,"1059617336", { "status": "Accepted" }]
[ocpp ] TRACE 2024/12/20 16:57:46 recv wb3: [2,"251836754","MeterValues", { "meterValue": [ { "timestamp": "2024-12-20T15:57:46Z", "sampledValue": [ { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Power.Active.Import", "unit": "W", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Energy.Active.Import.Register", "unit": "Wh", "value": "458122" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1", "measurand": "Current.Import", "unit": "A", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1-N", "measurand": "Voltage", "unit": "V", "value": "228.22" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "Current.Offered", "unit": "V", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "Power.Offered", "unit": "V", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "SoC", "unit": "V", "value": "0" } ] } ], "connectorId": 1, "transactionId": 0 }]
[ocpp ] TRACE 2024/12/20 16:57:46 send wb3: [3,"251836754",{}]
.
.
.
[ocpp ] TRACE 2024/12/20 15:32:55 send wb4: [2,"2777259992","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":0,"stackLevel":2,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-12-20T14:31:55Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp ] TRACE 2024/12/20 15:32:55 recv wb4: [3,"2777259992", { "status": "NotSupported" }]
[lp-4 ] ERROR 2024/12/20 15:32:56 max charge current 6A: set charging profile: NotSupported
.
.
.
[ocpp ] TRACE 2024/12/20 15:06:11 send wb3: [2,"2654811303","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":0,"stackLevel":2,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-12-20T14:05:11Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16}]}}}]
[ocpp ] TRACE 2024/12/20 15:06:11 recv wb3: [3,"2654811303", { "status": "NotSupported" }]
[lp-3 ] ERROR 2024/12/20 15:06:11 max charge current 16A: set charging profile: NotSupported
.
.
.
[ocpp ] TRACE 2024/12/20 15:34:55 send wb3: [2,"3142705710","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":0,"stackLevel":2,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-12-20T14:33:55Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]
[ocpp ] TRACE 2024/12/20 15:34:55 recv wb3: [3,"3142705710", { "status": "NotSupported" }]
[lp-3 ] ERROR 2024/12/20 15:34:55 charger disable: set charging profile: NotSupported
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.12

Saueee, 2024-12-20:
Ich habe gerade gesehen, es gibt noch weitere Errors nach jedem Neustart:

`[ocpp ] TRACE 2024/12/20 17:03:11 send wb3: [2,"2410947932","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp ] TRACE 2024/12/20 17:03:11 recv wb3: [3,"2410947932", { "status": "Rejected", "connectorId": 0, "chargingSchedule": { "duration": 0, "chargingSchedulePeriod": [], "startSchedule": "0", "chargingRateUnit": "" }, "scheduleStart": "0" }]
[ocpp ] ERROR 2024/12/20 17:03:11 ocpp message (2410947932): FormationViolation - iso8601: Cannot parse "0": month 0 is not in range 1-12 ([3,"2410947932", { "status": "Rejected", "connectorId": 0, "chargingSchedule": { "duration": 0, "chargingSchedulePeriod": [], "startSchedule": "0", "chargingRateUnit": "" }, "scheduleStart": "0" }])`

`[ocpp ] ERROR 2024/12/20 17:02:15 ocpp message (669842954): OccurrenceConstraintViolation - Field Call.Payload.MeterValue[0].SampledValue[0].Value required but not found for feature MeterValues ([2,"669842954","MeterValues", { "meterValue": [ { "timestamp": "2024-12-20T16:02:16Z", "sampledValue": [ { "context": "Trigger", "measurand": "Power.Active.Import", "value": "" }, { "context": "Trigger", "measurand": "Energy.Active.Import.Register", "value": "" }, { "context": "Trigger", "measurand": "Current.Import", "value": "" }, { "context": "Trigger", "measurand": "Voltage", "value": "" }, { "context": "Trigger", "measurand": "Current.Offered", "value": "" }, { "context": "Trigger", "measurand": "Power.Offered", "value": "" }, { "context": "Trigger", "measurand": "SoC", "value": "" } ] } ], "connectorId": 0, "transactionId": 0 }])`

`[ocpp ] TRACE 2024/12/20 17:02:17 send wb4: [2,"701849447","GetConfiguration",{}]
[ocpp ] TRACE 2024/12/20 17:02:17 recv wb4: [3,"701849447", { "configurationKey": [ { "readonly": false, "key": "AllowOfflineTxForUnknownId", "value": "false" }, { "readonly": false, "key": "AuthorizationCacheEnabled", "value": "true" }, { "readonly": false, "key": "AuthorizeRemoteTxRequests", "value": "true" }, { "readonly": false, "key": "BlinkRepeat", "value": "10" }, { "readonly": false, "key": "ClockAlignedDataInterval", "value": "900" }, { "readonly": false, "key": "ConnectionTimeOut", "value": "60" }, { "readonly": false, "key": "ConnectorPhaseRotation", "value": "0.RST, 1.RST, 2.RTS" }, { "readonly": true, "key": "ConnectorPhaseRotationMaxLength", "value": "5" }, { "readonly": true, "key": "GetConfigurationMaxKeys", "value": "50" }, { "readonly": false, "key": "HeartbeatInterval", "value": "60" }, { "readonly": false, "key": "LightIntensity", "value": "50" }, { "readonly": false, "key": "LocalAuthorizeOffline", "value": "true" }, { "readonly": false, "key": "LocalPreAuthorize", "value": "true" }, { "readonly": false, "key": "MaxEnergyOnInvalidId", "value": "0" }, { "readonly": false, "key": "MeterValuesAlignedData", "value": "Power.Active.Import,Energy.Active.Import.Register,Voltage.Outlet,Current.Import.Inlet.L1,Current.Import.Inlet.L2,Current.Import.Inlet.L3,Voltage.Inlet.L1,Voltage.Inlet.L2,Voltage.Inlet.L3,Power.Active.Import.L1-N,Power.Active.Import.L2-N,Power.Active.Import.L3-N" }, { "readonly": true, "key": "MeterValuesAlignedDataMaxLength", "value": "700" }, { "readonly": false, "key": "MeterValuesSampledData", "value": "Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered,Power.Offered,SoC" }, { "readonly": true, "key": "MeterValuesSampledDataMaxLength", "value": "700" }, { "readonly": false, "key": "MeterValueSampleInterval", "value": "10" }, { "readonly": false, "key": "MinimumStatusDuration", "value": "1" }, { "readonly": true, "key": "NumberOfConnectors", "value": "1" }, { "readonly": false, "key": "ResetRetries", "value": "3" }, { "readonly": true, "key": "SupportedFileTransferProtocols", "value": "FTP,FTPS,HTTP,HTTPS" }, { "readonly": false, "key": "StopTransactionOnEVSideDisconnect", "value": "true" }, { "readonly": false, "key": "StopTransactionOnInvalidId", "value": "true" }, { "readonly": false, "key": "StopTxnAlignedData", "value": "Energy.Active.Import.Register" }, { "readonly": true, "key": "StopTxnAlignedDataMaxLength", "value": "1" }, { "readonly": false, "key": "StopTxnSampledData", "value": "Energy.Active.Import.Register" }, { "readonly": true, "key": "StopTxnSampledDataMaxLength", "value": "1" }, { "readonly": true, "key": "SupportedFeatureProfiles", "value": "Core, FirmwareManagement, LocalAuthListManagement, Reservation, RemoteTrigger" }, { "readonly": true, "key": "SupportedFeatureProfilesMaxLength", "value": "5" }, { "readonly": false, "key": "TransactionMessageAttempts", "value": "1" }, { "readonly": false, "key": "TransactionMessageRetryInterval", "value": "1" }, { "readonly": false, "key": "UnlockConnectorOnEVSideDisconnect", "value": "true" }, { "readonly": false, "key": "WebSocketPingInterval", "value": "30" }, { "readonly": false, "key": "LocalAuthListEnabled", "value": "true" }, { "readonly": true, "key": "LocalAuthListMaxLength", "value": "100" }, { "readonly": true, "key": "SendLocalListMaxLength", "value": "100" }, { "readonly": true, "key": "ReserveConnectorZeroSupported", "value": "false" }, { "readonly": true, "key": "ChargeProfileMaxStackLevel", "value": "2" }, { "readonly": true, "key": "ChargingScheduleAllowedChargingRateUnit", "value": "Current, Power" }, { "readonly": true, "key": "ChargingScheduleMaxPeriods", "value": "2" }, { "readonly": true, "key": "ConnectorSwitch3to1PhaseSupported", "value": "false" }, { "readonly": true, "key": "MaxChargingProfilesInstalled", "value": "0" }, { "readonly": false, "key": "WebSocketUrl", "value": "ws://10.0.90.111:8887/wb4" } ] }]
[wb4-1 ] WARN 2024/12/20 17:02:17 the required SmartCharging feature profile is not indicated as supported`

andig, 2024-12-20:
```
[ocpp ] TRACE 2024/12/20 15:32:55 send wb4: [2,"2777259992","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":0,"stackLevel":2,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-12-20T14:31:55Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp ] TRACE 2024/12/20 15:32:55 recv wb4: [3,"2777259992", { "status": "NotSupported" }]
[lp-4 ] ERROR 2024/12/20 15:32:56 max charge current 6A: set charging profile: NotSupported
```

Die Box ist nicht durch evcc steuerbar. Frage an Heidelberg: was gefällt der Box an diesem perfekt gültigen OCPP Kommando nicht? Bitte Ticket beim Herstellersupport öffnen.

/cc @premultiply

mfuchs1984, 2024-12-20:
{"SupportedFeatureProfiles", "value": "Core, FirmwareManagement, LocalAuthListManagement, Reservation, RemoteTrigger" }

Ich denke sie unterstützt das smartCharging Profil einfach nicht.

andig, 2024-12-20:
Leider untauglich 🙁

Saueee, 2024-12-20:
Kurze Rückfrage:
Steuerung von Strom bzw. Leistung ist ja dann nicht möglich an dieser Station.
Ist überhaupt Starten/Stoppen eines Ladevorgangs möglich wenn keine SmartCharging unterstützt wird?
Die Station dürfte ja dann überhaupt nicht (auch keine anderen OCPP Server) steuerbar sein?

andig, 2024-12-20:
Start/Stopp geht immer