evcc-io/evcc #17838: Allow charging based on "price limit" and "charging plan" to be active at the same time.

vhkristof, 2024-12-20:
**Is your feature request related to a problem? Please describe.**
I'd like to combine a repeating charging plan ("charging goal" set to 50%) and the "price limit" (for dynamic prices) to be active at the same time. 
Charge limit for my car is set to 80%.

In this case it would make sure the car is charged to 50% every day, but when prices are below the price limit it would continue to charge till either prices are too high again or the max SoC is reached.

Currently the cheap prices are ignored when a plan is active.

When prices would be low for a very long time overnight, charging would be stopped to soon. In this case I want it to charge as long as possible.

**Describe the solution you'd like**
Take into account the "charging goal" and "price limit" at the same time so the car can be "extra charged" when electricity is cheap.

**Describe alternatives you've considered**
Manually changing the "charging goal" every day after looking at the electricity prices.

andig, 2024-12-20:
Alresdy fixed in master

closed 2024-12-20 by andig

vhkristof, 2024-12-21:
Awesome, I'll give it a spin. Thx !