evcc-io/evcc #17839: Make wallbox optional if charging can be controlled through vehicle

elekktrisch, 2024-12-21:
**Is your feature request related to a problem? Please describe.**
According to https://docs.evcc.io/docs/reference/configuration/loadpoints for loadpoints, always a charger has to be specified. I do have a charger, but it is offline. The car on the other hand has a very good API which allows me to controll it through Home Assistant. But with evcc, that doesn't seem to help, since the loadpoint can't be used just with the vehicle. Also, if I just specify the vehicle and no loadpoint, it doesn't even show up in the UI.

**Describe the solution you'd like**
Why not make the charger optional if the vehicle has enough functionality to control everything needed?

**Describe alternatives you've considered**
I tried adding a demo-charger with static configs as a workaround, but that way, the loadpoint shows up as "disconnected"

**Additional context**
In my specific case, it's a Tesla (note: I bought it before we knew Elon is crazy), but I assume it could be useful for most brands if there is only a "dumb" charger available.

VolkerK62, 2024-12-21:
It is possible, with a custom charger, but evcc will not control the charge power through the vehice.
So it will only be for logging purpose

elekktrisch, 2024-12-21:
Yes that's what I understand too. But I'd specifically like to control charge power. And since that's possible through the vehicle API, I don't see why it should not become a feature of evcc. Am I missing something?

elekktrisch, 2024-12-21:
Also, according to https://docs.evcc.io/docs/devices/chargers#tesla-twc3 in the case of Tesla, it is already controlled through the vehicle. But since I don't have a TWC, it doesn't help me yet. I tried adding this, in the hopes that it doesn't matter if the connection actually works:
```
chargers:
  - name: juice_booster
    type: template
    template: twc3
    host: 127.0.0.1 # IP-Adresse oder Hostname
```
but the vehicle still shows up as "disconnected"

andig, 2024-12-21:
We don‘t plan to change this. You can always create a custom fake charger that gives status, most importantly if vehicle is connected.

elekktrisch, 2024-12-21:
Thanks for the idea. That would be like a flag which tells the custom charger to behave like TWC? This would definitely help. So the connected-status would then be piped from the vehicle to the custom charger? Or would it have to come from somewhere else?

andig, 2024-12-21:
We would need to figure out which part of the charge API (status, enable, enabled, maxcurrent) the vehicle could provide and make that optional config on the charger side and relay to vehicle.

Jabe, 2024-12-22:
I would love to have this setup in EVCC. I use a wallbox that uses OCPP, all is good and it works fine. But the Tesla can be set to 1 A through the API (I use TeslaBleHttpProxy [custom car](https://github.com/wimaha/TeslaBleHttpProxy?tab=readme-ov-file#setup-evcc)), that way I could eliminate the phase switching and just keep it at 3 ph all the time. This will increase efficiency and extend the lifetime of the wallbox. Bonus for lower starting threshold because 3x1 A is lower then 1x6 A. 😃

michiproep, 2024-12-23:
Just an idea: You can try to setup a custom "integrated device"

andig, 2025-01-01:
I guess for BLE you want https://github.com/evcc-io/evcc/pull/17866

elekktrisch, 2025-01-02:
One other idea (more like a workaround though): 

Just start a mock-server imitating twc3, and point evcc to that mock-server. The mock could be something like this:
```
const express = require('express');
const morgan = require('morgan')

const app = express();
const port = 80;

app.use(express.json());
app.use(morgan('combined'));

const status = {
 vitals: {"contactor_closed": false,
              "vehicle_connected": false,
              "session_s": 0,
              "grid_v": 230,
              "grid_hz": 50,
              "vehicle_current_a": 0.1,
              "currentA_a": 0.0,
              "currentB_a": 0.1,
              "currentC_a": 0.0,
              "currentN_a": 0.0,
              "voltageA_v": 0.0,
              "voltageB_v": 0.0,
              "voltageC_v": 0.0,
              "relay_coil_v": 11.8,
              "pcba_temp_c": 19.2,
              "handle_temp_c": 15.3,
              "mcu_temp_c": 25.1,
              "uptime_s": 0,
              "input_thermopile_uv": -233,
              "prox_v": 0.0,
              "pilot_high_v": 11.9,
              "pilot_low_v": 11.9,
              "session_energy_wh": 0,
              "config_status": 5,
              "evse_state": 1,
              "current_alerts": []}};

app.get('/api/1/vitals', (req, res) => { 
  res.setHeader('Content-Type', 'application/json'); 
  res.send(status.vitals);
});

app.post('/api/update', (req, res) => { 
  status.vitals = req.body;
  console.log(`received new vitals: ${JSON.stringify(status.vitals)}`);
  res.end();
});

app.listen(port, () => { 
  console.log(`Server running at http://localhost:${port}`);
});
```

Then, the only thing that's needed is that someone calls the update endpoint with the correct flags and values. This should be feasible quite easily through home-assistant automations.

Jabe, 2025-01-02:
This is pretty much what I built:  
https://github.com/Jabe/ocpp2twc

AI did most of the work actually 😉

closed 2025-01-14 by github-actions[bot]