evcc-io/evcc #17843: Hausbatterie Netzladen kenntlich machen

jori0001, 2024-12-21:
Wäre toll, wenn ihr einbauen könntet, dass auf der Hauptseite angezeigt wird, wenn das Hausbatterie Netzladen aktiviert ist. 
Des Weiteren wäre es cool, wenn das Netzladen nur für das einmalige Laden aktiviert wäre. 

Bsp.: man aktiviert es zum Laden <28 Cent, weil die Preise gerade sehr hoch sind. Am nächsten Tag vergisst man es manuell zu deaktivieren. Die Preise fallen aber auf z.B. 19-27 Cent. Da das Netzladen weiterhin aktiv ist, lädt die Batterie für 27 Cent (da <28 vom Vortag noch aktiv). 

Zusätzlich könntet ihr vielleicht noch eine Checkbox einbauen, mit der man das Netzladen dauerhaft aktiv schalten kann (so wie es jetzt ist), falls das jemand haben möchte.

naltatis, 2024-12-22:
Zu deinem ersten Punkt:
In der zugeklappten Ansicht wechselt das Batterie Icon in ein 🔋⇪ Icon, wenn die Batterie gerade aus dem Netz geladen wird.
[image]

Klappt man die Detailansicht auf gibts auch eine Erklärung mit der gesetzten Preisgrenze dazu
[image]

Das Thema "Limit nach einer Zeit zurücksetzen" oder dynamischere Möglichkeiten (günstigste 2 Stunden, ...) wird aktuell hier diskutiert. Da werden wir auf jeden Fall noch mehr machen.
https://github.com/evcc-io/evcc/discussions/16364

Was man aktuell in der Tat nicht sieht ist ob noch ein Preislimit gesetzt ist oder nicht. Beim Ladepunkt zeigen wir das ja inzwischen permanent an. Ist das das was du hier meinst?

jori0001, 2024-12-22:
Hi. Danke für die Rückmeldung. 
Ja genau das meine ich. So wie es beim Ladepunkt gelöst ist, wäre es auch für die Hausbatterie toll. 
mal Ladepunkt sieht man sofort, wenn das Netzladen zu nem bestimmten Preis aktiviert ist. 🙂

closed 2025-01-04 by andig