evcc-io/evcc #17850: Wiederholender Ladeplan: request failed with status code 400

An-Wi, 2024-12-21:
### Describe the bug

Beim Versuch, einen wiederholenden Plan hinzuzufügen, passiert zunächst „nichts“. In den Meldungen ist dann folgendes zu finden:
[image]

### Steps to reproduce

1. Ladeplanung
2. Wiederholenden Plan hinzufügen
3. Meldungen einsehen

### Configuration details

```yaml
.
```

### Log details

```text
(keine dazu passenden Einträge)
```

### What type of operating system are you running?

Windows

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.12

andig, 2024-12-22:
Kannst Du das reproduzieren? Wie sieht der Request aus (Entwicklerkonsole des Browsers, Netzwerktab)?

naltatis, 2024-12-22:
Ok, das ist merkwürdig. Wir ermitteln in der Tat die aktuelle Zeitzone für den wiederholenden Plan im Browser und nutzen auch genau die in Go um dann die Plan auszuwerten. Letzteres scheint hier ja fehlzuschlagen obwohl "Europe/Rome" eine valide Zeitzone ist. https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
Bist du in Italien unterwegs? Bei mir ists "Europe/Berlin"

An-Wi, 2024-12-22:
Screenshot oben ist vom iPhone.
Wenn ich es auf dem iPad oder am PC (Google Chrome) teste, kommt die gleiche Fehlermeldung, aber mit „unknown time zone Europe/Berlin.“

andig, 2024-12-22:
> https://github.com/evcc-io/evcc/issues/17850#issuecomment-2558379219

?

An-Wi, 2024-12-23:
> Kannst Du das reproduzieren? Wie sieht der Request aus (Entwicklerkonsole des Browsers, Netzwerktab)?

Console:
[image]

Network:
[image]
[image]
[image]
[image]
[image]

naltatis, 2024-12-23:
Hm, könnte das ein Go/Windows Problem sein?
https://github.com/golang/go/issues/62065

naltatis, 2024-12-23:
See https://github.com/bpg/terraform-provider-proxmox/pull/1436

andig, 2024-12-23:
Die Lösung wäre anscheinend die TZ Informationen in evcc einzubauen. Warum Europe/Berlin aber nicht klappen sollte ist nur ein Rätsel. Alternativ: immer UTC für die APIs verwenden?

naltatis, 2024-12-24:
> Option: immer UTC für die APIs verwenden?

Das funktioniert leider nicht. Wir speichern ja keinen festen Zeitpunkt ab sondern Wochentage + Uhrzeit. Um daraus einen Zeitpunkt errechnen zu können müssen wir die Zeitzone/Location des Anwenders kennen. Sonst verschiebt sich zur Sommer/Winterzeit der Ladeplan.

naltatis, 2024-12-24:
Die `pun` Implementierung nutzt den TZ Mechanismus auch bereits. Das müssten auf Windows heute auch bereits nen Fehler geben. Vermutlich nutzt die Kombination niemand :D

closed 2024-12-25 by andig

An-Wi, 2024-12-30:
Funktioniert nun auf Windows-Server - Danke. 👍