evcc-io/evcc #17851: Growatt SPH4600 Batterie wird nicht gesteuert

DerP9, 2024-12-21:
### Describe the bug

Hallo,

ich habe die folgenden Situation. 

1. Growatt SPH4600 angebunden in evcc per MODBUS TCP/IP
2. Growatt TLX Wechselrichter in evcc per Modbus TCP/IP
3. waveshare  RS485 TO POE ETH Converter für MODBUS RTU - TCP
4. CFOS Powerbrain Wallbox 
5. Tibber mit Tibber Puls und der API Abfrage.

### Steps to reproduce

1. Beim Laden per PV Strom im Modus schnell wird die Batterie entleert. 
2. Ich habe den Tibber Vertrag integriert. Hier wird unter der definierten oder überhaupt bei einer definierte Preisgrenze nicht geladen. 

Verhindere Entladung im Schnell-Modus und bei geplantem Laden. ist aktiviert
AC Laden aus dem Stromnetz ist aktiviert. Zeitfenster ist gesetzt und deaktivert. 

### Configuration details

```yaml
meters:
- type: template
  template: tibber-pulse
  usage: grid
  token: XXXXXX
  name: grid1
- type: template
  template: growatt-hybrid
  id: 1
  host: 192.168.13.225
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: growatt-hybrid
  id: 1
  host: 192.168.13.225
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
- type: template
  template: growatt-hybrid-tlxh
  id: 1
  host: 192.168.13.111
  port: 502
  usage: pv
  modbus: tcpip
  name: pv1
chargers:
- type: template
  template: cfos
  host: 192.168.13.224
  name: wallbox4

tariffs:
  grid:
    type: template
    template: tibber
    token: XXXXXXXX
loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv

site:
[... 923 lines cut ...]
[site  ] DEBUG 2024/12/21 23:21:56 site power: 543W
[lp-1  ] DEBUG 2024/12/21 23:21:56 charge total import: 905.990kWh
[lp-1  ] DEBUG 2024/12/21 23:21:56 charger status: A
[site  ] DEBUG 2024/12/21 23:22:10 ----
[lp-1  ] DEBUG 2024/12/21 23:22:10 charge power: 1W
[lp-1  ] DEBUG 2024/12/21 23:22:10 charge currents: [0 0 0]A
[site  ] DEBUG 2024/12/21 23:22:10 grid power: 455W
[site  ] DEBUG 2024/12/21 23:22:10 grid currents: [1 0.5 0.4]A
[site  ] DEBUG 2024/12/21 23:22:10 pv 1 power: 0W
[site  ] DEBUG 2024/12/21 23:22:11 pv 2 power: 0W
[site  ] DEBUG 2024/12/21 23:22:11 pv power: 0W
[site  ] DEBUG 2024/12/21 23:22:11 battery soc: 14%
[site  ] DEBUG 2024/12/21 23:22:11 battery power: 0W
[site  ] DEBUG 2024/12/21 23:22:11 site power: 555W
[lp-1  ] DEBUG 2024/12/21 23:22:11 charge total import: 905.990kWh
[lp-1  ] DEBUG 2024/12/21 23:22:11 charger status: A
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

DerP9, 2024-12-21:
[evcc-20241221-232848-trace.log](https://github.com/user-attachments/files/18220243/evcc-20241221-232848-trace.log)

andig, 2024-12-22:
/cc @schulzh das Template war dankbarerweise von Dir- vllt. weisst Du was hier los ist?

andig, 2024-12-22:
@DerP9 das Setup gemäß Doku hast Du durchgeführt?!

DerP9, 2024-12-22:
Meine Vermutung ist, dass ich die Modbus Register nicht korrekt geschrieben habe. Wenn ich die Werte 1000 bis 1002 abfrage, steht da nicht die aus Doku hinterlegten Werte.

Könnt ihr mir evtl. bei dem Modbus Write Befehl helfen?

schulzh, 2024-12-22:
Wichtig ist, dass alle drei Werte auf einmal geschrieben werden (über write multiple), sonst akzeptiert der Wechselrichter diese nicht. Mit der [Modbus CLI](https://github.com/favalex/modbus-cli) sollte der Befehl für dich `modbus -s 1 192.168.13.225:502 H@1100=0 H@1101=5947 H@1102=0` sein. Mit `modbus -s 1 192.168.13.225:502 H@1100 H@1101 H@1102` kannst du überprüfen, ob diese richtig geschrieben wurden.

DerP9, 2024-12-22:
pascal@home-evcc01:~$ modbus -s 1 192.168.13.225:502 H@1100=0 H@1101=5947 H@1102=0
Parsed 0 registers definitions from 1 files
pascal@home-evcc01:~$ modbus -s 1 192.168.13.225:502 H@1100 H@1101 H@1102
Parsed 0 registers definitions from 1 files
1100: 0 0x0
1101: 5947 0x173b
1102: 0 0x0
pascal@home-evcc01:~$

gerade mal getestet. Was sagst du dazu?

schulzh, 2024-12-22:
Ja das sieht richtig aus, Funktioniert es jetzt?

DerP9, 2024-12-22:
[image]

DerP9, 2024-12-22:
Das sieht echt gut aus so. 

Vielen Dank für super schnelle Hilfe. 

Ich wünsche euch einen schönen 4 Advent.

closed 2024-12-22 by DerP9

MaZ181, 2024-12-29:
Hallo könnte mir jemand bitt erklären wie ich Modbus CLI auf meinem Mac installieren kann um die Vorbereitung für den Wechselrichter durchzuführen Vielen dank.  Ich habe leider im Internet nichts gefunden.