evcc-io/evcc #17852: Tesla with TWC3 - evcc cannot wake to charge.

delbz, 2024-12-22:
### Describe the bug

I have to wake the Tesla via the app etc before evcc will charge.  Both via excess solar and plan triggers.  
Using 0.131.12 with Home assistant add-on. 
I have not tried the nightly build. 

### Steps to reproduce

1. Set to solar. 
2. Get error message. 
3. No charge. 
...

### Configuration details

```yaml
Will share if necessary.
```

### Log details

```text
`[lp-1 ] DEBUG 2024/12/22 13:25:16 pv enable in 0s
[lp-1 ] DEBUG 2024/12/22 13:25:16 pv enable timer elapsed
[lp-1 ] DEBUG 2024/12/22 13:25:18 max charge current: waking up vehicle
[lp-1 ] ERROR 2024/12/22 13:25:18 max charge current 6A: asleep `
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.12

andig, 2024-12-22:
Needs trace log for Tesla. The error is not enough to see whats going on.

closed 2024-12-22 by andig

delbz, 2024-12-22:
Does this help?

`[site ] DEBUG 2024/12/22 16:18:15 ----
[twc3 ] TRACE 2024/12/22 16:18:15 GET http://192.168.0.157/api/1/vitals
[twc3 ] TRACE 2024/12/22 16:18:15 {"contactor_closed":false,"vehicle_connected":true,"session_s":12345,"grid_v":239.7,"grid_hz":49.958,"vehicle_current_a":0.4,"currentA_a":0.4,"currentB_a":0.0,"currentC_a":0.3,"currentN_a":0.0,"voltageA_v":3.9,"voltageB_v":0.0,"voltageC_v":0.0,"relay_coil_v":11.9,"pcba_temp_c":36.2,"handle_temp_c":31.8,"mcu_temp_c":43.2,"uptime_s":334283,"input_thermopile_uv":-206,"prox_v":0.0,"pilot_high_v":8.7,"pilot_low_v":-11.7,"session_energy_wh":3485.300,"config_status":5,"evse_state":4,"current_alerts":[],"evse_not_ready_reasons":[1]}
[lp-1 ] DEBUG 2024/12/22 16:18:15 charge power: 0W
[lp-1 ] DEBUG 2024/12/22 16:18:15 charge currents: [0.4 0 0.3]A
[grid1 ] TRACE 2024/12/22 16:18:15 GET http://192.168.0.196/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[pv2 ] TRACE 2024/12/22 16:18:15 GET http://192.168.0.196/solar_api/v1/GetPowerFlowRealtimeData.fcgi
[grid1 ] TRACE 2024/12/22 16:18:16 { "Body" : { "Data" : { "Inverters" : { "1" : { "DT" : 76, "E_Day" : 37290, "E_Total" : 82314600, "E_Year" : 7761645.5, "P" : 2234 }, "2" : { "DT" : 76, "E_Day" : 32780, "E_Total" : 32311398, "E_Year" : 7295629, "P" : 1721 } }, "Site" : { "E_Day" : 70070, "E_Total" : 114625998, "E_Year" : 15057274.5, "Meter_Location" : "grid", "Mode" : "meter", "P_Akku" : null, "P_Grid" : -2848.6700000000001, "P_Load" : -1106.3299999999999, "P_PV" : 3955, "rel_Autonomy" : 100, "rel_SelfConsumption" : 27.97294563843236 }, "Version" : "12" } }, "Head" : { "RequestArguments" : {}, "Status" : { "Code" : 0, "Reason" : "", "UserMessage" : "" }, "Timestamp" : "2024-12-22T16:18:15+10:00" } }
[site ] DEBUG 2024/12/22 16:18:16 grid power: -2849W
[pv2 ] TRACE 2024/12/22 16:18:16 { "Body" : { "Data" : { "Inverters" : { "1" : { "DT" : 76, "E_Day" : 37290, "E_Total" : 82314600, "E_Year" : 7761645.5, "P" : 2234 }, "2" : { "DT" : 76, "E_Day" : 32780, "E_Total" : 32311398, "E_Year" : 7295629, "P" : 1721 } }, "Site" : { "E_Day" : 70070, "E_Total" : 114625998, "E_Year" : 15057274.5, "Meter_Location" : "grid", "Mode" : "meter", "P_Akku" : null, "P_Grid" : -2848.6700000000001, "P_Load" : -1106.3299999999999, "P_PV" : 3955, "rel_Autonomy" : 100, "rel_SelfConsumption" : 27.97294563843236 }, "Version" : "12" } }, "Head" : { "RequestArguments" : {}, "Status" : { "Code" : 0, "Reason" : "", "UserMessage" : "" }, "Timestamp" : "2024-12-22T16:18:16+10:00" } }
[site ] DEBUG 2024/12/22 16:18:16 pv power: 3955W
[site ] DEBUG 2024/12/22 16:18:16 site power: -2849W
[lp-1 ] DEBUG 2024/12/22 16:18:16 charge voltages: [3.9 0 0]V
[lp-1 ] DEBUG 2024/12/22 16:18:16 charger status: B
[lp-1 ] DEBUG 2024/12/22 16:18:16 plan: charge 1h43m59s between 2024-12-23 04:16:01 +1000 AEST until 2024-12-23 06:00:00 +1000 AEST (power: 7360W, avg cost: 0.080)
[lp-1 ] TRACE 2024/12/22 16:18:16 slot from: 2024-12-23 04:16:01 +1000 AEST to 2024-12-23 05:00:00 +1000 AEST cost 0.080
[lp-1 ] TRACE 2024/12/22 16:18:16 slot from: 2024-12-23 05:00:00 +1000 AEST to 2024-12-23 06:00:00 +1000 AEST cost 0.080
[lp-1 ] DEBUG 2024/12/22 16:18:16 pv charge current: 12.4A = 0A + 12.4A (-2849W @ 1p)
[lp-1 ] DEBUG 2024/12/22 16:18:16 site power -2849W <= 0W enable threshold
[lp-1 ] DEBUG 2024/12/22 16:18:16 pv enable in 0s
[lp-1 ] DEBUG 2024/12/22 16:18:16 pv enable timer elapsed
[tesla ] TRACE 2024/12/22 16:18:16 POST https://tesla.evcc.io/api/1/vehicles/LRW3F7EK8NXXXXXX/command/set_charging_amps
[tesla ] TRACE 2024/12/22 16:18:19 {"charging_amps": 6} -- {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
[lp-1 ] DEBUG 2024/12/22 16:18:19 max charge current: waking up vehicle
[tesla ] TRACE 2024/12/22 16:18:19 POST https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/LRW3F7EKXXXXXXXXXX/wake_up
[tesla ] TRACE 2024/12/22 16:18:20 {"response":{"id":3744515789066467,"user_id":283109XXXXXX,"vehicle_id":2252136XXXXXXXX,"vin":"LRW3F7EK8NXXXXXX","color":null,"access_type":"OWNER","granular_access":{"hide_private":false},"tokens":null,"state":"offline","in_service":false,"id_s":"3744515789066467","calendar_enabled":true,"api_version":82,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}}
[lp-1 ] ERROR 2024/12/22 16:18:20 max charge current 6A: asleep
[site ] DEBUG 2024/12/22 16:18:45 ----`

andig, 2024-12-22:
Its unreadable. No.

delbz, 2024-12-22:
[evcc trace log Tesla.docx](https://github.com/user-attachments/files/18223501/evcc.trace.log.Tesla.docx)

andig, 2024-12-23:
It would be nice to post the logs here rather than attached files. In any case:

```
[lp-1 ] DEBUG 2024/12/22 16:18:19 max charge current: waking up vehicle
[tesla ] TRACE 2024/12/22 16:18:19 POST https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/LRW3F7EKXXXXXXXXXX/wake_up
[tesla ] TRACE 2024/12/22 16:18:20 {"response":{"id":3744515789066467,"user_id":283109XXXXXX,"vehicle_id":2252136XXXXXXXX,"vin":"LRW3F7EK8NXXXXXX","color":null,"access_type":"OWNER","granular_access":{"hide_private":false},"tokens":null,"state":"offline","in_service":false,"id_s":"3744515789066467","calendar_enabled":true,"api_version":82,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}}
```

The car _is_ correctly being woken up. If that doesn't work then it's a question to Tesla why not. Maybe it's parked in a spot without reception?

    "state":"offline"

Might indicate this.