evcc-io/evcc #17854: MQTT API funktioniert nicht mehr mit QoS 1 Nachrichten

erikgieseler, 2024-12-22:
### Describe the bug

In der aktuellen evcc Version funktioniert die MQTT API nur mit QoS 0 Nachrichten.
QoS 1 Nachrichten werden scheinbar nicht berücksichtigt.
In der Vergangenheit hat das auch immer mit QoS 1 Nachrichten funktioniert.

### Steps to reproduce

Beispiel: Ladepunkt 4 auf PV Modus umstellen:

Topic: evcc/loadpoints/4/mode/set
Payload: pv
QoS 0 = funktioniert
QoS 1 = funktioniert nicht

Im beigefügten Log sieht man nur den Befehl mit QoS 0
[evcc-20241222-111229-trace.log](https://github.com/user-attachments/files/18221474/evcc-20241222-111229-trace.log)

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

andig, 2024-12-22:
How are you testing this?

erikgieseler, 2024-12-22:
Ich habe das mit dem MQTT Explorer getestet: https://mqtt-explorer.com/

[image]

andig, 2024-12-22:
Vllt ist der ja kaputt?

erikgieseler, 2024-12-22:
Ist eher unwahrscheinlich, mit Home Assisstant Automatisierungen geht QoS 1 auch nicht.

erikgieseler, 2024-12-22:
Hab das Image nochmal gepullt, jetzt geht es auf einmal wieder.
Danke für die Rückmeldung!