evcc-io/evcc #17862: MQTT separate imported/exported grid energy

filipSu, 2024-12-22:
**Is your feature request related to a problem? Please describe.**
For integration purposes (specifically with [Homey](https://homey.app/)), it would be nice to have grid energy separated into imported and exported values. Right now, only `gridEnergy` property is exposed, which as I understand it, means imported energy.

**Describe the solution you'd like**
Expose `gridEnergyImported` and `gridEnergyExported` through MQTT integration

andig, 2024-12-23:
We don't collect these values, hence can't export them :(

closed 2024-12-23 by andig

filipSu, 2024-12-24:
@andig and would it be a problem to start collecting them? A lot of power meters expose these values (f.e. my deye 3p inverter). Because it is a very valuable information when you want to easily analyze your energy usage over a longer period of time, so in my opinion it fits to the overall purpose of evcc as a service

andig, 2024-12-24:
Not a problem per se but a big change- without use for evcc.