evcc-io/evcc #17872: KIA EV6 has no SOC state anymore

MasterGoofy, 2024-12-23:
### Describe the bug

Since some days I have no SOC from my car anymore. It is red with 0.0%. It was working for a long time and I haven't changed the configuration. Was there a change on KIA or EVCC side?

### Steps to reproduce

For example planned charging shows no SOC state.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.0.160
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# trial token, valid until 2024-09-18
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjY2NTM2MDAsImlhdCI6MTcyNTg3NjAwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.cR5Gfj9vnHxnGQoho5GIDJlo5tL6exD5yI5vSm2YbVE

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
- name: huawei_grid
[... 239 lines cut ...]
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
No log entry
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.12

VolkerK62, 2024-12-23:
My Hyundai is working.
Pls show output of `evcc vehicle --log trace`

andig, 2024-12-23:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-12-23 by andig

MasterGoofy, 2024-12-23:
I was able to fix the issue by myself. I deleted the car in the evcc.yaml and added it in the UI. Now the SOC is visible again and can be used in the charging plan.

VolkerK62, 2024-12-24:
afaik there is no difference btw yaml and UI configuration.
I think, the restart would have solved the problem.