evcc-io/evcc #17879: EVCC Version V0.131.2 or newer getting timeout issues when connecting to EM2GO Home

StefanA-LU, 2024-12-23:
### Describe the bug

Tested different versions now to connect to EM2GO Home.
Hardware Version: E22_1.2
Software Version: A03C_1.3
UI Version: 010.5

App connection works immediately without problems. Network connection is stable.

### Steps to reproduce

1. Install different versions and restart EVCC
2. Error messages appear at start with version V0.131.6 / 8 / 12 and it doesn't connect. Doesn't work.
3. Version V0.131.0 / 1 / 2 / 3 starts without an error but messages are thrown several times that connection was refused or timeout appeared. But seems to work.
4. Other versions not tested
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxx

interval: 30s # control cycle interval

meters:
- type: template
  template: e3dc-rscp
  usage: grid
  host: 10.0.0.30
  port: 5033
  user: xxxxx
  password: xxxxx
  key: xxxxx
  name: grid1
- type: template
  template: e3dc-rscp
  usage: pv
  host: 10.0.0.30
  port: 5033
  user: xxxxx
  password: xxxxx
  key: xxxxx
  name: pv2
- type: template
  template: e3dc-rscp
  usage: battery
  host: 10.0.0.30
  port: 5033
  user: xxxxx
  password: xxxxx
  key: xxxxx
  name: battery3

chargers:
- type: template
  template: em2go-home
  host: 10.0.0.66
  name: Wallbox

loadpoints:
- title: Garage
  charger: Wallbox
  mode: pv

site:
  title: Haus
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
time="2024-12-24T00:03:49+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-12-24T00:03:49+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-12-24T00:03:49+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site ] DEBUG 2024/12/24 00:03:49 pv power: 0W
[site ] DEBUG 2024/12/24 00:03:49 grid power: 791W
time="2024-12-24T00:03:49+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
[site ] DEBUG 2024/12/24 00:03:49 battery soc: 30%
[site ] DEBUG 2024/12/24 00:03:49 battery power: 0W
[site ] DEBUG 2024/12/24 00:03:49 site power: 791W
[lp-1 ] ERROR 2024/12/24 00:03:58 charge voltages: dial tcp 10.0.0.66:502: connectex: Es konnte keine Verbindung hergestellt werden, da der Zielcomputer die Verbindung verweigerte.
[lp-1 ] ERROR 2024/12/24 00:03:59 charge rater: dial tcp 10.0.0.66:502: connectex: Es konnte keine Verbindung hergestellt werden, da der Zielcomputer die Verbindung verweigerte.
[lp-1 ] ERROR 2024/12/24 00:04:00 charge timer: dial tcp 10.0.0.66:502: connectex: Es konnte keine Verbindung hergestellt werden, da der Zielcomputer die Verbindung verweigerte.
[lp-1 ] ERROR 2024/12/24 00:04:01 charge total import: dial tcp 10.0.0.66:502: connectex: Es konnte keine Verbindung hergestellt werden, da der Zielcomputer die Verbindung verweigerte.
[lp-1 ] ERROR 2024/12/24 00:04:02 charger status: dial tcp 10.0.0.66:502: connectex: Es konnte keine Verbindung hergestellt werden, da der Zielcomputer die Verbindung verweigerte.
[site ] DEBUG 2024/12/24 00:04:02 ----
[lp-1 ] ERROR 2024/12/24 00:04:13 charge power: read tcp 10.0.0.6:49229->10.0.0.66:502: i/o timeout
```

### What type of operating system are you running?

Windows

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.0

StefanA-LU, 2024-12-24:
Can be closed, solved. This happens in connection with a Power-LAN WLAN adapter if it is connected to a socket in whose circuit an interfering power supply is connected.

StefanA-LU, 2024-12-24:
With version 0.131.12, the process has to be restarted several times before the wallbox adapter can be read correctly. There seems to be a synchronization or time problem with the connection to the EM2GO Home. These connection attempts should become more robust (longer timeout, multiple attempts before abort)

StefanA-LU, 2024-12-24:
After a successful start, however, the error messages described above no longer appear. The only critical thing is the launch of EVCC.

andig, 2024-12-24:
> dial tcp 10.0.0.66:502: connectex: Es konnte keine Verbindung hergestellt werden, da der Zielcomputer die Verbindung verweigerte.

That sounds like a network problem. 

> EVCC Version V0.131.0 or newer getting timeout issues when connecting to EM2GO Home 

vs

> Version V0.131.0 / 1 / 2 / 3 starts without an error but messages are thrown several times that connection was refused or timeout appeared. But seems to work.

Which is true? Please add trace logs of working and non-working.

StefanA-LU, 2024-12-25:
It's true that the first reason was my notebook ac adapter disturbing the Power-LAN traffic  because it was plugged-in the socket directly beside the Power-LAN-WLAN adapter. I moved the Power-LAN-WLAN adapter to another socket and it works without any message. Only the startup process fails by random and requires 2 or 3 attempts to run. Because this is a timeout, I assume that the timeout limits should be adjusted or it should try it several times before aborting the connection attempts with an error message.

StefanA-LU, 2024-12-25:
[main ] INFO 2024/12/25 08:59:14 evcc 0.131.12
[main ] INFO 2024/12/25 08:59:14 using config file: C:\EVCC\evcc.yaml
[db ] INFO 2024/12/25 08:59:14 using sqlite database: C:\Windows\system32\config\systemprofile\.evcc\evcc.db
[db ] TRACE 2024/12/25 08:59:14 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db ] TRACE 2024/12/25 08:59:14 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2024/12/25 08:59:14 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db ] TRACE 2024/12/25 08:59:14 SELECT * FROM `settings` 9 <nil>
[db ] TRACE 2024/12/25 08:59:15 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db ] TRACE 2024/12/25 08:59:15 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db ] TRACE 2024/12/25 08:59:15 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db ] TRACE 2024/12/25 08:59:15 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2024/12/25 08:59:15 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db ] TRACE 2024/12/25 08:59:15 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db ] TRACE 2024/12/25 08:59:15 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main ] INFO 2024/12/25 08:59:15 listening at :7070
[db ] TRACE 2024/12/25 08:59:15 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db ] TRACE 2024/12/25 08:59:15 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[em2go ] TRACE 2024/12/25 08:59:15 modbus: send 00 01 00 00 00 09 ff 10 00 5b 00 01 02 00 3d
[main ] FATAL 2024/12/25 08:59:25 cannot create charger 'Wallbox': cannot create charger type 'template': cannot create charger type 'em2go-home': read tcp 10.0.0.6:50580->10.0.0.66:502: i/o timeout
[main ] TRACE 2024/12/25 08:59:25 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
[main ] FATAL 2024/12/25 08:59:25 will attempt restart in: 15m0s
[main ] TRACE 2024/12/25 08:59:25 {"url":"https://api.github.com/repos/evcc-io/evcc/releases/191516848","assets_url":"https://api.github.com/repos/evcc-io/evcc/releases/191516848/assets","upload_url":"https://uploads.github.com/repos/evcc-io/evcc/releases/191516848/assets{?name,label}","html_url":"https://github.com/evcc-io/evcc/releases/tag/0.131.12","id":191516848,"author":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"node_id":"RE_kwDODX4bUs4LalCw","tag_name":"0.131.12","target_commitish":"master","name":"0.131.12","draft":false,"prerelease":false,"created_at":"2024-12-18T19:14:15Z","published_at":"2024-12-18T19:15:32Z","assets":[{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/214273433","id":214273433,"node_id":"RA_kwDODX4bUs4MxY2Z","name":"checksums.txt","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"content_type":"text/plain; charset=utf-8","state":"uploaded","size":762,"download_count":2,"created_at":"2024-12-18T19:27:57Z","updated_at":"2024-12-18T19:27:57Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.131.12/checksums.txt"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/214273432","id":214273432,"node_id":"RA_kwDODX4bUs4MxY2Y","name":"evcc_0.131.12_amd64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.gi [... cut ...]
time="2024-12-25T08:59:28+01:00" level=info msg="successfully connected to 10.0.0.30:5033"
time="2024-12-25T08:59:28+01:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2024-12-25T08:59:28+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2024-12-25T08:59:29+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2024-12-25T08:59:29+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"

StefanA-LU, 2024-12-25:
Did a restart

StefanA-LU, 2024-12-25:
[image]

andig, 2024-12-25:
Please reproduce this with a wired LAN connection.

closed 2024-12-25 by andig

StefanA-LU, 2024-12-25:
That's not possible. There isn't any socket in the vicinity of the EM2GO Home. And the EM2GO Home doesn't have a LAN socket too. It's only usable with WLAN.

And I had to configure an own WLAN because EM2GO Home has password limits of 30 digits but my WLAN password has 32 digits. And I can't change all settings of over 20 devices because one piece of hardware seems not to consider network standards.

StefanA-LU, 2024-12-25:
[image]

How fast and short is it required?

StefanA-LU, 2024-12-25:
I assume that the Firmware is weak. It fails at smallest connection issues.

Now I can't get it to start again after the test for creating the TRACE. I already rebooted the network parts and the EM2GO. I've a lot of work and will abort it at this point. I'm short before sending back the EM2GO.

StefanA-LU, 2024-12-25:
Now I set the display time to 5 minutes, which was my last change to reduce it to 1 minute. Don't know if that was the reason. I switched off modbus, load-control and WLAN on the box, switched everything on again and did a cold restart. Know I got it to run again.

From my view as IT engineer it's a bad implementation of network traffic in the box itself.

But how to deal with it? I already informed D-Parts. But maybe there is a possibility to improve the handshake from outside too.

StefanA-LU, 2024-12-25:
Additional test:
1. evcc -migrate
2. start fails always
3. restore C:\Windows\System32\config\systemprofile\.evcc from tonight's synology ABB
4. start fails always
5. settings on wallbox: modbus off/on, load-control off/on, WLAN off/on
6. first attempt of starting EVCC works.

I tried it, because I did the migration today morning and after failing I restored it too, but it works only if there are the additional steps on the EM2GO Home. It seems that the connection attempts cause an internal module crash in the wallbox which even wasn't fixed by cold restart. During last test I didn't perform a cold restart, only the above commands.

growers12, 2025-08-20:
> It seems that the connection attempts cause an internal module crash in the wallbox which even wasn't fixed by cold restart, because during the first case today morning I tried only a cold restart and this didn't help. During last test I didn't perform a cold restart, only the above commands and it worked. Can reproduce this.

It seems that the EM2Go internal software crashes when there are multiple Modbus connections in parallel. As a workaround I use a second instance of EVCC as a pure modbus proxy and connect to this instead directly to the wallbox. This seems to help.

andig, 2025-08-20:
@growers12 there's no need for a second instance. One suffices.

growers12, 2025-08-21:
Checked the nightly, but still isn't working

```
[proxy-5022] DEBUG 2025/08/21 07:52:57 modbus proxy for 192.168.11.248:502 listening at :5022
[em2go ] TRACE 2025/08/21 07:52:57 modbus: send 00 01 00 00 00 09 ff 10 00 5b 00 01 02 00 3d
[proxy-5022] TRACE 2025/08/21 07:52:57 write holdings: id 255 addr 91 qty 1 val 003d
[em2go ] TRACE 2025/08/21 07:53:00 modbus: close connection: read tcp [::1]:51892->[::1]:5022: i/o timeout
[proxy-5022] TRACE 2025/08/21 07:53:00 write multiple holding: read tcp 192.168.11.146:58780->192.168.11.248:502: i/o timeout
```

Could be that the modbus proxy needs some time to start up and connect to the wallbox first.

andig, 2025-08-30:
Had to revert the early proxy start due to https://github.com/evcc-io/evcc/issues/23293#issuecomment-3238928079