evcc-io/evcc #17882: BMW X1 xDrive30E can't be created manually or in configuration GUI

StefanA-LU, 2024-12-24:
### Describe the bug

See logs. Entered every data as described including Captcha

### Steps to reproduce

1. Enter data manually in config file
2. Use GUI experimental configuration
3.
4.
...

### Configuration details

```yaml
Request failed with status code 400.
cannot create vehicle type 'template': cannot create vehicle type 'bmw': Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method): unexpected status: 401 (Unauthorized).
POST http://10.0.0.6:7070/api/config/test/vehicle
```

### Log details

```text
[bmw ] DEBUG 2024/12/24 09:46:00 identity.Login - no database token found. Proceeding to login via user, password and captcha.
[bmw ] TRACE 2024/12/24 09:46:00 POST https://customer.bmwgroup.com/gcdm/oauth/authenticate
[bmw ] TRACE 2024/12/24 09:46:01 client_id=31c357a0-7a1d-4590-aa99-33b97244d048&code_challenge=fXI17GfzrmS93r86kDGo_0Bq769-cQdTVzxKSM2Mn30&code_challenge_method=S256&grant_type=authorization_code&nonce=login_nonce&password=***&redirect_uri=com.bmw.connected%3A%2F%2Foauth&response_type=code&scope=openid+profile+email+offline_access+smacc+vehicle_data+perseus+dlm+svds+cesim+vsapi+remote_services+fupo+authenticate_user&state=cEG9eLAIi6Nv-aaCAniziE_B6FPoobva3qr5gukilYw&username=*** -- {"error": "invalid_client", "error_description": "Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method)"}
[bmw ] DEBUG 2024/12/24 09:53:30 identity.Login - no database token found. Proceeding to login via user, password and captcha.
[bmw ] TRACE 2024/12/24 09:53:30 POST https://customer.bmwgroup.com/gcdm/oauth/authenticate
[bmw ] TRACE 2024/12/24 09:53:30 client_id=31c357a0-7a1d-4590-aa99-33b97244d048&code_challenge=7BUz5vzSDzhfXd1dCDdDZwh1uMbbJYWIrrdkRzFcNO0&code_challenge_method=S256&grant_type=authorization_code&nonce=login_nonce&password=***&redirect_uri=com.bmw.connected%3A%2F%2Foauth&response_type=code&scope=openid+profile+email+offline_access+smacc+vehicle_data+perseus+dlm+svds+cesim+vsapi+remote_services+fupo+authenticate_user&state=cEG9eLAIi6Nv-aaCAniziE_B6FPoobva3qr5gukilYw&username=*** -- {"error": "invalid_client", "error_description": "Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method)"}
[bmw ] DEBUG 2024/12/24 09:53:33 identity.Login - no database token found. Proceeding to login via user, password and captcha.
[bmw ] TRACE 2024/12/24 09:53:33 POST https://customer.bmwgroup.com/gcdm/oauth/authenticate
[bmw ] TRACE 2024/12/24 09:53:33 client_id=31c357a0-7a1d-4590-aa99-33b97244d048&code_challenge=J3yYQXRsFPJz11SPqHinwblDRfWJIIQHpcjQkQiVGDE&code_challenge_method=S256&grant_type=authorization_code&nonce=login_nonce&password=***&redirect_uri=com.bmw.connected%3A%2F%2Foauth&response_type=code&scope=openid+profile+email+offline_access+smacc+vehicle_data+perseus+dlm+svds+cesim+vsapi+remote_services+fupo+authenticate_user&state=cEG9eLAIi6Nv-aaCAniziE_B6FPoobva3qr5gukilYw&username=*** -- {"error": "invalid_client", "error_description": "Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method)"}
```

### What type of operating system are you running?

Windows

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.12

andig, 2024-12-24:
Needs a trace log. Most likely wrong user/password.

StefanA-LU, 2024-12-24:
I logged out in the App and immediately logged in in the App again. Thereafter it worked with creating the vehicle in EVCC.

Maybe there should be a hint to just do a REAL logout in the App after weeks of only closing it. I found this in another thread and it worked. Should be added to the advices together with the captcha hints maybe.

closed 2024-12-25 by andig